openwrt/packages #23706: Preparations to switch to APK package manager

aparcar, 2024-03-21:
Hi all, some fellow developers and me worked for some time now on making [APK](https://gitlab.alpinelinux.org/alpine/apk-tools) the new package manager for OpenWrt, replacing the unmaintained OPKG fork we've been using for the longest time.

APK is actively developed and used in multiple other distributions, e.g. Alpine Linux 🎉 

While there is till some work ahead, I'd like to prepare everyone who maintains a package to verify that the `PKG_VERSION` follows [Semantic Versioning](https://semver.org). APK uses a deterministic algorithm to compare versions and does not like random strings, except a valid hash prefixed with a `~`.

If in doubt, please use the Docker container below to verify the used version is valid:

```
docker run --rm -it docker.io/aparcar/apk-version-check <VERSION> [<VERSION> ...]
```

It will print whatever version is not valid, if you get a zero exit code, you're fine.

Please feel free to reach out for assistance and have a look at the [core migration of versions](https://github.com/openwrt/openwrt/pull/14918).

aparcar, 2024-03-21:
Below packages that would cause trouble. Please bear in mind that `PKG_RELEASE` is soonish prefixed with an `r` and may only contain a number. 

**Naughty list:**

- [ ] ack                              `3.7.0+perl5.38-r1`
- [ ] atlas-probe                      `0~2.6.3-r1`
- [ ] bcm2711-eeprom                   `v.2024.01.05-2712-r2`
- [ ] bcm2712-eeprom                   `v.2024.01.05-2712-r2`
- [ ] bcm27xx-eeprom                   `v.2024.01.05-2712-r2`
- [ ] bluez-tools                      `20201025.f653217-r2`
- [ ] convbin                          `2.4.3_b34-r1`
- [ ] debootstrap                      `1.0.128+nmu2+deb12u1-r1`
- [ ] faad2                            `0~2.11.1-r1`
- [ ] fft-eval                         `2019-11-27-r4d3b6faee428e3bd9f44ab6a3d70585ec50484a1`
- [ ] hfsfsck                          `540.1.linux3-r2`
- [ ] https-dns-proxy                  `2023-11-19-r1`
- [ ] isc-dhcp-client-ipv4             `4.4.3-P1-r7`
- [ ] isc-dhcp-client-ipv6             `4.4.3-P1-r7`
- [ ] isc-dhcp-dyndns                  `4.4.3-P1-r7`
- [ ] isc-dhcp-omshell-ipv4            `4.4.3-P1-r7`
- [ ] isc-dhcp-omshell-ipv6            `4.4.3-P1-r7`
- [ ] isc-dhcp-relay-ipv4              `4.4.3-P1-r7`
- [ ] isc-dhcp-relay-ipv6              `4.4.3-P1-r7`
- [ ] isc-dhcp-server-ipv4             `4.4.3-P1-r7`
- [ ] isc-dhcp-server-ipv6             `4.4.3-P1-r7`
- [ ] ksmbd-avahi-service              `0~3.5.1-r1`
- [ ] ksmbd-hotplug                    `0~3.5.1-r1`
- [ ] ksmbd-server                     `0~3.5.1-r1`
- [ ] libcli                           `2022.07.06~V1.10.7`
- [ ] libdb47                          `4.7.25.4.NC-r7`
- [ ] libdb47xx                        `4.7.25.4.NC-r7`
- [ ] libedit                          `20230828-3.1-r1`
- [ ] libfaad2                         `0~2.11.1-r1`
- [ ] libinih                          `r58-r1`
- [ ] libinireader                     `r58-r1`
- [ ] libmad                           `0~0.16.4-r1`
- [ ] libmbim                          `0~1.30.0-r1`
- [ ] libmspack                        `0.10.1alpha-r1`
- [ ] libnet-1.2.x                     `1.2-rc3-r4`
- [ ] libopenssl-afalg_sync            `1.2.0-beta.1-r5`
- [ ] libow                            `3.2p4-r2`
- [ ] libow-capi                       `3.2p4-r2`
- [ ] libpagekite                      `0.91.201110C-r2`
- [ ] libqmi                           `0~1.34.0-r1`
- [ ] libradiotap                      `2020-06-22-r5`
- [ ] libuwifi                         `2020-03-10-r2`
- [ ] libuwsc-mbedtls                  `3.3.5-r5.1`
- [ ] libuwsc-nossl                    `3.3.5-r5.1`
- [ ] libuwsc-openssl                  `3.3.5-r5.1`
- [ ] libuwsc-wolfssl                  `3.3.5-r5.1`
- [ ] libvorbisidec                    `1.0.3-20180319-r2`
- [ ] libx264                          `2024-01-13-r1`
- [ ] lua-curl-v3                      `0.3.13-snapshot-r2`
- [ ] lua-openssl                      `0~0.8.5-1-r1`
- [ ] luajit                           `2.1.0-beta3-r7`
- [ ] luajit2                          `2.1-20231117-r1`
- [ ] luasoap                          `2014-08-21-raf1e100281cee4b972df10121e37e51d53367a98`
- [ ] luv                              `1.40.0-0-r2`
- [ ] mbim-utils                       `0~1.30.0-r1`
- [ ] mdio-tools                       `0~1.3.1-r1`
- [ ] mdns-utils                       `IETF104-r5`
- [ ] mdnsd                            `IETF104-r5`
- [ ] mdnsresponder                    `IETF104-r5`
- [ ] mkhfs                            `540.1.linux3-r2`
- [ ] modemmanager                     `0~1.22.0-r12`
- [ ] netdiscover                      `0.3-pre-beta7-r2`
- [ ] netifyd                          `2023.08.10~v4.4.7-r1`
- [ ] netjson-monitoring               `0~0.1.1-r2`
- [ ] node                             `v20.11.1-r1`
- [ ] node-npm                         `v20.11.1-r1`
[... 88 lines cut ...]
- [ ] softethervpn-bridge              `4.38-9760-r2`
- [ ] softethervpn-client              `4.38-9760-r2`
- [ ] softethervpn-server              `4.38-9760-r2`
- [ ] spamassassin                     `4.0.0+perl5.38-r1`
- [ ] spamc                            `4.0.0+perl5.38-r1`
- [ ] spamc-ssl                        `4.0.0+perl5.38-r1`
- [ ] sqm-scripts-extra                `2016-06-08-r1`
- [ ] squeezelite-custom               `1.9.9-1432-r2`
- [ ] squeezelite-dynamic              `1.9.9-1432-r2`
- [ ] squeezelite-full                 `1.9.9-1432-r2`
- [ ] sslh                             `v1.22c-r1`
- [ ] str2str                          `2.4.3_b34-r1`
- [ ] sudo                             `1.9.15p5-r1`
- [ ] svox                             `1.0+git20130326-r3`
- [ ] tinc                             `1.1pre18-r3`
- [ ] tmate-ssh-server                 `511fd2bd852464e76824279609a34ee93fe148a4-r2`
- [ ] tunneldigger-broker              `0~v0.4.0-r1`
- [ ] tunneldigger-broker-src          `0~v0.4.0-r1`
- [ ] tvheadend                        `2023-06-05-r2`
- [ ] udpxy                            `1.0-25.1-r1`
- [ ] v2ray-geoip                      `202403210039-1`
- [ ] v2ray-geosite                    `20240316051411-1`
- [ ] v2ray-geosite-ir                 `202403180026-1`
- [ ] xupnpd                           `2018-11-20-r2`
- [ ] zlog                             `0~1.2.17-r1`

brada4, 2024-03-21:
isc-dhcp is EOL https://www.isc.org/dhcp/

BKPepe, 2024-03-23:
Dear packages maintainers,

There is no need to open individual pull requests for your packages, you are maintaining. That will be really time-consuming for you and for us. We will do it here exactly the same as it was done in the main repo via treewide commit.

stokito, 2024-03-24:
I didn't know that the OpenWrt is going to switch to APK. Here is more details on this https://forum.openwrt.org/t/rfc-announcing-apkwrt/120096

pesa1234, 2024-03-25:
Dear, I've update ksmbd https://github.com/openwrt/packages/pull/23750, I'm not maintainer but if I can why don't help? Thanks guys

stangri, 2024-03-25:
@aparcar Just a heads up, most of my packages either:
* depended on the old style of version/revision to be presented by the binary to be compatible with upstream-maintained products (client to a server or a peer in a mixed network for example)
* contained dependencies on opkg and were incompatible with upcoming APK in the ways other than version/release number

I believe I have fixed most of those, but I may have to submit further PRs if needed.

Also, the https-dns-proxy should be fixed with the latest PR. 

PS. If the PR needs to be made against packages/luci repo for reasons other than APK migration, should I refrain from moving to the new release syntax?

neheb, 2024-04-07:
libmad was fixed.

pesa1234, 2024-04-07:
ksmbd-* fixed

neheb, 2024-04-21:
I noticed mtd-rw has a bad version too. https://github.com/openwrt/packages/pull/23977 fixes.

ttc0419, 2024-06-12:
Quick question, is 24x going to use apk?

aparcar, 2024-06-12:
> Quick question, is 24x going to use apk?

No final decision on this.

stangri, 2024-07-09:
> > Quick question, is 24x going to use apk?
> 
> No final decision on this.

If 24.xx is NOT going to use APK by default, would the 24.xx building tools (toolchain, SDK, CI) still prepend `r` to the `PKG_RELEASE`?

aparcar, 2024-08-13:
@openwrt/packages-write I kindly ask every maintainer to check the failing package builds here and see if the PKG_VERSION need some adoption: https://buildbot.aparcar.org/faillogs/x86_64/packages/

hnyman, 2024-08-14:
@aparcar
In addition of versioning, there appears to be also dependency trouble.
Some packages depend explicitly on opkg, and building an actual firmware will fail if apk is selected. Apk also finds more conflicts that opkg.

I built yesterday an APK enabled firmware, and had to remove one package (openssh sftp server) due to the versioning reasons, like expected.
But then at image finalization I ran into conflicts with e.g. @dibdot popular 'adblock' which explicitly depends on opkg.  
Similarly, some packages depend on uclient-fetch,, which apk finds to be in conflict with GNU wget, unlike opkg does.

ErwanMAS, 2024-08-14:
> @aparcar
> 
> ```
> root@debian:~# docker run --rm -it ghcr.io/aparcar/apk-valid-version
> Unable to find image 'ghcr.io/aparcar/apk-valid-version:latest' locally
> docker: Error response from daemon: Head "https://ghcr.io/v2/aparcar/apk-valid-version/manifests/latest": unauthorized.
> See 'docker run --help'.
> ```

@aparcar i want to try your docker image and i have this error ^

dibdot, 2024-08-14:
> But then at image finalization I ran into conflicts with e.g. @dibdot popular 'adblock' which explicitly depends on opkg.

@hnyman I'll remove the opkg dependency in adblock with the next minor update - thanks for heads up!

nxhack, 2024-08-27:
node, node-npm: I have sent a pull request.

cotequeiroz, 2024-08-27:
libopenssl-afalg_sync:  fixed by #24897

neheb, 2024-10-22:
Unlikely unfortunately.

Neustradamus, 2024-11-04:
For the moment, it is not sure for 24.10.x?

hnyman, 2024-11-09:
List of still failing packages can be seen from the test/staging apk buildbot faillogs:
https://downloads.staging.openwrt.org/snapshots/faillogs/

only a few architectures have been compiled so far, but seems that e.g. ddns-scripts, nginx, ntfs-3g, many perl packages etc. still fail due to versioning reasons.

vortexilation, 2024-11-11:
Does anyone have a clue about perl-file-next or perl related packages?

```
ERROR: info field 'version' has invalid value: package version is invalid
ERROR: failed to create package: /home/user/works/openwrt/bin/packages/x86_64/packages/perl-file-next-1.18+perl5.40-r1.apk: package version is invalid
```

Looking at for example perl-file-next Makefile PKG_VERSION, i don't see anything wrong with it : 
```
PKG_NAME:=perl-file-next
PKG_VERSION:=1.18
PKG_RELEASE:=1
```

What would be the fixes then?.

aparcar, 2024-11-11:
Seems like you attach `perl5.40` to the version itself, which doesn't fly. How about something like `perl-file-next-1.18.5.40-r1`? This is what we did for some kernel packages, first the package version attached by the Kernel. You may not any arbitrary strings to the version anymore.

1715173329, 2024-11-11:
guess we need change
https://github.com/openwrt/packages/blob/ca503cc4054d9a13558c7b552886a6dba359a0eb/lang/perl/perlmod.mk#L11
to `PKG_VERSION:=$(PKG_VERSION).$(PERL_VERSION2)`

vortexilation, 2024-11-11:
@aparcar 
Thank you, will be helpful for fixing other packages version related things.

@1715173329 
Compiled fine now, thanks, I think you should open a PR as using APK is becoming default now.

hnyman, 2024-11-12:
> How to fix nginx makefile?

By removing the extra version comparison there.
@Ansuel probably thinks about better fix, but that should be enough for your own use.

(Not sure if actually just "1.5-r1" would be enough there, as old "1.5-1" is not accepted by apk)

Ra2-IFV, 2024-11-13:
I see a lot of package use this:
```
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
```
Then `PKG_VERSION` must match upstream version, this breaks `apk` support if upstream version use other naming syntax. Do we have a solution for this?

Ra2-IFV, 2024-11-13:
> I found that when compiling nginx-ssl I get an error with the following output
> 
> `"depends:nginx-ssl-util>=1.5-1 libc libopenssl3 libpthread libpcre2 nginx-ssl-util zlib"`
> 
> How to fix nginx makefile?

#25335

hnyman, 2024-11-13:
If you check the currently failing packages in faillogs, you notice that there is substantial variation to the upstream names.  There is no single cure.
https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/

hnyman, 2024-11-17:
The list of currently failing packages in the packages feed is already below 10 packages.

https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/

`debootstrap/ hfsprogs/ kafs-client/ micropython-lib/ netifyd/ oci-runtime-tools/ python-dateutil/ qcsuper/ sox/`

( and also lua-curl-v3 and netdiscover, but there is already a PR for them )

Ansuel, 2024-11-17:
@hnyman always dreap to have 0 faillogs there... at least for some targets :(

Ansuel, 2024-11-17:
This error looks interesting

https://downloads.openwrt.org/snapshots/faillogs/x86_64/telephony/pcapsipdump/compile.txt

Ansuel, 2024-11-17:
`/home/ansuel/openwrt-ansuel/staging_dir/host/bin/fakeroot /home/ansuel/openwrt-ansuel/staging_dir/host/bin/apk mkpkg --info "name:pcapsipdump" --info "version:2020.03.03~157-r1" --info "description:pcapsipdump is a tool for dumping SIP sessions (plus RTP traffic, if available) to disk in a fashion similar to "tcpdump -w" (format is exactly the same), but one file per SIP session (even if there are thousands of concurrent SIP sessions)." --info "arch:aarch64_cortex-a53" --info "license:GPL-2.0+" --info "origin:feeds/telephony/net/pcapsipdump" --info "url:http://sourceforge.net/projects/pcapsipdump/" --info "maintainer:" --info "provides:"  --script "post-install:/home/ansuel/openwrt-ansuel/build_dir/target-aarch64_cortex-a53_musl/pcapsipdump-2020.03.03~157/apk-aarch64_cortex-a53/pcapsipdump/post-install" --script "pre-deinstall:/home/ansuel/openwrt-ansuel/build_dir/target-aarch64_cortex-a53_musl/pcapsipdump-2020.03.03~157/apk-aarch64_cortex-a53/pcapsipdump/pre-deinstall" --info "depends:libc libstdcpp6 libpcap1" --files "/home/ansuel/openwrt-ansuel/build_dir/target-aarch64_cortex-a53_musl/pcapsipdump-2020.03.03~157/ipkg-aarch64_cortex-a53/pcapsipdump" --output "/home/ansuel/openwrt-ansuel/bin/packages/aarch64_cortex-a53/telephony/pcapsipdump-2020.03.03~157-r1.apk" --sign "/home/ansuel/openwrt-ansuel/private-key.pem"`

Ok we need to escape the description...

Ansuel, 2024-11-17:
@hnyman I sanitized the package description and also pushed a commit that escape them just to handle anything that is not officially supported.

hnyman, 2024-11-23:
All packages in main/master should now be fixed for apk versioning.
In the last two weeks I have patched the last remaining dozen failing packages or so.

Buildbot doesn't currently show any apk related failures for main OpenWrt, LuCI, packages & telephony. 
In routing feed there is still PR https://github.com/openwrt/routing/pull/1089 waiting for @aparcar @Ansuel or somebody else with rights to the routing feed repo.

I have backported my own commits to 24.10, but I suspect that some other fixes are still failing there., as fixes have not been backported. (Note: Buildbot does not yet show failures there, as opkg is still currently the default in 24.10.)

qyl27, 2025-02-05:
Still no document for migrating to the alpine package manager for existing systems?

aparcar, 2025-02-05:
@qyl27 did you login to a snapshot device recently? https://github.com/openwrt/openwrt/commit/c19c31f7aa6f9952b5d5e1b313eca79d4e7ed59c

qyl27, 2025-02-05:
> [@qyl27](https://github.com/qyl27) did you login to a snapshot device recently? [openwrt/openwrt@c19c31f](https://github.com/openwrt/openwrt/commit/c19c31f7aa6f9952b5d5e1b313eca79d4e7ed59c)

I have read the cheatsheet at https://openwrt.org/docs/guide-user/additional-software/opkg-to-apk-cheatsheet. So will there be some documents for migrating existing packages and configurations?

hnyman, 2025-02-05:
> will there be some documents for migrating existing packages and configurations?

Not quite sure what you mean. No special migration is generally needed. 

In main/master all existing packages are already compiling ok with apk enabled. In general, the packages themselves have usually not changed, only the version strings have been modified to match apk requirements.  

From end-user perspective you just create a new current main/master image that has apk by default and include the same packages in the image as earlier. (or install them afterward with apk)

(Regarding package contents & authoring packages: possible issues might only arise if you have a package that includes scripts that explicitly use opkg for something.)

qyl27, 2025-02-05:
> Not quite sure what you mean. No special migration is generally needed.
>
> In main/master all existing packages are already compiling ok with apk enabled. In general, the packages themselves have [...]

Ok, I see, thank you for your explainations.

aparcar, 2025-03-06:
I'll unpin this, things look good

Djfe, 2025-03-17:
Tunneldigger-Broker was fixed a few months back
https://github.com/openwrt/packages/commit/bec27934112d5ebfcc7d3914aab4f73804d76438

here the same for tinc:
https://github.com/openwrt/packages/commit/3d6cfc732b67549a770b5a4016024ebaef51dfda

qcsuper:
https://github.com/openwrt/packages/commit/6ed915b06067e0a5c37d6bada2869f07d98cb300

python3-dateutil:
https://github.com/openwrt/packages/commit/40cfff8fe9594c4507db705111461973a9bbb0f0

netifyd:
https://github.com/openwrt/packages/commit/7ce6b6866e07b87dda76b769e03b0c3386fb2257

luasoap:
https://github.com/openwrt/packages/commit/30f8c9a769072e24f77c5849b4f13ba0a2db77ee

luajit (not luajit2):
https://github.com/openwrt/packages/commit/fc130a0d32301a788be6f7996bba9d0f75297ad8

lua-openssl:
https://github.com/openwrt/packages/commit/c8a46db23df2aa932c2935796d71e804a72e68a5

libow:
https://github.com/openwrt/packages/commit/252efdc342432e50026c9116762d86b0c2448a6d

slightly offtopic question:
Will manual apk installations from /tmp/ always require "--allow-untrusted"? Or will OpenWrt apk files (from official builds) eventually be signed in a way that installing them isn't deemed untrusted?

aparcar, 2025-03-18:
> Will manual apk installations from /tmp/ always require "--allow-untrusted"? Or will OpenWrt apk files (from official builds) eventually be signed in a way that installing them isn't deemed untrusted?

We use a lot of remote build workers and while we could use some CA style, right now we only sign indexes with a trusted key.

BKPepe, 2025-05-21:
Can we close this issue or there are some remaining things, which needs to be fixed?

closed 2025-05-22 by hnyman