openwrt/packages #23707: Lighttpd directory listening bug in 1.4.74-1 version

Loafter, 2024-03-21:
Looks like we have bug with directory listening feature 
it fixed in next version
https://redmine.lighttpd.net/issues/3242

BKPepe, 2024-03-21:
Which OpenWrt version do you have? It is unclear from your post. There was issue template, which you completely removed. We do have Lighttpd version 1.4.75 in the master branch.

Loafter, 2024-03-22:
OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-24.073.29889-cd7e519
[image]

Loafter, 2024-03-22:
[image]

Loafter, 2024-03-22:
> Which OpenWrt version do you have? It is unclear from your post. There was issue template, which you completely removed. We do have Lighttpd version 1.4.75 in the master branch.

[image]

gstrauss, 2024-03-23:
Sorry for the regression.  #23724 and #23725 filed for backports.

BKPepe, 2024-03-23:
- OpenWrt 23.05: ed3cba66e18632197e862b94c04ea2fe21fbc24f
- OpenWrt 22.03: 5a20abc059c72e4c2c27e1a5ef2a1225559e19f8

closed 2024-03-23 by BKPepe

gstrauss, 2024-03-23:
@BKPepe Thank you for the correction.