openwrt/packages #23713: Update package ovpn-dco for support 6.6 kernel

remittor, 2024-03-22:
Package `ovpn-dco` needed update for support 6.6 kernel.
Makefile changes:
```
PKG_VERSION:=0.2.20240320
PKG_HASH:=83a02dc3e6e40b0ef128cd32ce7f47da7fccd759af68657f44925d64a88db37b
```

BKPepe, 2024-03-23:
Feel free to submit PR.

MaxSKash, 2024-03-27:
`/root/work/wrt/x86/build_dir/target-x86_64_musl/linux-x86_64/ovpn-dco-0.2.20240320/drivers/net/ovpn-dco/ovpn.c:25:10: fatal error: net/gso.h: No such file or directory
   25 | #include <net/gso.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[4]: *** [scripts/Makefile.build:250: /root/work/wrt/x86/build_dir/target-x86_64_musl/linux-x86_64/ovpn-dco-0.2.20240320/drivers/net/ovpn-dco/ovpn.o] Error 1
`

trippleflux, 2024-03-27:
Use this patch : 

001-use_gso_header.patch : 
```
diff -Naur a/linux-compat.h b/linux-compat.h
--- a/linux-compat.h
+++ b/linux-compat.h
@@ -88,6 +88,10 @@
 
 #endif /* LINUX_VERSION_CODE < KERNEL_VERSION(5, 10, 0) && !defined(EL8) */
 
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 4, 10)
+#include_next <net/gso.h>
+#endif
+
 #if LINUX_VERSION_CODE < KERNEL_VERSION(5, 6, 0)
 
 /* Iterate through singly-linked GSO fragments of an skb. */
```

MaxSKash, 2024-03-27:
Makefile:
```
NOSTDINC_FLAGS += \
  $(KERNEL_NOSTDINC_FLAGS) \
     -I$(PKG_BUILD_DIR)/include \
 +   -I$(PKG_BUILD_DIR)/compat-include \
     -include $(PKG_BUILD_DIR)/linux-compat.h
```

So I compiled it. Still doesn't work.