openwrt/packages #23721: Nginx compilation error on mipsel but works good on filogic

zcy85611, 2024-03-23:
it seems to have errors on compile ngx_http_dav_ext_module.so,  could anyone help to look into this please? thanks

```
make[3]: Entering directory '/openwrt/feeds/packages/net/nginx'
rm -f /openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3/.built
touch /openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3=nginx-1.25.3 -mips16 -minterlink-mips16 -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -DNGX_LUA_NO_BY_LUA_BLOCK  -I/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/include -I/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include -I/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include/fortify   " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3=nginx-1.25.3 -mips16 -minterlink-mips16 -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -DNGX_LUA_NO_BY_LUA_BLOCK -mips16 -minterlink-mips16 -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects  -I/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/include -I/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include -I/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include/fortify   " LDFLAGS="-L/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/lib -L/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/lib -fuse-ld=bfd -Wl,--gc-sections -flto=auto -fuse-linker-plugin -znow -zrelro   " make  -C /openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3/. AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3=nginx-1.25.3 -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNGX_LUA_NO_BY_LUA_BLOCK" LD="mipsel-openwrt-linux-musl-ld.bfd" NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CROSS="mipsel-openwrt-linux-musl-" ARCH="mipsel" ;
make[4]: Entering directory '/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3'
make -f objs/Makefile
make[5]: Entering directory '/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3'
mipsel-openwrt-linux-musl-gcc -o objs/ngx_http_dav_ext_module.so \
objs/addon/nginx-mod-dav-ext/ngx_http_dav_ext_module.o \
objs/ngx_http_dav_ext_module_modules.o \
-L/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/lib -L/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/lib -fuse-ld=bfd -Wl,--gc-sections -flto=auto -fuse-linker-plugin -znow -zrelro -lxml2 \
-shared
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_strip_uri':
<artificial>:(.text+0x3a): relocation R_MIPS16_26 against `strncmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x4e): relocation R_MIPS16_26 against `strncmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x70): relocation R_MIPS16_26 against `strncmp' cannot be used when making a shared object; recompile with -fPIC
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_format_token':
<artificial>:(.text+0xce): relocation R_MIPS16_26 against `memcpy' cannot be used when making a shared object; recompile with -fPIC
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_propfind_xml_start':
<artificial>:(.text+0x128): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x13a): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x14c): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x15e): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_propfind_xml_end':
<artificial>:(.text+0x19a): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x1ac): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x1be): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x1d0): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x1e2): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x1f4): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x22e): relocation R_MIPS16_26 against `a local symbol' cannot be used when making a shared object; recompile with -fPIC
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_depth':
<artificial>:(.text+0x282): relocation R_MIPS16_26 against `ngx_log_error_core' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x296): relocation R_MIPS16_26 against `strcmp' cannot be used when making a shared object; recompile with -fPIC
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_init':
<artificial>:(.text+0x2c8): relocation R_MIPS16_26 against `ngx_array_push' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x2de): relocation R_MIPS16_26 against `ngx_array_push' cannot be used when making a shared object; recompile with -fPIC
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_if':
<artificial>:(.text+0x3a8): relocation R_MIPS16_26 against `a local symbol' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x3da): relocation R_MIPS16_26 against `memcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x412): relocation R_MIPS16_26 against `strncmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x44e): relocation R_MIPS16_26 against `strncmp' cannot be used when making a shared object; recompile with -fPIC
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_lock_lookup.constprop.0':
<artificial>:(.text+0x4f6): relocation R_MIPS16_26 against `ngx_slab_free_locked' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x50e): relocation R_MIPS16_26 against `memcmp' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x564): relocation R_MIPS16_26 against `memcmp' cannot be used when making a shared object; recompile with -fPIC
/openwrt/tmp/ccWMQ06V.ltrans0.ltrans.o: in function `ngx_http_dav_ext_verify_lock':
<artificial>:(.text+0x5a2): relocation R_MIPS16_26 against `a local symbol' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x5be): relocation R_MIPS16_26 against `ngx_shmtx_lock' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x5ca): relocation R_MIPS16_26 against `a local symbol' cannot be used when making a shared object; recompile with -fPIC
<artificial>:(.text+0x5d4): relocation R_MIPS16_26 against `ngx_shmtx_unlock' cannot be used when making a shared object; recompile with -fPIC
[... cut ...]
make[3]: *** [Makefile:461: /openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3/.built] Error 2
make[2]: *** [package/Makefile:129: package/feeds/packages/nginx/compile] Error 1
make[5]: Leaving directory '/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3'
make[4]: Leaving directory '/openwrt/build_dir/target-mipsel_24kc_musl/nginx-ssl/nginx-1.25.3'
make[3]: Leaving directory '/openwrt/feeds/packages/net/nginx'
time: package/feeds/packages/nginx/ssl/compile#0.46#0.09#0.53
    ERROR: package/feeds/packages/nginx failed to build (build variant: ssl).
```

brada4, 2024-03-23:
Can you follow issue template, i looked at your copypasta, so you can close this issue.

ptpt52, 2024-03-24:
@brada4 @zcy85611 @Rondom @neheb 
the same build error on my mt7621/ramips build.
since commit: https://github.com/openwrt/packages/commit/3b13b08ad98dd15999e83c98f80674c9965d7a10

neheb, 2024-03-26:
Why?

anyway, this just sounds like missing fPIC.