openwrt/packages #23722: nginx: Feature request - add module fancyindeex

alex9434, 2024-03-23:
Maintainer: @Ansuel @neheb @hnyman @jefferyto @dangowrt @thess @commodo @BKPepe @feckert @sbyx 
Environment: OpenWrt 23.05 and snapshot

Tried to include ngx-fancyindex to the nginx build
https://github.com/aperezdc/ngx-fancyindex

Here are the modifications I made to the existing Nginx Makefile (... to indicate unchanged parts)
```
...
PKG_MOD_EXTRA := \
	...
	fancyindex \
	...

PKG_CONFIG_DEPENDS := \
	...
	CONFIG_NGINX_HTTP_ADDITION \
	...

define Download/nginx-mod-fancyindex
  VERSION:=6cdffdb0af493b7c21eedba2f09d8486579f3922
  URL:=https://github.com/aperezdc/ngx-fancyindex.git
  MIRROR_HASH:=skip
  PROTO:=git
endef

...
	$(if $(call IsEnabled,NGINX_HTTP_ADDITION),--with-http_addition_module) \
...
$(eval $(call BuildModule,fancyindex,+@NGINX_HTTP_ADDITION,nginx-fancyindex, \
	Nginx Fancy Index module.))
...
```
I also added the entry for http_addition to Config_ssl.in.

Unfortunately the build does not work. I get an error that the ipk for Fancyindex cannot be found. For some reason the makefile does not build fancyindex. What am I doing wrong?

Any help greatly appreciated!

brada4, 2024-03-23:
You should ask in forums.
Guide https://openwrt.org/docs/guide-developer/packages describes a step or five you left out.

commodo, 2024-03-24:
can you post the entire Makefile?
maybe that helps to see things easier;

alex9434, 2024-03-25:
Hi @commodo and @brada4 . Happy to post it. As I said, digest is working fine but fancyindex is not. I also upgraded some of the other modules to the most recent version and applied an additional patch from cloudflare to speed-up http2 upload to cloudflare CDN and added MP4 pseudo streaming.

Makefile
```
#
# Copyright (C) 2012-2016 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=nginx
PKG_VERSION:=1.21.3
PKG_RELEASE:=2

PKG_SOURCE:=nginx-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://nginx.org/download/
PKG_HASH:=14774aae0d151da350417efc4afda5cce5035056e71894836797e1f6e2d1175a

PKG_MAINTAINER:=Thomas Heil <heil@terminal-consulting.de> \
				Christian Marangi <ansuelsmth@gmail.com>
PKG_LICENSE:=2-clause BSD-like license
PKG_CPE_ID:=cpe:/a:nginx:nginx

PKG_FIXUP:=autoreconf
PKG_BUILD_PARALLEL:=1
PKG_INSTALL:=1
PKG_BUILD_FLAGS:=gc-sections

# 3rd-party modules
PKG_MOD_EXTRA := \
	geoip2 \
	lua \
	lua-resty-core \
	lua-resty-lrucache \
	rtmp \
	dav-ext \
	fancyindex \
	naxsi \
	brotli \
	digest \
	headers-more \
	ts \
	ubus
# official and 3rd-party modules
PKG_MOD_ALL := \
	$(PKG_MOD_EXTRA) \
	stream
PKG_MOD_PATCHED := $(shell find patches -mindepth 1 -maxdepth 1 -type d -name 'nginx-mod-*' | sed -E 's,^patches/nginx-mod-,,')

PKG_CONFIG_DEPENDS := \
	CONFIG_NGINX_DAV \
	CONFIG_NGINX_FLV \
	CONFIG_NGINX_MP4 \
	CONFIG_NGINX_STUB_STATUS \
	CONFIG_NGINX_HTTP_CHARSET \
	CONFIG_NGINX_HTTP_GZIP \
	CONFIG_NGINX_HTTP_GZIP_STATIC \
	CONFIG_NGINX_HTTP_SSI \
	CONFIG_NGINX_HTTP_USERID \
	CONFIG_NGINX_HTTP_ACCESS \
	CONFIG_NGINX_HTTP_AUTH_BASIC \
	CONFIG_NGINX_HTTP_AUTH_REQUEST \
	CONFIG_NGINX_HTTP_AUTOINDEX \
	CONFIG_NGINX_HTTP_GEO \
	CONFIG_NGINX_HTTP_MAP \
	CONFIG_NGINX_HTTP_SPLIT_CLIENTS \
	CONFIG_NGINX_HTTP_REFERER \
[... 673 lines cut ...]
	prompt "Enable PCRE library usage"
	default y

config NGINX_HTTP_REAL_IP
	bool
	prompt "Enable HTTP real ip module"
	default n

config NGINX_HTTP_SECURE_LINK
	bool
	prompt "Enable HTTP secure link module"
	default n

config NGINX_HTTP_SUB
	bool
	prompt "Enable HTTP sub module"
	default n

config NGINX_STREAM_REAL_IP
	bool
	prompt "Enable STREAM real ip module"
	default n

endmenu
```

brada4, 2024-03-25:
You need to fix your build directory. You are building v22 package tree on v23. See:
https://github.com/openwrt/packages/blob/53896bd1a7cf7d886cafcd42178d58ada4d6e697/net/nginx/Makefile#L11
https://github.com/openwrt/packages/blob/1f10633db02f03b4e4f18cb83ca94fc838dc985d/net/nginx/Makefile#L11

alex9434, 2024-03-25:
Hi @brada4 
I used the the Makefile from staging because it creates the loadable modules. The older Makefile does not do that. The fancyindex module also does not compile with the latest stable version of nginx.

brada4, 2024-03-25:
Im afraid you go to forums. To request new package/option you have to provide a working PR.

closed 2025-05-21 by BKPepe