openwrt/packages #23733: schroot:  Chroot setup failed: stage=setup-start

alex9434, 2024-03-24:
Maintainer: @jmarcet @ptpt52 @neheb 
Environment: OpenWrt 23.05.2

Description: Unable to start schroot environment: Chroot setup failed: stage=setup-start
 
1. Create chroot with debootstrap --no-check-gpg --arch=arm64 bullseye /srv/chroot/android
2. Create file in /etc/schroot/chroot.d
```[android]
description=Android dev environment
directory=/srv/chroot/android
aliases=bullseye
type=directory
users=root
groups=root
root-groups=root
```
3. Start with schroot -c android
```E: 20copyfiles: /etc/schroot/setup.d/20copyfiles: line 42: /bin/stat: not found
E: 70services: /etc/schroot/setup.d/70services: line 48: tac: not found
E: android-1711263744-18335: Chroot setup failed: stage=setup-start
```

brada4, 2024-03-29:
Install coreutils-stat and coreutils-tac as hinted by those messages.

alex9434, 2024-04-05:
@jmarcet Apologies for the delay in getting back. Yes, installing coreutils-stat & coreutils-tac solved the issue. Since I am building from source I was thinking about just using stat and tac from busybox. Would that also work? 

I only now get an error message because ash is not included in the debian setup.
```
W: Shell ‘/bin/ash’ not available: /bin/ash: Failed to stat file: No such file or directory
W: Falling back to shell ‘/bin/bash’
```
schell=/bin/bash in the config solved the issue. You might want to include that in the package.

There is an additional dependency on coreutils-realpath when exiting the schroot:
```
E: 15killprocs: /etc/schroot/setup.d/15killprocs: line 42: realpath: not found
E: android-1712316017-2771: Chroot setup failed: stage=setup-stop
```
Installing coreutils-realpath solved the issue. 

The only question would be whether to include in the Makefile the dependency on Busybox or coreutils. I think we can close the issue now.

jmarcet, 2024-04-05:
> @jmarcet Apologies for the delay in getting back. Yes, installing `coreutils-stat` & `coreutils-tac` solved the issue. Since I am building from source I was thinking about just using stat and tac from busybox. Would that also work?

@alex9434 See the [PR](https://github.com/openwrt/packages/pull/23793) I've opened, it includes an updated `01-openwrt.patch` which brings compatibility with busybox's stat (built with `BUSYBOX_CONFIG_FEATURE_STAT_FORMAT` option).

> I only now get an error message because ash is not included in the debian setup.
> 
> ```
> W: Shell ‘/bin/ash’ not available: /bin/ash: Failed to stat file: No such file or directory
> W: Falling back to shell ‘/bin/bash’
> ```
> 
> `shell=/bin/bash` in the config solved the issue. You might want to include that in the package.

Where does that `ash` come from? I can't find it within `/etc/schroot` or any other of the files in the `schroot` ipk package, which is what I could check now.

> There is an additional dependency on coreutils-realpath when exiting the schroot:
> 
> ```
> E: 15killprocs: /etc/schroot/setup.d/15killprocs: line 42: realpath: not found
> E: android-1712316017-2771: Chroot setup failed: stage=setup-stop
> ```
> 
> Installing `coreutils-realpath` solved the issue.

Great! Thanks for the feedback :) I had missed that one.

> The only question would be whether to include in the Makefile the dependency on Busybox or coreutils. I think we can close the issue now.

In the [PR](https://github.com/openwrt/packages/pull/23793) I've added both as options, it is missing `realpath` ATM though.

alex9434, 2024-04-05:
> Where does that `ash` come from? I can't find it within `/etc/schroot` or any other of the files in the `schroot` ipk package, which is what I could check now.
> 
ash is the default shell in OpenWrt. I think schroot tries to call the same shell.  However, debootstrap installs only bash by default hence the error message.

brada4, 2024-04-05:
`SHELL=/bin/sh chroot bootstrap`

jmarcet, 2024-04-06:
> ash is the default shell in OpenWrt. I think schroot tries to call the same shell. However, debootstrap installs only bash by default hence the error message.

I'm used to specify which command I want to execute with `schroot`, be it a shell or any other command. I was not remembering it defaults to the host's `$SHELL`.

> BTW: realpath is also available in Busybox

I've updated the [PR](https://github.com/openwrt/packages/pull/23793) to also include `realpath`, whether from `busybox` or from `coreutils`.

alex9434, 2024-05-12:
Hi @jmarcet , I found another issue:

```
E: 20copyfiles: cp: unrecognized option: preserve=all
E: 20copyfiles: BusyBox v1.36.1 (2024-03-22 22:09:42 UTC) multi-call binary.
E: 20copyfiles: 
E: 20copyfiles: Usage: cp [-arPLHpfinlsTu] SOURCE DEST
E: 20copyfiles: or: cp [-arPLHpfinlsu] SOURCE... { -t DIRECTORY | DIRECTORY }
E: 20copyfiles: 
E: 20copyfiles: Copy SOURCEs to DEST
E: 20copyfiles: 
E: 20copyfiles: 	-a	Same as -dpR
E: 20copyfiles: 	-R,-r	Recurse
E: 20copyfiles: 	-d,-P	Preserve symlinks (default if -R)
E: 20copyfiles: 	-L	Follow all symlinks
E: 20copyfiles: 	-H	Follow symlinks on command line
E: 20copyfiles: 	-p	Preserve file attributes if possible
E: 20copyfiles: 	-f	Overwrite
E: 20copyfiles: 	-i	Prompt before overwrite
E: 20copyfiles: 	-n	Don't overwrite
E: 20copyfiles: 	-l,-s	Create (sym)links
E: 20copyfiles: 	-T	Refuse to copy if DEST is a directory
E: 20copyfiles: 	-t DIR	Copy all SOURCEs into DIR
E: 20copyfiles: 	-u	Copy only newer files
E: ubuntu-1715273858-15347: Chroot setup failed: stage=setup-start
```
It is caused by busybox. I solved the errror by replacing `preserve=all` with `-a`.

brada4, 2024-05-12:
First gather more edits needed by busybox, sure worth adding patch toward flawless ux