openwrt/packages #23735: Failed to configuring knot package

trippleflux, 2024-03-24:
I don't change much of my OpenWRT .config but somehow suddenly after using latest OpenWRT snapshot 8753022aeae14eaee02e0c1e59244c5b36e454db for x86_64 , I am getting the following when configuring knot at the end of packages compilation : 
```
Configuring strongswan-mod-cmaCollected errors:
 * resolve_conffiles: Existing conffile /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/root-x86/etc/ppp/options-opkg.
 * resolve_conffiles: Existing conffile /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/root-x86/etc/config/crowdsec is different from the conffile in the new package. The new conffile will be placed at /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/root-x86/etc/config/crowdsec-opkg.
 * pkg_hash_check_unresolved: cannot find dependency knot-libs (= 3.3.5-1) for knot
 * pkg_hash_check_unresolved: cannot find dependency knot-libzscanner (= 3.3.5-1) for knot
 * pkg_hash_fetch_best_installation_candidate: Packages for knot found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package knot.
 * pkg_hash_check_unresolved: cannot find dependency knot-libs (= 3.3.5-1) for knot-dig
 * pkg_hash_fetch_best_installation_candidate: Packages for knot-dig found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package knot-dig.
 * pkg_hash_check_unresolved: cannot find dependency knot-libs (= 3.3.5-1) for knot-host
 * pkg_hash_fetch_best_installation_candidate: Packages for knot-host found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package knot-host.
 * pkg_hash_check_unresolved: cannot find dependency knot-libs (= 3.3.5-1) for knot-nsupdate
 * pkg_hash_check_unresolved: cannot find dependency knot-libzscanner (= 3.3.5-1) for knot-nsupdate
 * pkg_hash_fetch_best_installation_candidate: Packages for knot-nsupdate found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package knot-nsupdate.
 * pkg_hash_check_unresolved: cannot find dependency knot-libs (= 3.3.5-1) for knot-zonecheck
 * pkg_hash_check_unresolved: cannot find dependency knot-libzscanner (= 3.3.5-1) for knot-zonecheck
 * pkg_hash_fetch_best_installation_candidate: Packages for knot-zonecheck found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package knot-zonecheck.
 * pkg_hash_check_unresolved: cannot find dependency knot-libs (= 3.3.5-1) for knot-keymgr
 * pkg_hash_check_unresolved: cannot find dependency knot-libzscanner (= 3.3.5-1) for knot-keymgr
 * pkg_hash_fetch_best_installation_candidate: Packages for knot-keymgr found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package knot-keymgr.
 * pkg_hash_check_unresolved: cannot find dependency knot-libs (= 3.3.5-1) for knot-tests
 * pkg_hash_check_unresolved: cannot find dependency knot-libzscanner (= 3.3.5-1) for knot-tests
 * pkg_hash_fetch_best_installation_candidate: Packages for knot-tests found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package knot-tests.
 * pkg_alternatives_update_path: /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/root-x86/usr/bin/groups exists but is not a symlink
c.
Configuring strongswan-mod-ctr.
```

Perhaps this is related with [recent apk commit](https://github.com/openwrt/openwrt/commit/e8725a932e16eaf6ec51add8c084d959cbe32ff2) ?.

BKPepe, 2024-03-24:
Could be, it seems there is now missing character "r" in ``EXTRA_DEPENDS`` as it done in the commit, which you referenced: https://github.com/openwrt/openwrt/pull/14918/files#diff-30d3ce228a1284069df1e00e9a7911b595532052a822a01b546f9007b3a30462R258 , so thats will be probably fixed by doing treewide commit here as well.

trippleflux, 2024-03-24:
It's compiling, thanks for the pointer.

closed 2024-06-23 by neheb