openwrt/packages #23756: luci-app-acme: Page is not loading

JKSTAFF, 2024-03-26:
## Steps to reproduce:
1. Make a fiemware with luci-app-acme and its depends build in
2. go to: Services → ACME certs
[image]

## Actual behavior:
In browser luci shows 
```
RPCError
RPC call to uci/get failed with ubus code 9: Unspecified error
  at ClassConstructor.handleCallReply (https://192.168.0.1/luci-static/resources/rpc.js?v=git-24.080.71710-b07cf9d:15:3)
```

## Expected behavior:
ACME gui displayed correctly

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05-SNAPSHOT'
DISTRIB_REVISION='r23812-0a2047cf77'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05-SNAPSHOT r23812-0a2047cf77'
DISTRIB_TAINTS='no-all'
```

saudiqbal, 2024-03-27:
Same problem when installing the package. https://forum.openwrt.org/t/installing-luci-app-acme-not-working-on-23-05-2/192303

tohojo, 2024-03-27:
Ah, thanks @jow- ! I will push a fix to the 23.05 branch :)

user8446, 2024-03-27:
I am on 23.05.3 installing acme for the 1st time. Even after the https://github.com/openwrt/packages/commit/d701cae191a4025292b15c6f908b823ad3b88985 fix it does not start:

`Failed to execute "/etc/init.d/acme enable" action: Command failed`

If I run `/usr/lib/acme/client/acme.sh` the script itself runs but of course isn't usable since nothing is configured.

tohojo, 2024-03-27:
Fixed with 73bee1f9b0e4dafddf45baea02cdf06e5a20dffe

closed 2024-03-27 by tohojo

stokito, 2024-04-19:
I forgot to say sorry fot this bug. I added the option in the VM when tested but when copied back to sources forgot the `option`.
But, I would say that the UCI should be more lenient for syntax errors and better to follow the [Postel Law](https://en.wikipedia.org/wiki/Robustness_principle). This is a good sample of a minor error on the embedded device that broke the world.
We should try to make the UCI more smart and not fail in such cases.

user8446, 2024-04-19:
No worries my friend, thank you for the development!

saudiqbal, 2024-04-20:
Using acme plugin in Openwrt does not replace the default certificate.