openwrt/packages #23761: nano: ucode syntax highlighting works only for template files

eric-j-ason, 2024-03-27:
Maintainer: @hnyman
Environment: master

Description:

The current syntax highlighting rules are applied to file extensions `uc` (scripts) and `ut` (templates):

https://github.com/openwrt/packages/blob/d8a574f7f0eb2f5970119a2b0527048583054180/utils/nano/files/ucode.nanorc#L3

However, the actual rule are correct only for template files, as they include things like template string expressions, and thus break the highlighting for scripts.

jow-, 2024-03-27:
Start the `.uc` script with `#!ucode` or `#!/usr/bin/env ucode` and it should work correct.

eric-j-ason, 2024-03-28:
Here is an example:

```
#!/usr/bin/env ucode

print("Hello, world!\n");
```
[image]

An example [in the wild](https://github.com/openwrt/openwrt/blob/d4a40827baf010763a1bf60a6435a5ca7da7ea62/package/network/config/wifi-scripts/files/usr/share/hostap/wdev.uc) looks like this:

[image]

hnyman, 2024-04-02:
> can't reproduce a problem from the given description. Can you provide an example file and a screenshot?

@jow- 
I can reproduce it with up-to-date master build: 

Of the 5 .uc files in `/usr/share/hostap/` three get syntax highlighted just fine, but two do not. The failing ones are those with shebang: `wdev.uc` and `wifi-detect.uc`
Removing (or commenting out) the shebang from the .uc file enables syntax highlighting. 
The shebang doesn't even need to be a full shebang, but just having "#!" there prevents highlighting.

```
root@router6000:/usr/share/nano# ls -1 /usr/share/hostap/*uc
/usr/share/hostap/common.uc
/usr/share/hostap/hostapd.uc
/usr/share/hostap/wdev.uc
/usr/share/hostap/wifi-detect.uc
/usr/share/hostap/wpa_supplicant.uc
```

Ok:
[image]

Failing file wdev,uc
[image]

Removed shebang:
[image]

Extra # in shebang to invalidate it:
[image]

The behaviour seems strange, as the filename based syntax should be the primary selection criteria and header just a fallback.

I didn't figure out what is wrong with the header def.

jow-, 2024-04-04:
This should fix it:

```diff
diff --git a/utils/nano/files/ucode.nanorc b/utils/nano/files/ucode.nanorc
index 96c7848e2..647f21d85 100644
--- a/utils/nano/files/ucode.nanorc
+++ b/utils/nano/files/ucode.nanorc
@@ -43,7 +43,7 @@ color ,green "[[:space:]]+$"
 
 # Text outside template directives
 color slate start="[}%#]\}" end="\{[{%#]"
-color slate start="^#!" end="\{[{%#]"
+color slate start="^#!.*(\<utpl\>|[[:space:]]-[[:alnum:]]*T[[:alnum:]]*\>)" end="\{[{%#]"
 color slate "^([^{%#}]|\{[^{%#]|[%#}][^}])+\{[{%#]"
 
 # Template tags
```

eric-j-ason, 2024-04-05:
> This should fix it:

A quick test indicates that this works. At least keywords and strings get highlighted.

[image]

PR #23820 (2024-04-05): nano: fix syntax highlighting for raw ucode scripts
Maintainer: @hnyman 
Compile tested: n/a
Run tested: n/a

Description:

Text between interpreter line and start of first directive should only highlighted as uninterpreted when running in template mode, so adjust the match rule accordingly.

Fixes: #23761
commit 8f956438 (2024-04-05): nano: fix syntax highlighting for raw ucode scripts

Text between interpreter line and start of first directive should only
highlighted as uninterpreted when running in template mode, so adjust
the match rule accordingly.

Fixes: #23761
commit d9d6eb1d (2024-04-05): nano: fix syntax highlighting for raw ucode scripts

Text between interpreter line and start of first directive should only
highlighted as uninterpreted when running in template mode, so adjust
the match rule accordingly.

Fixes: #23761

(cherry picked from commit 8f9564387d136c2a09c763b4c4ac7e4aa16baeb5)
commit 8c02297d (2024-04-05): nano: fix syntax highlighting for raw ucode scripts

Text between interpreter line and start of first directive should only
highlighted as uninterpreted when running in template mode, so adjust
the match rule accordingly.

Fixes: #23761

(cherry picked from commit 8f9564387d136c2a09c763b4c4ac7e4aa16baeb5)
closed 2024-04-05 by hnyman