openwrt/packages #23768: auc:

akoskoronka, 2024-03-27:
Package: auc

Maintainer: @dangowrt 

Environment:
- Architecture: mediatek/mt7622
- Model: linksys,e8450-ubi
- OpenWrt Version: 23.05.3

- Model: tplink,archer-c7-v5
- Architecture: ath79/generic
- OpenWrt Version: 23.05.2

Description:
Attended sysupgrade through luci works just fine, errors only appear on the cli.

Issue:
When running the AUC module via ssh, the following errors are encountered:
```
root@openwrt:~# auc
auc/0.3.2-1
Server: https://sysupgrade.openwrt.org/
Running: 23.05.3 r23809-234f1a2efa on mediatek/mt7622 (linksys,e8450-ubi)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```
The first error message is most likely caused by a bug in the `get_current_branch` function, where the `strncasecmp` comparison can lead to incorrect behavior if `br->name` is shorter than `version`.
The problem lies in the condition inside the if statement:
`if (!strncasecmp(br->name, version, strlen(br->name)))`
The strncasecmp function is comparing the branch name (br->name) with the version string using the length of br->name as the comparison limit. This could lead to incorrect behavior if br->name is shorter than version. To fix this, we should use the minimum length between br->name and version as the comparison limit, like this:
`if (!strncasecmp(br->name, version, strlen(version)))`

Also, there is a typo is in the error message "cannot determing currently running branch." where "determing" should be corrected to "determine." 

For the second error message, there is an existing open issue: "auc: Invalid argument (22)" #20377, however if I remember well there were some file problems earlier with the ASU, which were fixed.

Thank you.

peekpeekpeekpeek, 2024-03-27:
This is on TP-Link AC1750 Archer A7

auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ath79/generic (tplink,archer-a7-v5)
WARNING: cannot determing currently running branch.
Invalid argument (22)

And the same error on X86 system
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on x86/64 (generic)
WARNING: cannot determing currently running branch.
Invalid argument (22)

also notice the typo in "determing", should be "determine"

opty77, 2024-03-31:
/me joins the party:

```
root@tplink:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ath79/generic (tplink,tl-wdr3600-v1)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

opty77, 2024-04-01:
It changed to https://github.com/openwrt/asu/issues/780 this morning and works now so /me leaves the party.

akoskoronka, 2024-04-02:
It works for me as well, so closing this issue.

closed 2024-04-02 by akoskoronka

grandiego, 2024-07-09:
This is still an issue on my end
`root@ap:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.3 r23809-234f1a2efa on mediatek/filogic (cudy,wr3000-v1)
WARNING: cannot determing currently running branch.
Invalid argument (22)
`

mangkoran, 2024-07-09:
For anyone that still have issue with `auc`, this tool will be replaced with `owut` in the next stable release. It's intended to be `auc` replacement (not drop in but should cover all `auc` use cases) but also offer more features. You may find more detail in [the forum post](https://forum.openwrt.org/t/owut-openwrt-upgrade-tool). Please give it a try!

grandiego, 2024-07-09:
`owut` is not available on 23.05.3 repo. Will it be backported? If not, I guess I'll wait for next stable release. Thanks anyway