openwrt/packages #23771: [openwrt-23.05] ksmbd-server: Shared files/directories are not visible in root folder of the share

JKSTAFF, 2024-03-28:
Description:  
Same issue as #23088 . Happend on 23.05 with ksmbd v3.4.8 and kernel v5.15.150.  
To reproduce this, create user `admin` with `ksmbd.adduser`, mount the mmc storage, create a share setting to allow users or guests or both.
Actually there are file and folders there in the path which is mounted and shared. But I can't see or write in to.

[image]

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05-SNAPSHOT'
DISTRIB_REVISION='r23812-0a2047cf77'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05-SNAPSHOT r23812-0a2047cf77'
DISTRIB_TAINTS='no-all'
```

sppmasterspp, 2024-03-28:
@neheb 
Can you help here too?
Like you [helped for main branch](https://github.com/openwrt/packages/pull/23095).

pesa1234, 2024-03-29:
> Description: Same issue as #23088 . Happend on 23.05 with ksmbd v3.4.8 and kernel v5.15.150. To reproduce this, create `whateverusername` with `ksmbd.adduser`, mount the mmc storage, create a share setting to allow users or guests or both. Actually there are file and folders there in the path which is mounted and shared. But I can't see or write in to.
> 
> ![Screenshot_20240328_205755](https://private-user-images.githubusercontent.com/27397756/317734898-9813e08d-3463-4e42-9d1f-291a1722215c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE3MjAwODUsIm5iZiI6MTcxMTcxOTc4NSwicGF0aCI6Ii8yNzM5Nzc1Ni8zMTc3MzQ4OTgtOTgxM2UwOGQtMzQ2My00ZTQyLTlkMWYtMjkxYTE3MjIyMTVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI5VDEzNDMwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1ZTE3ZjFkOTNhMDFjYjVhZjE0YTI5YmViYTQyZWFhZjdjYTQ0YWE4Yjc1OThiM2E0YjI4ODRmNzk0MDBmNzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Rh-PrwCMYq9xEDAetXBR3e_hY5ltS4SXhWrnjUtotks)
> ## Additional Information:
> 
> ```
> DISTRIB_ID='OpenWrt'
> DISTRIB_RELEASE='23.05-SNAPSHOT'
> DISTRIB_REVISION='r23812-0a2047cf77'
> DISTRIB_TARGET='mediatek/filogic'
> DISTRIB_ARCH='aarch64_cortex-a53'
> DISTRIB_DESCRIPTION='OpenWrt 23.05-SNAPSHOT r23812-0a2047cf77'
> DISTRIB_TAINTS='no-all'
> ```

Hello, do you have same issue with latest snapshot and other operating system?
I'm on latest snapshot ksmbd-tools 3.5.1 with ubuntu 22.04, raspbian, win 10, win 11 and android phones and I didn't get this issue. If no issue with ksmbd-tools 3.5.1 I try to backport to 23.05. Thanks

JKSTAFF, 2024-03-30:
> do you have same issue with latest snapshot and other operating system?
Hello ,I make a snapshot on the same device with ksmbd 3.5.1 works fine. I saw you already update version for 23.05. I will try and report later.

JKSTAFF, 2024-04-02:
I've tried ksmbd 3.4.6-3.4.8 in the last two days in branch 23.05 and both have display issues. And when using the Solid-Explorer on android to access the SMB network storage, the router will get into a stuck that has to be unplugged to recover.  
In order to exclude interference from other packages, I did not add third-party software, mainly selected ksmbd, block-mount, btrfs and other necessary kmod and luci modules

pesa1234, 2024-04-02:
Please try to cherry-pick this commit https://github.com/openwrt/packages/pull/23785/commits/dc5fa137787a351d58f46c4724871554537c4efd and check with new version 3.5.1 is ok. let me know.

JKSTAFF, 2024-04-03:
problem fixed with 3.5.1

closed 2024-04-03 by JKSTAFF

pesa1234, 2024-04-03:
> problem fixed with 3.5.1

Thanks for feedback

Andrea