openwrt/packages #23774: p910nd: reset bedirectional option after printer hotlug/reboot

ivanmara555, 2024-03-28:
owrt 23.05.3.

problem: After printer hotplug/reboot, running script: /etc/hotplug.d/usbmisc/20-p910nd always rewrite ``` option bidirectional 'X' ```  in /etc/config/p910nd because /etc/hotplug.d/usbmisc/20-p910nd doesn't check for existence of bidirectional option and always rewrite it. Probably this is bug.

On my HP LaserJet 1200 printer (usb, need firmware on boot) and I need manually set ``` option bidirectional '0' ``` in /etc/config/p910nd because my printer incorrectly defined as bidirectional 1 (probably because of the firmware).

My workaround is change 285 string in /etc/hotplug.d/usbmisc/20-p910nd from:
``` eval "$uqsddu_cmd.bidirectional='$BIDIR'" ```
to
``` [ -z "$(eval "$uqgddu_cmd".bidirectional)" ] && eval "$uqsddu_cmd.bidirectional='$BIDIR'" ```

The second problem is:
In owrt 23.05.3 you need necessarily option  ``` runas_root '1' ``` otherwise p910nd does not start with log string:
```daemon.info procd: Instance p910nd::p910nd s in a crash loop 6 crashes, 0 seconds since last crash``` 
In owrt 23.05.2 p910nd process started with unprivileged p910nd user (``` runas_root '0' ```)

BKPepe, 2024-03-29:
Let's ping @systemcrash about this.

brada4, 2024-03-29:
Problem 1: hotplug script reads bidirectional status from port, disregarding (not modifying) config file. Probably you can write your own that starts/stops init script as printer is plugged in.
Problem 2: yes, the permissions of printer port are never changed to p910n user.

ivanmara555, 2024-03-29:
> Problem 1: hotplug script reads bidirectional status from port, disregarding (not modifying) config file. Probably you can write your own that starts/stops init script as printer is plugged in.

Sorry you are wrong. Hotplug script reads bidirectional status from port and always rewrite (modifying) bidirectional option in /etc/config/p910nd. Lets see code from: /etc/hotplug.d/usbmisc/20-p910nd (from owrt 23.05.3):

```
# Port directionality                               <-  read from port
BIP=$( cat "$ACTUAL_DEVPATH/bInterfaceProtocol" )
[ "$DEBUG" ] && echo BIP: "$BIP"
case $BIP in
	01 )
		BIDIR=0
		;;
	02 | 03 )
		BIDIR=1
		;;
esac
...
eval "$uqsddu_cmd.bidirectional='$BIDIR'"           <- rewrite bidirectional option to /etc/config/p910nd
```
But the user should be able to set the bidirectional option in config that he needs to be able to print. Therefore, before writing, you need to check for the presence of a bidirectional option like this:
```
[ -z "$(eval "$uqgddu_cmd".bidirectional)" ] && eval "$uqsddu_cmd.bidirectional='$BIDIR'"
```

systemcrash, 2024-03-29:
> Sorry you are wrong. 

😆 

I agree with the bidirectional suggestions. I can get those in. 

The root thing, however, I disagree. I've never had to run as root. You should not need to. 

re permissions.  The hotplug script does:

```sh
		[ -n "${DEVNAME}" ] && [ "${DEVNAME##usb/lp*}" = "" ] && {
			chmod 660 /dev/"$DEVNAME"
			chgrp lp /dev/"$DEVNAME"
		}
```

Maybe we should skip the check for it in the usb path.

Can you provide some more info about your printer please?

```sh
ls /sys/class/usbmisc/lp0/device/
cat /sys/class/usbmisc/lp0/device/ieee1284_id
cat /sys/class/usbmisc/lp0/device/bInterfaceClass
cat /sys/class/usbmisc/lp0/device/bInterfaceProtocol
cat /sys/class/usbmisc/lp0/device/bInterfaceSubClass
cat /sys/class/usbmisc/lp0/device/bNumEndpoints
find -L /dev -maxdepth 3 -type c -name lp[0-9]
```

ivanmara555, 2024-03-29:
> re permissions. The hotplug script does:

Yes hotplug script set right permissions:
```
root@router:/etc/config$ ls -al /dev/usb/lp0
crw-rw----    1 root     lp        180,   0 Mar 29 19:44 /dev/usb/lp0
```
But in owrt 23.05.3 service p910nd can not start with p910nd user. I need use `runas_root '1'`. 
In owrt 23.05.2 everything was ok and p910nd started with low privileged user like this:
```
23.05.2: ps | grep p910nd
1637 p910nd    1064 S    {p910nd} /usr/sbin/p9100d -d -f /dev/usb/lp0 0
```
I dont know why )

Additional information upon your request:
```
root@router:/etc/config$ ls /sys/class/usbmisc/lp0/device/
authorized            bInterfaceNumber      bNumEndpoints         ep_81                 of_node               uevent
bAlternateSetting     bInterfaceProtocol    driver                ieee1284_id           subsystem             usbmisc
bInterfaceClass       bInterfaceSubClass    ep_01                 modalias              supports_autosuspend

root@router:/etc/config$ cat /sys/class/usbmisc/lp0/device/ieee1284_id
MFG:Hewlett-Packard;MDL:HP LaserJet 1020;CMD:ACL;CLS:PRINTER;DES:HP LaserJet 1020;FWVER:20050309;

root@router:/etc/config$ cat /sys/class/usbmisc/lp0/device/bInterfaceClass
07

root@router:/etc/config$ cat /sys/class/usbmisc/lp0/device/bInterfaceProtocol
02

root@router:/etc/config$ cat /sys/class/usbmisc/lp0/device/bInterfaceSubClass
01

root@router:/etc/config$ cat /sys/class/usbmisc/lp0/device/bNumEndpoints
02

root@router:/etc/config$ find -L /dev -maxdepth 3 -type c -name lp[0-9]
/dev/usb/lp0

```

Ready to provide additional information.

systemcrash, 2024-03-29:
> Ready to provide additional information.

Try settings the device permissions not as root but as user 660 (which ideally should be in passwd files as p910nd). You can check other p910nd issues for more info on that.

ivanmara555, 2024-03-29:
> Try setting the device permissions not as root but as user 660 (which ideally should be in passwd files as p910nd). You can check other p910nd issues for more info on that.

Ok. 
```
Before:
service p910nd stop
root@router:/$ ls -al /dev/usb/lp0
crw-rw----    1 root     lp        180,   0 Mar 29 19:59 /dev/usb/lp0
root@router:/etc/config$ uci get p910nd.@p910nd[0].runas_root
1

after:
root@router:/etc/config$ chown p910nd:lp /dev/usb/lp0
root@router:/etc/config$ ls -al /dev/usb/lp0
crw-rw----    1 p910nd   lp        180,   0 Mar 29 19:59 /dev/usb/lp0

# set runas_root=0
uci set p910nd.@p910nd[0].runas_root=0
uci commit p910nd
luci-reload

service p910nd restart
ps | grep p910nd
NOTHING! p910nd does not work
```

After chown p910nd:lp /dev/usb/lp0 and runas_root=0 p910nd does not work with log string:
```
Fri Mar 29 21:05:22 2024 daemon.info procd: Instance p910nd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
```

problem still exists

systemcrash, 2024-03-29:
Ok. Thanks for testing.

systemcrash, 2024-03-31:
> After chown p910nd:lp /dev/usb/lp0 and runas_root=0 p910nd does not work with log string:
> 
> ```
> Fri Mar 29 21:05:22 2024 daemon.info procd: Instance p910nd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
> ```

Please install `strace` to your device and see what specifically turns up.

`strace p910nd ... (rest of usual command line)`.

This should help determine what and where it is going wrong in your case.

systemcrash, 2024-03-31:
> On my HP LaserJet 1200 printer (usb, need firmware on boot) and I need manually set `option bidirectional '0'` in /etc/config/p910nd because my printer incorrectly defined as bidirectional 1 (probably because of the firmware).

I also have a cousin of this printer, the 1018. Do you use a blob for this printer?

ivanmara555, 2024-03-31:
> I also have a cousin of this printer, the 1018. Do you use a blob for this printer?

Yes. I use blob from ubuntu: getweb 1020
Linux and windows clients printing fine.
My working config in 23.05.3:
```
config p910nd
	option device '/dev/usb/lp0'
	option bidirectional '0'
	option port '0'
	option enabled '1'
	option runas_root '1'
	option usbvidpid '03f0/2b17'
	option mdns_mfg 'Hewlett-Packard'
	option mdns_note 'Located near router'
	option mdns_sn 'JL07RZF'
	option mdns '0'
	option mdns_mdl 'HP LaserJet 1020'
	option driver_file '/opt/p910nd_drivers/Hewlett-Packard_HP_LaserJet_1020_03f0_2b17.bin'
	option mdns_cmd 'ACL'
	option mdns_cls 'PRINTER'
	option mdns_ty 'HP LaserJet 1020'
	option mdns_product '(HP LaserJet 1020)'

config hotplug
	option driver_home '/opt/p910nd_drivers'

```

> Please install `strace` to your device and see what specifically turns up.

Ok. strace test I'll do it soon.!

ivanmara555, 2024-03-31:
> Please install `strace` to your device and see what specifically turns up.

owrt 23.05.3

root case:
All ok. process start and working:
```
root@router:~# strace -u root /usr/sbin/p910nd -d -f /dev/usb/lp0 0

execve("/usr/sbin/p910nd", ["/usr/sbin/p910nd", "-d", "-f", "/dev/usb/lp0", "0"], 0x7f73fd28 /* 13 vars */) = 0
set_thread_area(0x77e70dfc)             = 0
set_tid_address(0x77e686ac)             = 13207
brk(NULL)                               = 0x1515000
brk(0x1517000)                          = 0x1517000
mmap2(0x1515000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x1515000
open("/etc/ld-musl-mipsel-sf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=131351, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 200704, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77d76000
mmap2(0x77da5000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1f000) = 0x77da5000
close(3)                                = 0
mprotect(0x77da5000, 4096, PROT_READ)   = 0
mprotect(0x41f000, 4096, PROT_READ)     = 0
open("/tmp/p9100d", O_RDWR|O_CREAT|O_LARGEFILE, 0666) = 3
getpid()                                = 13207
fcntl64(3, F_SETLKW64, {l_type=F_WRLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
socket(AF_INET, SOCK_DGRAM|SOCK_CLOEXEC, IPPROTO_UDP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(65535), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
getsockname(4, {sa_family=AF_INET, sin_port=htons(48277), sin_addr=inet_addr("127.0.0.1")}, [16]) = 0
close(4)                                = 0
socket(AF_INET6, SOCK_DGRAM|SOCK_CLOEXEC, IPPROTO_UDP) = 4
connect(4, {sa_family=AF_INET6, sin6_port=htons(65535), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "::", &sin6_addr), sin6_scope_id=0}, 28) = 0
getsockname(4, {sa_family=AF_INET6, sin6_port=htons(33934), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "::1", &sin6_addr), sin6_scope_id=0}, [28]) = 0
close(4)                                = 0
socket(AF_INET, SOCK_STREAM, IPPROTO_IP) = 4
setsockopt(4, SOL_SOCKET, SO_RCVBUF, [65536], 4) = 0
setsockopt(4, SOL_SOCKET, SO_SNDBUF, [65536], 4) = 0
setsockopt(4, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
bind(4, {sa_family=AF_INET, sin_port=htons(9100), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
listen(4, 5)                            = 0
accept(4, 

# another console:
root@router:~# ps | grep p910nd
13204 root      2116 S    strace -u root /usr/sbin/p910nd -d -f /dev/usb/lp0 0
13207 root      1064 S    {p910nd} /usr/sbin/p9100d -d -f /dev/usb/lp0 0
```

p910nd low privileged user case:
process does not start!:
```
root@router:~# strace -u p910nd /usr/sbin/p910nd -d -f /dev/usb/lp0 0
execve("/usr/sbin/p910nd", ["/usr/sbin/p910nd", "-d", "-f", "/dev/usb/lp0", "0"], 0x7f72da28 /* 13 vars */) = 0
set_thread_area(0x77ea8dfc)             = 0
set_tid_address(0x77ea06ac)             = 13275
brk(NULL)                               = 0xaf3000
brk(0xaf5000)                           = 0xaf5000
mmap2(0xaf3000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xaf3000
open("/etc/ld-musl-mipsel-sf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=131351, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 200704, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77dae000
mmap2(0x77ddd000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1f000) = 0x77ddd000
close(3)                                = 0
mprotect(0x77ddd000, 4096, PROT_READ)   = 0
mprotect(0x41f000, 4096, PROT_READ)     = 0
open("/tmp/p9100d", O_RDWR|O_CREAT|O_LARGEFILE, 0666) = -1 EACCES (Permission denied)
ioctl(1, TIOCGWINSZ, {ws_row=43, ws_col=158, ws_xpixel=0, ws_ypixel=0}) = 0
writev(1, [{iov_base="/tmp/p9100d: Permission denied", iov_len=30}, {iov_base="\n", iov_len=1}], 2/tmp/p9100d: Permission denied
) = 31
exit_group(1)                           = ?
+++ exited with 1 +++
root@router:~# 

```

any ideas ?

systemcrash, 2024-03-31:
At its simplest level, stop the privileged process, delete the /tmp/p910nd file. Then run p910nd unprivileged. It cannot start because it cannot write to that temp file.

Either: the file owned by root is still there; there's no space; something else weird.

ivanmara555, 2024-04-01:
> At its simplest level, stop the privileged process, delete the /tmp/p9100d file. Then run p910nd unprivileged. It cannot start because it cannot write to that temp file.
> 
> Either: the file owned by root is still there; there's no space; something else weird.

Yes! After manually deleting file /tmp/p9100d i could run p910nd with unprivileged user:

```
root@router:/tmp$ strace -u p910nd /usr/sbin/p910nd -d -f /dev/usb/lp0 0

strace: Exit of unknown pid 18135 ignored
execve("/usr/sbin/p910nd", ["/usr/sbin/p910nd", "-d", "-f", "/dev/usb/lp0", "0"], 0x7fcfba48 /* 16 vars */) = 0
set_thread_area(0x77ee7dfc)             = 0
set_tid_address(0x77edf6ac)             = 18136
brk(NULL)                               = 0x7e0000
brk(0x7e2000)                           = 0x7e2000
mmap2(0x7e0000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7e0000
open("/etc/ld-musl-mipsel-sf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=131351, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 200704, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77ded000
mmap2(0x77e1c000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1f000) = 0x77e1c000
close(3)                                = 0
mprotect(0x77e1c000, 4096, PROT_READ)   = 0
mprotect(0x41f000, 4096, PROT_READ)     = 0
open("/tmp/p9100d", O_RDWR|O_CREAT|O_LARGEFILE, 0666) = 3
getpid()                                = 18136
fcntl64(3, F_SETLKW64, {l_type=F_WRLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
socket(AF_INET, SOCK_DGRAM|SOCK_CLOEXEC, IPPROTO_UDP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(65535), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
getsockname(4, {sa_family=AF_INET, sin_port=htons(54569), sin_addr=inet_addr("127.0.0.1")}, [16]) = 0
close(4)                                = 0
socket(AF_INET6, SOCK_DGRAM|SOCK_CLOEXEC, IPPROTO_UDP) = 4
connect(4, {sa_family=AF_INET6, sin6_port=htons(65535), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "::", &sin6_addr), sin6_scope_id=0}, 28) = 0
getsockname(4, {sa_family=AF_INET6, sin6_port=htons(55514), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "::1", &sin6_addr), sin6_scope_id=0}, [28]) = 0
close(4)                                = 0
socket(AF_INET, SOCK_STREAM, IPPROTO_IP) = 4
setsockopt(4, SOL_SOCKET, SO_RCVBUF, [65536], 4) = 0
setsockopt(4, SOL_SOCKET, SO_SNDBUF, [65536], 4) = 0
setsockopt(4, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
bind(4, {sa_family=AF_INET, sin_port=htons(9100), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
listen(4, 5)                            = 0
accept(4, 

...

root@router:~# ps | grep p910nd
18133 root      2116 S    strace -u p910nd /usr/sbin/p910nd -d -f /dev/usb/lp0 0
18136 p910nd    1064 S    {p910nd} /usr/sbin/p9100d -d -f /dev/usb/lp0 0
```

there are a lot of resources:

```
root@router:~# free -h
              total        used        free      shared  buff/cache   available
Mem:         250316       69744       30688        6040      149884      121564
Swap:        262160           0      262160
root@router:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.8M      3.8M         0 100% /rom
tmpfs                   122.2M      5.9M    116.3M   5% /tmp
/dev/ubi0_1             179.4M     27.4M    147.3M  16% /overlay
overlayfs:/overlay      179.4M     27.4M    147.3M  16% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/sda1               458.1G    106.5G    328.3G  24% /mnt/sda1
```

I noticed that after the process is completed (or killing), the file /tmp/p9100d not deleting automatically. Apparently this is the problem.

closed 2024-04-26 by neheb

ivanmara555, 2024-07-20:
commit https://github.com/systemcrash/openwrt_packages/commit/73c026b4810ac036bce0139151b2e404cd0665dc (about + [ -z "$(eval "$uqgddu_cmd".bidirectional)" ] && eval "$uqsddu_cmd.bidirectional='$BIDIR'" ) does not appear in release OpenWrt 23.05.4 ! The problem still exists.

systemcrash, 2024-07-22:
It went into master. I don't think it got cherry picked to 23 branch yet.

systemcrash, 2024-07-23:
It's now in 22 and 23. package and new image builds will contain it
tomorrow.

commit 4628b6bd (2024-04-26): p910nd: set bidi only if not already set

Closes #23774
commit c2422e74 (2024-07-23): p910nd: set bidi only if not already set

Closes #23774

(cherry picked from commit 4628b6bd43ac1c212363535fe4d32739c5f7e622)
commit 028474f0 (2024-07-23): p910nd: set bidi only if not already set

Closes #23774

(cherry picked from commit 4628b6bd43ac1c212363535fe4d32739c5f7e622)