openwrt/packages #23778: Reopen #16548 -- NOT FIXED  grep: /usr/libexec/{egrep-gnu,fgrep-gnu} uses #!/bin/bash as interpreter

BlessDeix92, 2024-03-29:
Reopen #16548 -- NOT FIXED  grep: /usr/libexec/{egrep-gnu,fgrep-gnu} uses #!/bin/bash as interpreter

BKPepe, 2024-03-30:
If you want to reopen some issues, please provide as many details as possible. You removed the issue template, and we don't have the crystal ball to know what is going on, the steps for reproduction, which versions you are using, etc. The issue template is there to help you and help us to look into it.

BlessDeix92, 2024-03-30:
> If you want to reopen some issues, please provide as many details as possible.

Can you please be specific about what information is missing? I included the original bug and proof that the bug is not fixed. Please be specific about what your problem is. Were you not able to reproduce the issue? What steps did you take to show that my bug report is inadequate?

BKPepe, 2024-03-31:
I see that you are offended by me, I am not sure why, but I would like to help you since you have not provided details:
- Which version of grep do you have installed
- Which OpenWrt version are you using, which target/router do you have
- Have you reach the OpenWrt maintainer of grep to look into it?

From your comment here: https://github.com/openwrt/packages/issues/16548#issuecomment-2027785561, where you replied after almost 3 years, I am missing these details, and if you are not going to provide them, that's perfectly fine, but I would not be the one, who can look into it. 

Keep in mind that this project is voluntary based and we are doing in our free time.

BlessDeix92, 2024-04-01:
> [~/download]
> pleb@computer>wget https://downloads.openwrt.org/releases/23.05.3/packages/arm_cortex-a15_neon-vfpv4/packages/grep_3.8-2_arm_cortex-a15_neon-vfpv4.ipk
> --2024-04-01 00:06:34--  https://downloads.openwrt.org/releases/23.05.3/packages/arm_cortex-a15_neon-vfpv4/packages/grep_3.8-2_arm_cortex-a15_neon-vfpv4.ipk
> Resolving downloads.openwrt.org (downloads.openwrt.org)... 151.101.2.132, 151.101.66.132, 151.101.130.132, ...
> Connecting to downloads.openwrt.org (downloads.openwrt.org)|151.101.2.132|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 90576 (88K) [application/octet-stream]
> Saving to: ‘grep_3.8-2_arm_cortex-a15_neon-vfpv4.ipk’
> 
> grep_3.8-2_arm_cortex-a15_neon-vfpv4.ipk   100%[=======================================================================================>]  88.45K  --.-KB/s    in 0.02s   
> 
> 2024-04-01 00:06:35 (3.61 MB/s) - ‘grep_3.8-2_arm_cortex-a15_neon-vfpv4.ipk’ saved [90576/90576]
> 
> [~/download]
> pleb@computer>tar -vxf grep_3.8-2_arm_cortex-a15_neon-vfpv4.ipk 
> ./debian-binary
> ./data.tar.gz
> ./control.tar.gz
> [~/download]
> pleb@computer>tar -vxf data.tar.gz 
> ./
> ./usr/
> ./usr/libexec/
> ./usr/libexec/egrep-gnu
> ./usr/libexec/fgrep-gnu
> ./usr/libexec/grep-gnu
> [~/download]
> pleb@computer>cat usr/libexec/egrep-gnu 
> #!/bin/bash
> cmd=${0##*/}
> echo "$cmd: warning: $cmd is obsolescent; using grep -E" >&2
> exec grep -E "$@"
> 
> 
>

closed 2024-04-01 by BlessDeix92