openwrt/packages #23780: vnstat2: vnstatd can not be stopped

BlessDeix92, 2024-03-30:
OpenWrt version

r23809
OpenWrt release

23.05.3

vnstat2, and vnstat v1 as well, have no functionality to tell procd to stop their service.

"/etc/init.d/vnstat disable" does nothing
"/etc/init.d/vnstat stop" does nothing

vnstatd just keeps on running and being restarted by procd.

There's no way to stop these daemons if, say, you are troubleshooting, or any other number of good reasons why you might want to stop it.

brada4, 2024-03-30:
> someone slapped 

helps to tag maintainer ipo blaming the world

pesa1234, 2024-03-31:
No message at all?
If you enable and then start or restart? What do you get example on `logread -f | grep vnsta`

BlessDeix92, 2024-04-01:
There is literally no stop function for this service. It is impossible to make this brain-dead obvious issue more apparent.

https://openwrt.org/docs/guide-developer/procd-init-scripts#stopping_services

https://github.com/openwrt/packages/blob/openwrt-22.03/net/vnstat2/files/vnstat.init

There is no "stop_service()" func in the init script.

BlessDeix92, 2024-04-01:
My bad. I filed this before, but nobody gives a fuck.

https://github.com/openwrt/packages/issues/14547

closed 2024-04-01 by BlessDeix92

pesa1234, 2024-04-01:
/etc/init.d/vnstat stop --> daemon.info vnstatd[29444]: Info: SIGTERM received, exiting.

/etc/init.d/vnstat disable --> removed from /etc/rc.d/vnstat

So, I didn't see your error