openwrt/packages #23781: vnstat2: Importing DBs from vnstat1 fails

BlessDeix92, 2024-03-30:
OpenWrt version

r23809
OpenWrt release

23.05.3

I tried to upgrade an openwrt system from vnstat v1 to v2. Unfortunately, none of the database could be upgraded, so vnstat loses all of the historical data.

I am not the only one with this problem: https://forum.openwrt.org/t/solved-vnstat-2-import-old-database/148981

However, I have no idea what the above person was doing, since those import/export arguments don't exist.

When running "./vnstatd -D -n" manually, I get the following errors:

```
        Importing data from legacy database "eth0.2".
        db: Database read failed for file "/srv/vnstat/eth0.2".
        Error: Database load failed even when using backup (No such file or directory). Aborting.

```
There is nothing wrong with the input DB files. They work fine on a v1 system. There is no permission or other strange problem here.

pesa1234, 2024-03-31:
https://github.com/vergoh/vnstat/wiki/Usage-%23-Migrating-data

BlessDeix92, 2024-04-01:
My bad. Nobody gives a fuck.

closed 2024-04-01 by BlessDeix92

vergoh, 2024-04-01:
I can try to fill at least some of the blanks:

- The import/export being done in https://forum.openwrt.org/t/solved-vnstat-2-import-old-database/148981 is for renaming the interface and it doesn't have much to do with the database migration itself as versions 2.x don't care if the interfaces are available or not during import
- The reason why the discussion linked above mentions https://github.com/vergoh/vnstat/wiki/Usage-%23-Migrating-data but acts as the "Migrating data from vnStat 1.x versions to 2.x versions" section isn't there is because that section was written later to better document the conditions for the data migration to happen
- "db: Database read failed for file" error occurs when the v1 file can be opened but it's structure doesn't fit or match the expected v1 database structure. Either this is a case of a really old v1 version being used (the exact vnStat version from which the database file came from isn't mentioned) or it's caused by the "The host CPU architecture is the same for both versions" condition not being met. This information is also missing from the report so I can't guess which one it is
- "Error: Database load failed even when using backup (No such file or directory)" indicates that the version upgrade isn't likely being done in the same system as the backup file would otherwise be there

In summary, the missing details are:

- which vnStat 1.x version has been used to create the v1 files?
- which vnStat 2.x version is being used for the data migration?
- is the system being upgraded in place or is the data also being migrated for one device to another?
- if a device to device migration is being done, what are the CPU architecture of both devices?