openwrt/packages #23782: vnstat: No version of vnstat can use db files from a previous build

BlessDeix92, 2024-03-30:
I tested OpenWRT release versions 18, 19, 21, 22, and 23, and every version fails to correctly import and re-use existing vnstat database files from the previous version.

I've known this for years. This isn't a new issue. It's possible it's never ever worked.  I have my own script which I run before an upgrade to export the files to txt format and then import them after the upgrade. However, now that I want to migrate to vnstat2, it's impossible because the --exportdb and --importdb arguments have been removed, and vnstat2 also fails to import the binary files from any previous vnstat1 version.

So, just to be clear, in addition to vnstat v1 not being able to read databases from any previous build, v2 can't read previous v1 dbs either.

Every vnstat db file seems to be build-unique and can't be recognized from any other.

Running "./vnstatd -D -n" does not give much info, except that it's broken

```
        Importing data from legacy database "eth0.2".
        db: Database read failed for file "/srv/vnstat/eth0.2".
        Error: Database load failed even when using backup (No such file or directory). Aborting.

```

Google results prove that db corruption problems are extremely common with vnstat. This is probably why it's author migrated to sqlite as a back-end.

BlessDeix92, 2024-03-30:
This is an ipq806x device, ea8500.

brada4, 2024-03-30:
Upstream documentation details actions to take:
https://github.com/vergoh/vnstat/wiki/Usage-%23-Migrating-data

brada4, 2024-03-30:
Please close duplicates.

BlessDeix92, 2024-03-30:
> Please close duplicates.

Which duplicates are you referring to?

BlessDeix92, 2024-03-30:
> Upstream documentation details actions to take: https://github.com/vergoh/vnstat/wiki/Usage-%23-Migrating-data

This does not address my issue. The documentation you referenced simply says "it will happen automatically". My report proves it does not. Further, the upstream documentation supports much of the information that I have provided.

brada4, 2024-03-30:
#23781 ?
#23870  not backed by any logs or confs either.

BlessDeix92, 2024-03-30:
Your attitude is abusive. Look into these issues and if you find I have made a mistake in my reporting, then we can talk 
about it.

I went through a lot of trouble to report these bugs. Why should I bother if I'm going to get shit on by toxic behavior?

If you don't want to fix bugs, just don't reply. Don't do any work. That's fine. It's your code base, but don't fucking abuse me with false accusations of not providing enough info, filing duplicate reports, or other lazy dumb shit. You are just wasting my time with boiler-plate copy-paste time-wasting crap.

Tell me how much work you've put into any of these bugs to demonstrate that I'm wrong.

> #23781 ?

What is your question? Question mark means nothing. You need to put some effort into typing. I put a lot of effort into reporting these bugs. You can be bothered by typing whole words and sentences.

> #23780 not backed by any logs or confs either.

None are needed. The code does not support stopping the service.  The init script simply does not support stopping at all. It's like someone slapped together some copy/paste from another service but forgot half of it.

brada4, 2024-03-30:
I dont get errors you describe, you need to write up on how to reproduce them. Really simple - gather eth0 to /mnt/sda for 1h, uninstall, install other, it imports database.

BlessDeix92, 2024-04-01:
My bad. Nobody gives a fuck.

https://github.com/openwrt/packages/issues/14547

closed 2024-04-01 by BlessDeix92