openwrt/packages #23786: micropython: cannot compile with glibc because of "__P" redefined [-Werror]

titobrasolin, 2024-03-31:
Package: micropython
Maintainer: @jefferyto 
OpenWrt Version: 23.05.3

Steps to reproduce:

    git clone https://git.openwrt.org/openwrt/openwrt.git
    cd openwrt
    git checkout v23.05.3
    ./scripts/feeds update -a
    ./scripts/feeds install -a
    make menuconfig
    # Use glibc, enable micropython, exit.
    make -j$(nproc) defconfig download clean world

Error Log:
```
[...]
In file included from ../../lib/berkeley-db-1.xx/PORT/include/db.h:40,
                 from ../../extmod/modbtree.c:35:
../../lib/berkeley-db-1.xx/PORT/include/cdefs.h:58: error: "__P" redefined [-Werror]
   58 | #define __P(protos)     protos          /* full-blown ANSI C */
      | 
In file included from /root/openwrt/staging_dir/toolchain-mips_24kc_gcc-12.3.0_glibc/include/features.h:491,
                 from /root/openwrt/staging_dir/toolchain-mips_24kc_gcc-12.3.0_glibc/include/bits/libc-header-start.h:33,
                 from /root/openwrt/staging_dir/toolchain-mips_24kc_gcc-12.3.0_glibc/include/stdint.h:26,
                 from /root/openwrt/staging_dir/toolchain-mips_24kc_gcc-12.3.0_glibc/lib/gcc/mips-openwrt-linux-gnu/12.3.0/include/stdint.h:9,
                 from ../../py/mpstate.h:29,
                 from ../../py/runtime.h:29,
                 from ../../extmod/modbtree.c:27:
/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-12.3.0_glibc/include/sys/cdefs.h:118: note: this is the location of the previous definition
  118 | #define __P(args)       args
      | 
cc1: all warnings being treated as errors
[...]
```

Workaround:
Disable all warnings with "-w" (additional compiler options)

Neustradamus, 2024-10-27:
@Ansuel, @gstrauss: What do you think about this ticket?

gstrauss, 2024-10-27:
@Neustradamus please do not tag me.  Ever.  I have no interest in working with you again based on my *many* prior interactions with you and your continued failure to learn and implement the basics of github.  *For me,* you are too noisy and disruptive and I plan to filter you out of my email.

Neustradamus, 2024-10-27:
gstrauss: Thanks for your answer.
You have done a lot of improvements, me and other people thank you.
I have already done my thanks previously.

Thanks to have said that you can not work on this ticket, hope that other guy can look it.