openwrt/packages #23794: lighttpd: missing dependency libcrypto.so.3

rodajc, 2024-04-01:
lighttpd: 1.4.74
Maintainer: @gstrauss
Environment: rockchip armv8, OpenWrt v23.05.3

Description:

Clean new compilation of v23.05.3 reports a missing dependency for lighttpd: libcrypto.so.3

```
make[3]: Entering directory '/home/user/openwrt/feeds/packages/net/lighttpd'
mkdir -p /home/user/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/user/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/.pkgdir/lighttpd/. /home/user/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/user/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.lighttpd_installed
mkdir -p /home/user/openwrt/bin/targets/rockchip/armv8/packages /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/CONTROL /home/user/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo
install -d -m0755 /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/etc/lighttpd
install -m0644 ./files/lighttpd.conf /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/etc/lighttpd/
install -m0644 /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/doc/config/conf.d/mime.conf /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/etc/lighttpd/
install -d -m0755 /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/etc/lighttpd/conf.d
install -d -m0755 /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/etc/init.d
install -m0755 ./files/lighttpd.init /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/etc/init.d/lighttpd
install -d -m0755 /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/usr/lib/lighttpd
for m in dirlisting h2; do cp -fpR /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-install/usr/lib/lighttpd/mod_${m}.so /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/usr/lib/lighttpd/ ; done
install -d -m0755 /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/usr/sbin
install -m0755 /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-install/usr/sbin/lighttpd /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd/usr/sbin/
find /home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.74/ipkg-aarch64_generic/lighttpd -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package lighttpd is missing dependencies for the following libraries:
libcrypto.so.3
make[3]: *** [Makefile:255: /home/user/openwrt/bin/packages/aarch64_generic/packages/lighttpd_1.4.74-1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/user/openwrt/feeds/packages/net/lighttpd'
time: package/feeds/packages/lighttpd/compile#0.57#0.38#1.33
    ERROR: package/feeds/packages/lighttpd failed to build.
make[2]: *** [package/Makefile:129: package/feeds/packages/lighttpd/compile] Error 1
make[2]: Leaving directory '/home/user/openwrt'
make[1]: *** [package/Makefile:123: /home/user/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:232: world] Error 2

```

gstrauss, 2024-04-01:
How have you configured lighttpd?  What selections were made in `make menuconfig`?

Latest release of lighttpd is lighttpd 1.4.75.  Please use lighttpd 1.4.75 instead of lighttpd 1.4.74.

rodajc, 2024-04-01:
lighttpd config:

```
CONFIG_LIGHTTPD_CRYPTOLIB_MBEDTLS=y
CONFIG_LIGHTTPD_PCRE2=y
CONFIG_LIGHTTPD_SSL=y
CONFIG_LIGHTTPD_SSL_SELECT=m
CONFIG_PACKAGE_lighttpd=y
CONFIG_PACKAGE_lighttpd-mod-accesslog=y
CONFIG_PACKAGE_lighttpd-mod-cgi=y
CONFIG_PACKAGE_lighttpd-mod-openssl=y
CONFIG_PACKAGE_lighttpd-mod-redirect=y
CONFIG_PACKAGE_luci-light=y

```

Compiling 1.4.75 gives:

```
make[2]: Entering directory '/home/user/openwrt/feeds/packages/net/lighttpd'
mkdir -p /home/user/openwrt/dl
SHELL= flock /home/user/openwrt/tmp/.lighttpd-1.4.75.tar.xz.flock -c '  	/home/user/openwrt/scripts/download.pl "/home/user/openwrt/dl" "lighttpd-1.4.75.tar.xz" "5c08736e83088f7e019797159f306e88ec729abe976dc98fb3bed71b9d3e53b5" "" "https://download.lighttpd.net/lighttpd/releases-1.4.x"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://download.lighttpd.net/lighttpd/releases-1.4.x/lighttpd-1.4.75.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1076k  100 1076k    0     0   864k      0  0:00:01  0:00:01 --:--:--  865k
Hash of the downloaded file does not match (file: 8b721ca939d312afaa6ef31dcbd6afb5161ed385ac828e6fccd4c5b76be189d6, requested: 5c08736e83088f7e019797159f306e88ec729abe976dc98fb3bed71b9d3e53b5) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/lighttpd-1.4.75.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1076k  100 1076k    0     0   656k      0  0:00:01  0:00:01 --:--:--  656k
Hash of the downloaded file does not match (file: 8b721ca939d312afaa6ef31dcbd6afb5161ed385ac828e6fccd4c5b76be189d6, requested: 5c08736e83088f7e019797159f306e88ec729abe976dc98fb3bed71b9d3e53b5) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/lighttpd-1.4.75.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1076k  100 1076k    0     0   534k      0  0:00:02  0:00:02 --:--:--  535k
Hash of the downloaded file does not match (file: 8b721ca939d312afaa6ef31dcbd6afb5161ed385ac828e6fccd4c5b76be189d6, requested: 5c08736e83088f7e019797159f306e88ec729abe976dc98fb3bed71b9d3e53b5) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/lighttpd-1.4.75.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
make[2]: *** [Makefile:239: /home/user/openwrt/dl/lighttpd-1.4.75.tar.xz] Error 2
make[2]: Leaving directory '/home/user/openwrt/feeds/packages/net/lighttpd'
time: package/feeds/packages/lighttpd/compile#0.84#0.29#6.01
    ERROR: package/feeds/packages/lighttpd failed to build.
make[1]: *** [package/Makefile:129: package/feeds/packages/lighttpd/compile] Error 1
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:232: package/lighttpd/compile] Error 2

```

brada4, 2024-04-01:
Just pull updates and sync feeds.
Mirror checksum is correct in repo https://github.com/openwrt/packages/blob/876b0bc1945e578106569760d9c138b6dee30643/net/lighttpd/Makefile#L18

rodajc, 2024-04-01:
After:

`./scripts/feeds update -a`

lighttpd 1.4.74 at least starts compiling but then hits the same dependency: libcrypto.so.3

rodajc, 2024-04-01:
Both lighttpd 1.4.74 and 1.4.75 compile fine after selecting the lighttpd-mod-mbedtls package.
This makes CONFIG_LIGHTTPD_SSL_SELECT=m change automatically into CONFIG_LIGHTTPD_SSL_DEPENDS=y.

Shouldn't menuconfig select lighttpd-mod-mbedtls automatically if you select libmbedtls as crypto library?

gstrauss, 2024-04-02:
> Both lighttpd 1.4.74 and 1.4.75 compile fine after selecting the lighttpd-mod-mbedtls package.
This makes CONFIG_LIGHTTPD_SSL_SELECT=m change automatically into CONFIG_LIGHTTPD_SSL_DEPENDS=y.

So if this issue resolved for you?

> Shouldn't menuconfig select lighttpd-mod-mbedtls automatically if you select libmbedtls as crypto library?

lighttpd Makefile config in openwrt defaults to use nettle for crypto functions (SHA-1, SHA-256, etc)  in the lighttpd core and non-TLS modules such as lighttpd mod_auth.  If you select libmbedtls as crypto library in in menuconfig for lighttpd, then lighttpd will compile the base lighttpd modules against mbedtls crypto library instead of nettle.  lighttpd mod_mbedtls is a separate lighttpd module, and it depends on mbedtls TLS library, mbedtls x509 library, and mbedtls crypto library.

rodajc, 2024-04-02:
Interesting. After changing the crypto library from mbedtls to nettle lighttpd compiles successfully.

The problem is that if you select mbedtls as crypto library, you need to select lighttpd-mod-mbedtls as well.

If you don't you end up with the missing dependency libcrypto.so.3 error.

ynezz, 2024-04-24:
> Package lighttpd is missing dependencies for the following libraries:
> libcrypto.so.3

I've hit the same issue as I would like to make OpenSSL only.

> lighttpd Makefile config in openwrt defaults to use nettle for crypto functions (SHA-1, SHA-256, etc) 

Ok, but it seems, that lighttpd itself supports OpenSSL as well, thus I'm wondering if there is any reason to not support this library additionally? I've probed that option in https://github.com/openwrt/packages/pull/24004

gstrauss, 2024-04-27:
Sorry, I got side-tracked.  Still fighting with Ubuntu and Debian to pretend like they know how to update software.

I'll try to review this some time next week.

gstrauss, 2024-05-03:
More detailed comment in #24004 regarding extending support to force the lighttpd base application to use openssl for crypto functions (such as SHA-256).

@rodajc 

> Shouldn't menuconfig select lighttpd-mod-mbedtls automatically if you select libmbedtls as crypto library?

> Interesting. After changing the crypto library from mbedtls to nettle lighttpd compiles successfully.
> 
> The problem is that if you select mbedtls as crypto library, you need to select lighttpd-mod-mbedtls as well.
> 
> If you don't you end up with the missing dependency libcrypto.so.3 error.

Yes, the openwrt lighttpd Makefile could do a better job here, independent of #24004

gstrauss, 2024-05-06:
This might address the issue raised by @rodajc.  Would someone else help confirm?
```
--- a/net/lighttpd/Makefile
+++ b/net/lighttpd/Makefile
@@ -133,9 +133,11 @@ if PACKAGE_lighttpd
 
                config LIGHTTPD_CRYPTOLIB_MBEDTLS
                        bool "libmbedtls"
+                       select PACKAGE_lighttpd-mod-mbedtls
 
                config LIGHTTPD_CRYPTOLIB_WOLFSSL
                        bool "libwolfssl"
+                       select PACKAGE_lighttpd-mod-wolfssl
        endchoice
 endif
 
```
@ynezz's PR #24004 is also an improvement and is being handled separately.

Another question: on OpenWRT, should the lighttpd Makefile default to using mbedtls as the crypto lib instead of Nettle?  If the default dependency of hostapd is mbedtls, then I think it makes sense for that to be the default for lighttpd, too.

gstrauss, 2024-05-12:
@rodajc any change to test the patch above?  I'm going to leave out this patch for now until I am unable to get feedback.

gstrauss, 2024-05-12:
Actually, @rodajc this patch seems better to me and follows the same pattern currently used for `CONFIG_LIGHTTPD_CRYPTOLIB_NETTLE`
```
--- a/net/lighttpd/Makefile
+++ b/net/lighttpd/Makefile
@@ -168,25 +168,25 @@ MESON_ARGS += \
        -Dwith_bzip=disabled \
        -Dwith_dbi=$(if $(CONFIG_PACKAGE_lighttpd-mod-authn_dbi)$(CONFIG_PACKAGE_lighttpd-mod-vhostdb_dbi),enabled,disabled) \
        -Dwith_fam=disabled \
-       -Dwith_gnutls=$(if $(CONFIG_PACKAGE_lighttpd-mod-gnutls),true,false) \
+       -Dwith_gnutls=$(if $(CONFIG_PACKAGE_lighttpd-mod-gnutls)$(CONFIG_LIGHTTPD_CRYPTOLIB_GNUTLS),true,false) \
        -Dwith_krb5=$(if $(CONFIG_PACKAGE_lighttpd-mod-authn_gssapi),enabled,disabled) \
        -Dwith_ldap=$(if $(CONFIG_PACKAGE_lighttpd-mod-authn_ldap)$(CONFIG_PACKAGE_lighttpd-mod-vhostdb_ldap),enabled,disabled) \
        -Dwith_libunwind=disabled \
        -Dwith_lua=$(if $(CONFIG_PACKAGE_lighttpd-mod-magnet),true,false) \
        -Dlua_version=lua \
        -Dwith_maxminddb=$(if $(CONFIG_PACKAGE_lighttpd-mod-maxminddb),enabled,disabled) \
-       -Dwith_mbedtls=$(if $(CONFIG_PACKAGE_lighttpd-mod-mbedtls),true,false) \
+       -Dwith_mbedtls=$(if $(CONFIG_PACKAGE_lighttpd-mod-mbedtls)$(CONFIG_LIGHTTPD_CRYPTOLIB_MBEDTLS),true,false) \
        -Dwith_mysql=$(if $(CONFIG_PACKAGE_lighttpd-mod-vhostdb_mysql),enabled,disabled) \
        -Dwith_nettle=$(if $(CONFIG_LIGHTTPD_CRYPTOLIB_NETTLE),true,false) \
        -Dwith_nss=$(if $(CONFIG_PACKAGE_lighttpd-mod-nss),true,false) \
-       -Dwith_openssl=$(if $(CONFIG_PACKAGE_lighttpd-mod-openssl),true,false) \
+       -Dwith_openssl=$(if $(CONFIG_PACKAGE_lighttpd-mod-openssl)$(CONFIG_LIGHTTPD_CRYPTOLIB_OPENSSL),true,false) \
        -Dwith_pam=$(if $(CONFIG_PACKAGE_lighttpd-mod-authn_pam),enabled,disabled) \
        -Dwith_pcre2=$(if $(CONFIG_LIGHTTPD_PCRE2),true,false) \
        -Dwith_pgsql=$(if $(CONFIG_PACKAGE_lighttpd-mod-vhostdb_pgsql),enabled,disabled) \
        -Dwith_sasl=$(if $(CONFIG_PACKAGE_lighttpd-mod-authn_sasl),enabled,disabled) \
        -Dwith_webdav_locks=$(if $(CONFIG_PACKAGE_lighttpd-mod-webdav),enabled,disabled) \
        -Dwith_webdav_props=$(if $(CONFIG_PACKAGE_lighttpd-mod-webdav),enabled,disabled) \
-       -Dwith_wolfssl=$(if $(CONFIG_PACKAGE_lighttpd-mod-wolfssl),true,false) \
+       -Dwith_wolfssl=$(if $(CONFIG_PACKAGE_lighttpd-mod-wolfssl)$(CONFIG_LIGHTTPD_CRYPTOLIB_WOLFSSL),true,false) \
        -Dwith_xattr=false \
        -Dwith_zlib=$(if $(CONFIG_PACKAGE_lighttpd-mod-deflate),enabled,disabled) \
        -Dwith_zstd=disabled
```

rodajc, 2024-05-12:
@gstrauss 
Where do I create the patch to [Makefile](https://github.com/openwrt/packages/issues/23794#issuecomment-2095038219)? In openwrt/build_dir? There I find lighttpd-1.4.75/src/Makefile.am and lighttpd-1.4.75/src/Makefile.in. Which do I need to patch?

gstrauss, 2024-05-12:
`openwrt/feeds/packages/net/lighttpd/Makefile`

rodajc, 2024-05-12:
```
user@laptop:~/openwrt$ make package/lighttpd/update V=s
Collecting package info: done
make[2]: Entering directory '/home/user/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/user/openwrt/scripts/config'
make[1]: Entering directory '/home/user/openwrt'
make[2]: Entering directory '/home/user/openwrt/feeds/packages/net/lighttpd'
if [ -s "/home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.75/patches/series" ]; then (cd "/home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.75"; if quilt --quiltrc=- next >/dev/null 2>&1; then quilt --quiltrc=- push -a; else quilt --quiltrc=- top >/dev/null 2>&1; fi ); fi
touch "/home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.75/.quilt_checked"
mkdir -p ./patches
rm -f ./patches/* 2>/dev/null >/dev/null
'/home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.75/patches/020-meson-mod_webdav_min.patch' -> './patches/020-meson-mod_webdav_min.patch'
cp: cannot stat '/home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.75/patches/030-missing_dep_libcrypto.patch': No such file or directory
make[2]: *** [Makefile:242: update] Error 1
make[2]: Leaving directory '/home/user/openwrt/feeds/packages/net/lighttpd'
time: package/feeds/packages/lighttpd/update#0.38#0.18#0.65
    ERROR: package/feeds/packages/lighttpd failed to build.
make[1]: *** [package/Makefile:129: package/feeds/packages/lighttpd/update] Error 1
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:232: package/lighttpd/update] Error 2

```

gstrauss, 2024-05-12:
`cp: cannot stat '/home/user/openwrt/build_dir/target-aarch64_generic_musl/lighttpd-1.4.75/patches/030-missing_dep_libcrypto.patch': No such file or directory`

Maybe a mistyped path in your patches/series file?  Do the filenames match?  You might remove `lighttpd/patches/series`

rodajc, 2024-05-12:
Yes. There was a reference to a patch I didn't finish in lighttpd/patches/series.

But now:

`bash: line 1: /home/sil/openwrt/staging_dir/host/bin/python3: No such file or directory`

gstrauss, 2024-05-13:
`bash: line 1: /home/sil/openwrt/staging_dir/host/bin/python3: No such file or directory`

There are numerous starting points to set up an openwrt build.

Please check the documentation (https://openwrt.org/docs/guide-developer/start)

I am guessing: perhaps you have not recently updated your working copy or rebuilt the `tools/install` or `toolchain/install` make targets.  However, further discussion of this is off-topic for this issue, so please see the documentation linked above.

closed 2024-05-13 by neheb

rodajc, 2024-05-15:
> Actually, @rodajc this patch seems better to me and follows the same pattern currently used for `CONFIG_LIGHTTPD_CRYPTOLIB_NETTLE`
>
> ```
> --- a/net/lighttpd/Makefile
> +++ b/net/lighttpd/Makefile
> @@ [...]

This patch solves the issue.

commit a300185d (2024-05-13): lighttpd: fix missing dependency for OpenSSL crypto library

This change will provide the necessary dependency resolution, fixing:

  Package lighttpd is missing dependencies for the following libraries:
  libcrypto.so.3

Fixes: #23794
commit 4eb73df5 (2024-06-20): lighttpd: fix missing dependency for OpenSSL crypto library

This change will provide the necessary dependency resolution, fixing:

  Package lighttpd is missing dependencies for the following libraries:
  libcrypto.so.3

Fixes: #23794

(cherry picked from commit a300185d49e87848f9b8027393196631841b32f6)
commit f514663e (2024-06-20): lighttpd: fix missing dependency for OpenSSL crypto library

This change will provide the necessary dependency resolution, fixing:

  Package lighttpd is missing dependencies for the following libraries:
  libcrypto.so.3

Fixes: #23794

(cherry picked from commit a300185d49e87848f9b8027393196631841b32f6)