openwrt/packages #23796: auc: `-n` doesn't work

opty77, 2024-04-01:
Maintainer: @dangowrt
Environment: MIPS 74Kc, TP-Link TL-WDR3600 v1.2, 23.05.2

Description:

`-n` doesn't work:

```
root@tplink:~# auc -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ath79/generic (tplink,tl-wdr3600-v1)
Available: 23.05.3 r23809-234f1a2efa
Requesting package lists...
 openssh-sftp-server: 9.5p1-1 -> 9.6p1-1
[...]
 kmod-cfg80211: 5.15.137+6.1.24-3 -> 5.15.150+6.1.24-3
Are you sure you want to continue the upgrade process? [N/y] N
root@tplink:~#
```

dangowrt, 2024-04-01:
> `Are you sure you want to continue the upgrade process? [N/y] N`

You have to answer `Y` if you want anything to happen, even just the download. True, it's bit misleading, might adjust the confirmation prompt to be different in case of `-n` was given on cmdline.

opty77, 2024-04-01:
Are you saying that I didn't have to press Ctrl+C several times in panic?

```
[...]
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build...^C..^C.^C^CSegmentation fault
root@tplink:~# ^C

root@tplink:~#
```

`logread`:

```
Mon Apr  1 21:21:39 2024 kern.info kernel: [11656272.630686] do_page_fault(): sending SIGSEGV to auc for invalid read access from 779cd034
Mon Apr  1 21:21:39 2024 kern.info kernel: [11656272.639674] epc = 0040303f in auc[400000+6000]
Mon Apr  1 21:21:39 2024 kern.info kernel: [11656272.644581] ra  = 004030a5 in auc[400000+6000]
```

dangowrt, 2024-04-01:
No, with `-n` given you can proceed. And yes, this segfault needs to be dealt with and is due to inappropriate use of libuclient... This whole thing should be re-written in ucode once we got support for a sane way to emmit HTTP requests from ucode.

opty77, 2024-04-02:
I'm going to try before the actual upgrade (not enough faith now) but looking at the broader context of the `dry_run` code (shame on me), it should really work.

Thank you very much for explanations!

closed 2024-04-02 by opty77

opty77, 2024-05-05:
@dangowrt: Just FYI, works:

```
[...]
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build.............................................................
Image available at https://sysupgrade.openwrt.org/store/xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx/openwrt-23.05.3-xxxxxxxxxxxx-ath79-generic-tplink_tl-wdr3600-v1-squashfs-sysupgrade.bin
done
```

vs. actual upgrade

```
[...]
Are you sure you want to continue the upgrade process? [N/y] y

Downloading image from https://sysupgrade.openwrt.org/store/xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx/openwrt-23.05.3-xxxxxxxxxxxx-ath79-generic-tplink_tl-wdr3600-v1-squashfs-sysupgrade.bin
Writing to 'openwrt-23.05.3-xxxxxxxxxxxx-ath79-generic-tplink_tl-wdr3600-v1-squashfs-sysupgrade.bin'
image verification succeeded
invoking sysupgrade
Connection to 192.168.1.1 closed by remote host.
Connection to 192.168.1.1 closed.
```

BTW, I redacted the hashes but I'm not sure if I had to.