openwrt/packages #23801: openwrt version 23.05.3 Raspberry pi 4 banip version 0.9.4.-3 conflicting with dnscryptproxy version 2019-08-20-07ac3825-3 cause cannot surf the web

mrusli70, 2024-04-02:
I am currently using Raspberry pi 4, install openwrt version 23.05.3. And having issues when i enable the banip (version 0.9.4-3) when the dnscryptproxy (version 2019-08-20-07ac3825-3).  That i can no longer surf the web. 

I have try to uninstall banip. But it does not seems to work at all. I have to start from scratch from the ground up. 

There are other packages that i install namely bcp38, squid. So i have no idea what so ever that when i enable banip with dnscryptproxy. The openwrt cannot seems to get to the internet.

I did the same configuration with the older version of openwrt. which is the previous version 23.05.2 and that one seems to work without any problems.

brada4, 2024-04-02:
Can you resolve via banip?
```
nslookup openwrt.org. 127.0.0.1:5353
```

mrusli70, 2024-04-03:
Nope! Once I enable banip the entire network is not connecting to the internet. "No internet" in the web browser. 

The only way is that i have to format and reinstall openwrt 23.05.3 again and i do not install banip because it is still conflicting with dnscryptproxy.

brada4, 2024-04-03:
dnscryptproxy uses quadnine by default, you need to find banip list that blocks it.

brada4, 2024-04-04:
it is non-default be+aviour configured by you, check linked ip4 list for quad9 servers. I.e no issue.
https://openwrt.org/docs/guide-user/services/banip#blocking_doh

dibdot, 2024-04-07:
Most probably a config error, please ask in the forum for support ... and provide your banIP config of course.

closed 2024-04-07 by dibdot