openwrt/packages #23843: auc: cannot determine currently running branch

trenchcoatjedi, 2024-04-07:
```
# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on x86/64 (generic)
WARNING: cannot determing currently running branch.
Invalid argument (22)

# auc -f -b 23.05 -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on x86/64 (generic)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

Trying to perform attendedsysupgrade through the LUCI app also fails similarly.

patman-cp, 2024-04-20:
I'm new to openwrt and was about to try my first upgrade, and was trying `auc -c` first.

Will this issue prevent an upgrade from working?

patman-cp, 2024-04-20:
OK I could not apparently update using auc, so I used luci.

I still see the same behavior with 23.05.3:

```
root@OpenWrt:~# auc -c
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.3 r23809-234f1a2efa on bcm27xx/bcm2711 (raspberrypi,4-model-b)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

benhedrington, 2024-04-20:
Seeing same behavior. x86/64

```
root@openwrt:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on x86/64 (generic)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

svilenstefanov, 2024-09-22:
Same error for me 

Opened auc issue https://github.com/openwrt/asu/issues/800

`auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on octeon/generic (ubnt,usg)
Invalid argument (22)`