openwrt/packages #23844: gpsd: update request

3ricj, 2024-04-07:
gpsd openwrt version is 3.23.  The newer versions include an important feature: TCP reconnect.  This allows tcp based gps devices to automatically reconnect if there are problems. The current version in openwrt just hangs and stops responding if the upstream gps disconnects.  

Please consider updating this package to 3.25.  

Maintainer: @Pushpal Sidhu

brada4, 2024-04-07:
just update with opkg?
https://github.com/openwrt/packages/blob/8602f63b8db7efd93ad83f8f36e5a0d2a6e1d56a/utils/gpsd/Makefile#L11

feckert, 2024-04-08:
@3ricj Pull requests are welcome.
However, I will close this issue. as it is not an issue!
It is rather a reminder to the maintainer to update the package.

closed 2024-04-08 by feckert

brada4, 2024-04-08:
@feckert well, the requested package has been for a while on mirrors ;-)

3ricj, 2024-04-08:
Ah, thanks folks.  I was using a 3rd party repo which didn't have the updated packages (Gl.inet).  Loading the main openwrt package worked great.  Thanks so much!