openwrt/packages #23846: libgd: libgd-full is built without JPEG support

M95D, 2024-04-07:
Maintainer: @jow-
Environment: mvebu / Turris Omnia / OpenWrt r25618

Description:
A php script that I tried to run wrote this in the log:
```PHP Warning:  imagecreatefromstring() [...]: No JPEG support in this PHP build in [...]```

More details:
```
# php8-cli -r "var_dump(gd_info());"
array(15) {
  ["GD Version"]=>
  string(5) "2.3.3"
  ["FreeType Support"]=>
  bool(true)
  ["FreeType Linkage"]=>
  string(13) "with freetype"
  ["GIF Read Support"]=>
  bool(true)
  ["GIF Create Support"]=>
  bool(true)
  ["JPEG Support"]=>
  bool(false)
  ["PNG Support"]=>
  bool(false)
  ["WBMP Support"]=>
  bool(true)
  ["XPM Support"]=>
  bool(false)
  ["XBM Support"]=>
  bool(true)
  ["WebP Support"]=>
  bool(false)
  ["BMP Support"]=>
  bool(false)
  ["AVIF Support"]=>
  bool(false)
  ["TGA Read Support"]=>
  bool(false)
  ["JIS-mapped Japanese Font Support"]=>
  bool(false)
}
```
I tried to remove php8-mod-gd and libgd-full that I built and install the official packages from the OpenWrt snapshots. Same problem.

Thanks.

mhei, 2024-04-17:
`php8-mod-gd` already pulls in `libgd-full`. The smaller version (`libgd`) should already have JPEG support enabled, also PNG support. The `libgd-full` variant enables additionally TIFF and Freetype support.
I tested it quickly on a Raspberry Pi installation and got the same result. However, running `ldd /usr/lib/php8/gd.so` shows that the module is linked against libpng and libjpeg.
To be honest, I'm a little bit confused...

M95D, 2024-05-23:
It seems that the bug is in php8-mod-gd, not libgd.

BMP vs. JPEG functions/symbols in gd.so:
```
root@GRAPHRPi3:/# objdump -T /usr/lib/php8/gd.so | grep -i bmp
0000000000000000      DF *UND*  0000000000000000  Base        gdImageCreateFromWBMP
0000000000000000      DF *UND*  0000000000000000  Base        gdImageWBMPCtx
0000000000000000      DF *UND*  0000000000000000  Base        gdImageCreateFromBmpCtx
0000000000000000      DF *UND*  0000000000000000  Base        gdImageCreateFromWBMPCtx
0000000000000000      D  *UND*  0000000000000000  Base        php_sig_bmp

root@GRAPHRPi3:/# objdump -T /usr/lib/php8/gd.so | grep -i jp
0000000000000000      D  *UND*  0000000000000000  Base        php_sig_jpg
```
BMP vs. JPEG in libgd.so:
```
root@GRAPHRPi3:/# objdump -T /usr/lib/libgd.so | grep -i bmp
0000000000023b54 g    DF .text  0000000000000054  Base        gdImageCreateFromWBMPPtr
0000000000018940 g    DF .text  0000000000000090  Base        gdImageBmpPtr
000000000001de84 g    DF .text  0000000000000074  Base        gdImageWBMP
000000000002388c g    DF .text  0000000000000278  Base        gdImageCreateFromWBMPCtx
0000000000018b74 g    DF .text  0000000000000054  Base        gdImageCreateFromBmpPtr
00000000000189d0 g    DF .text  0000000000000004  Base        gdImageBmpCtx
00000000000172c8 g    DF .text  0000000000000adc  Base        gdImageCreateFromBmpCtx
0000000000018a58 g    DF .text  0000000000000050  Base        gdImageCreateFromBmp
000000000001def8 g    DF .text  0000000000000090  Base        gdImageWBMPPtr
00000000000189d4 g    DF .text  0000000000000074  Base        gdImageBmp
0000000000023b04 g    DF .text  0000000000000050  Base        gdImageCreateFromWBMP
000000000001de80 g    DF .text  0000000000000004  Base        gdImageWBMPCtx

root@GRAPHRPi3:/# objdump -T /usr/lib/libgd.so | grep -i jp
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_set_defaults
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_read_header
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_set_quality
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_finish_compress
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_resync_to_restart
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_CreateDecompress
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_write_marker
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_destroy_compress
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_simple_progression
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_save_markers
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_CreateCompress
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_start_decompress
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_finish_decompress
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_write_scanlines
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_read_scanlines
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_destroy_decompress
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_std_error
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_start_compress
0000000000000000      DF *UND*  0000000000000000  <corrupt>   jpeg_destroy
000000000001d02c g    DF .text  0000000000000074  Base        gdImageJpeg
000000000001e964 g    DF .text  0000000000000058  Base        gdImageCreateFromJpegEx
000000000001e9bc g    DF .text  0000000000000010  Base        gdImageCreateFromJpeg
000000000001e354 g    DF .text  0000000000000610  Base        gdImageCreateFromJpegCtxEx
000000000001ea28 g    DF .text  0000000000000010  Base        gdImageCreateFromJpegPtr
000000000001cf98 g    DF .text  0000000000000090  Base        gdImageJpegPtr
000000000001e9cc g    DF .text  000000000000005c  Base        gdImageCreateFromJpegPtrEx
000000000001ea38 g    DF .text  0000000000000010  Base        gdImageCreateFromJpegCtx
000000000001d028 g    DF .text  0000000000000004  Base        gdImageJpegCtx
```
So, the functions are there, PHP just didn't link to them.
Any ideeas why?

M95D, 2024-05-23:
I looked at OpenWrt configure log. It's configured without jpeg:
```
checking for GD support... yes, shared
checking for external libgd... yes
checking for libavif... no
checking for libwebp... no
checking for libjpeg... no
checking for libXpm... no
checking for FreeType 2... no
checking whether to enable JIS-mapped Japanese font support in GD... no
checking for gdlib >= 2.1.0... yes
checking for working gdImageCreateFromPng in libgd... no
checking for working gdImageCreateFromAvif in libgd... no
checking for working gdImageCreateFromWebp in libgd... no
checking for working gdImageCreateFromJpeg in libgd... no
checking for working gdImageCreateFromXpm in libgd... no
checking for working gdImageCreateFromBmp in libgd... no
checking for working gdImageCreateFromTga in libgd... no
checking for gdFontCacheShutdown in -lgd... yes
checking for gdVersionString in -lgd... yes
checking for gdImageGetInterpolationMethod in -lgd... yes
checking for gdImageCreate in -lgd... yes
```

Then I looked at config.log in build_dir. The configure script doesn't seem to run any actual gcc tests for jpeg support:
```
configure:39983: checking for working gdImageCreateFromJpeg in libgd
configure:39994: result: no
```

Then I looked at configure script itself. I don't think this will ever work.
```
  { printf "%s\n" "$as_me:${as_lineno-$LINENO}: checking for working gdImageCreateFromJpeg in libgd" >&5
printf %s "checking for working gdImageCreateFromJpeg in libgd... " >&6; }
[...]
  if test "$cross_compiling" = yes
then :
    { printf "%s\n" "$as_me:${as_lineno-$LINENO}: result: no" >&5
printf "%s\n" "no" >&6; }
```

M95D, 2024-11-16:
Fixed by #24367

closed 2024-11-16 by M95D