openwrt/packages #23851: atlas-sw-probe: remains hung due to too early start since 23.05?

lukastribus, 2024-04-08:
Maintainer: @ja-pa @BKPepe 
Environment: x86_64 / PC Engines APU2 / OpenWrt 23.05.3

Hello,

my atlas (atlas-sw-probe) probe currently remains "hung" after rebooting:

Atlas starts before the WAN link is up, and remains in a hung state, the probe appears to be online/connected at RIPE, however it does not fulfil any measurements at all (which makes this particularly hard to notice).

Restarting Atlas when the WAN link is already up works around this issue, however when rebooting next time, the issue appears again.

It's possible this issue is happening since upgrading from 22.03 to 23.05, however I do not have 100% certainty on that.

Example:

    Mon Apr  8 12:00:47 2024 daemon.err ATLAS[2221]: evping: bad address 'yyy.xxxxxxxxxxxxxx.sos.atlas.ripe.net'
    Mon Apr  8 12:00:47 2024 daemon.err ATLAS[2221]: evping_ops.init failed
    [...]
    Mon Apr  8 12:00:59 2024 daemon.notice netifd: Interface 'wan' is now up
    Mon Apr  8 12:00:59 2024 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
    [...]
    Mon Apr  8 12:03:48 2024 daemon.err ATLAS[2221]: condmv: not moving, destination '/usr/libexec/atlas-probe-scripts/data/out/v6addr.txt' exists
    Mon Apr  8 12:03:48 2024 daemon.err ATLAS[2221]: condmv: not moving, destination '/usr/libexec/atlas-probe-scripts/data/out/simpleping' exists
    [...]
    Mon Apr  8 12:06:51 2024 daemon.err ATLAS[2221]: condmv: not moving, destination '/usr/libexec/atlas-probe-scripts/data/out/v6addr.txt' exists
    Mon Apr  8 12:06:51 2024 daemon.err ATLAS[2221]: condmv: not moving, destination '/usr/libexec/atlas-probe-scripts/data/out/simpleping' exists

The condmv errors reappear every 3 minutes as above.

After 1 hour uptime I can see a eperd error as below:

    Mon Apr  8 13:00:42 2024 local6.err eperd[2425]: root: No such file or directory

The condmv errors keep reappearing every 3 minutes.

Is atlas supposed to automatically be restarted on a WAN up event?

I noticed that the busybox binary dates back to Jan 15th despite this being 23.05.03, but I assume this is a non problem?

    root@up-rt1:/usr/libexec/atlas-probe/bin# ls -lha
    drwxr-xr-x    2 root     root        1.0K Mar 26 18:28 .
    drwxr-xr-x    5 root     root        1.0K Mar 26 18:28 ..
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 atlasinit -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 buddyinfo -> busybox
    -rwxr-xr-x    1 root     root      417.8K Jan 15  2023 busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 condmv -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 date -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 dfrm -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 eooqd -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 eperd -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 evhttpget -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 evntp -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 evping -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 evsslgetcert -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 evtdig -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 evtraceroute -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 httppost -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 onlyuptime -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 perd -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 rchoose -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 rptaddrs -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 rptra6 -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 rptuptime -> busybox
    lrwxr-xr-x    1 root     root           7 Mar 26 18:28 rxtxrpt -> busybox
    root@up-rt1:/usr/libexec/atlas-probe/bin#

Thank you,

Lukas

brada4, 2024-04-08:
Try START=99 in init script then re-enable service, it is meant to be ran on configured network.

lukastribus, 2024-04-08:
network has 20, atlas has 30. Bumping atlas from 30 to 99 did not solve this issue.

Network will be configured before starting atlas, but that doesn't mean that the WAN link will be fully initialized, up and running.

I tried reproducing the problem on a full VM (Oracle EL8 - similar to CentOS8) and the problem does not happen there. Simulated by dropping all outgoing packets with iptables (except my ssh session).

I retract the statement that it may have worked in 22.03, I think I never actually ran atlas on 22.03.

I'm working around this for now by triggering the init.d script on ifup instead of startup:

    /etc/init.d/atlas disable
    
    cat << "EOF" > /etc/hotplug.d/iface/99-atlas-on-demand
    [ "${ACTION}" = "ifup" ] && [ "${DEVICE}" = "pppoe-wan" ] && {
        logger -t hotplug "Device: ${DEVICE} / Action: ${ACTION} starting atlas from 99-atlas-on-demand"
        /etc/init.d/atlas start
    }
    [ "${ACTION}" = "ifdown" ] && [ "${DEVICE}" = "pppoe-wan" ] && {
        logger -t hotplug "Device: ${DEVICE} / Action: ${ACTION} stopping atlas from 99-atlas-on-demand"
        /etc/init.d/atlas stop
    } 
    EOF

Ansuel, 2024-04-28:
@lukastribus we should really investigate this... afaik our script lack support for reboot_probe function... Does your VM use the same scripts used by openwrt or use the VM one? (that have different implementation?)

lukastribus, 2024-04-29:
On the VM it's handled by systemd, so it's quite different.

fcucino, 2025-05-17:
Hi,
I am experiencing the same issue, with similar logs.

I have a dump access point with lan static IPv4 address and dynamic IPv6 assignment from the router.
- Firmware Version: OpenWrt 24.10.1 r28597-0425664679
- Target Platform: `ipq40xx/generic`
- Kernel Version: 6.6.86
- atlas-probe `v.2.6.3-r2` package
- atlas-sw-probe `v.5080-r2` package 

Until a manual restart of the service, the probe appears connected but no measurements are performed.
Putting START=99 in the init script seems not to solve the issue.

```
root@ap-cucina:/usr/libexec/atlas-probe/bin# ls -lha
drwxr-xr-x    2 root     root        1.6K Apr 23 22:52 .
drwxr-xr-x    5 root     root         352 Apr 23 22:52 ..
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 atlasinit -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 buddyinfo -> busybox
-rwxr-xr-x    1 root     root      431.9K Apr 20 14:50 busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 condmv -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 date -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 dfrm -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 eooqd -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 eperd -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 evhttpget -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 evntp -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 evping -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 evsslgetcert -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 evtdig -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 evtraceroute -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 httppost -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 onlyuptime -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 perd -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 rchoose -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 rptaddrs -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 rptra6 -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 rptuptime -> busybox
lrwxrwxrwx    1 root     root           7 Apr 23 22:52 rxtxrpt -> busybox
root@ap-cucina:/usr/libexec/atlas-probe/bin# ./busybox
BusyBox v1.26.2 (2025-04-22 20:45:40 UTC) multi-call binary.
BusyBox is copyrighted by many authors between 1998-2015.
Licensed under GPLv2. See source distribution for detailed
copyright notices.

Usage: busybox [function [arguments]...]
   or: busybox --list[-full]
   or: busybox --install [-s] [DIR]
   or: function [arguments]...

Currently defined functions:
        atlasinit, buddyinfo, condmv, date, dfrm, eooqd, eperd, evhttpget,
        evntp, evping, evsslgetcert, evtdig, evtraceroute, httppost,
        onlyuptime, perd, rchoose, rptaddrs, rptra6, rptuptime, rxtxrpt,
        telnetd
```