openwrt/packages #23852: cloudflared: compiled without support for post quantum curves.

alex9434, 2024-04-08:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @1715173329 @stokito 
Environment: arm64 (Raspberry Pi 4B), snapshot

Description:
Cloudflared Version 2024.3.0 does not connect with quic only with http2 on OpenWrt

To Reproduce
Steps to reproduce the behavior:

1. Build Cloudflared on OpenWrt using the makefile from Github https://github.com/openwrt/packages/blob/master/net/cloudflared/Makefile
2. Run service cloudflared start
3. See error log

```
{"level":"info","tunnelID":"*deleted*","time":"2024-04-05T15:16:57Z","message":"Starting tunnel"}
{"level":"info","time":"2024-04-05T15:16:57Z","message":"Version 2024.3.0"}
{"level":"info","time":"2024-04-05T15:16:57Z","message":"GOOS: linux, GOVersion: go1.21.6, GoArch: arm64"}
{"level":"info","time":"2024-04-05T15:16:57Z","message":"Settings: map[config:/etc/cloudflared/config.yml cred-file:/etc/cloudflared/<Tunnel-UUID>.json credentials-file:/etc/cloudflared/<Tunnel-UUID>.json logfile:/var/log/cloudflared.log loglevel:info no-autoupdate:true origincert:/etc/cloudflared/cert.pem token:***** url:http://localhost:8000]"}
{"level":"info","time":"2024-04-05T15:16:57Z","message":"Generated Connector ID: deleted"}
{"level":"info","time":"2024-04-05T15:16:57Z","message":"Initial protocol quic"}
{"level":"info","time":"2024-04-05T15:16:58Z","message":"ICMP proxy will use *deleted* as source for IPv4"}
{"level":"info","time":"2024-04-05T15:16:58Z","message":"ICMP proxy will use *deleted* in zone br-lan as source for IPv6"}
{"level":"warn","error":"Group ID 0 is not between ping group 1 to 0","time":"2024-04-05T15:16:58Z","message":"The user running cloudflared process has a GID (group ID) that is not within ping_group_range. You might need to add that user to a group within that range, or instead update the range to encompass a group the user is already in by modifying /proc/sys/net/ipv4/ping_group_range. Otherwise cloudflared will not be able to ping this network"}
{"level":"warn","error":"cannot create ICMPv4 proxy: Group ID 0 is not between ping group 1 to 0 nor ICMPv6 proxy: socket: permission denied","time":"2024-04-05T15:16:58Z","message":"ICMP proxy feature is disabled"}
{"level":"info","time":"2024-04-05T15:16:58Z","message":"Starting metrics server on 127.0.0.1:44091/metrics"}
{"level":"error","event":0,"ip":"198.41.200.13","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:16:58Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.200.13","connIndex":0,"time":"2024-04-05T15:16:58Z","message":"Retrying connection in up to 2s"}
{"level":"error","event":0,"ip":"198.41.200.33","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:16:58Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.200.33","connIndex":0,"time":"2024-04-05T15:16:58Z","message":"Retrying connection in up to 4s"}
{"level":"error","event":0,"ip":"198.41.200.43","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:17:00Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.200.43","connIndex":0,"time":"2024-04-05T15:17:00Z","message":"Retrying connection in up to 8s"}
{"level":"error","event":0,"ip":"198.41.200.23","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:17:03Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.200.23","connIndex":0,"time":"2024-04-05T15:17:03Z","message":"Retrying connection in up to 16s"}
{"level":"error","event":0,"ip":"198.41.200.43","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:17:03Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.200.43","connIndex":0,"time":"2024-04-05T15:17:03Z","message":"Retrying connection in up to 32s"}
{"level":"error","event":0,"ip":"198.41.200.193","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:17:17Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.200.193","connIndex":0,"time":"2024-04-05T15:17:17Z","message":"Retrying connection in up to 1m4s"}
{"level":"error","event":0,"ip":"198.41.200.13","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:17:27Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.200.13","connIndex":0,"time":"2024-04-05T15:17:27Z","message":"Retrying connection in up to 1m4s"}
{"level":"error","event":0,"ip":"198.41.192.57","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:17:59Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.192.57","connIndex":0,"time":"2024-04-05T15:17:59Z","message":"Retrying connection in up to 1m4s"}
{"level":"error","event":0,"ip":"198.41.192.107","connIndex":0,"error":"failed to dial to edge with quic: INTERNAL_ERROR (local): tls: CurvePreferences includes unsupported curve","time":"2024-04-05T15:18:28Z","message":"Failed to create new quic connection"}
{"level":"info","event":0,"ip":"198.41.192.107","connIndex":0,"time":"2024-04-05T15:18:28Z","message":"Retrying connection in up to 1m4s"}
{"level":"info","event":0,"ip":"198.41.192.107","connIndex":0,"time":"2024-04-05T15:18:54Z","message":"Switching to fallback protocol http2"}
{"level":"info","event":0,"connection":"91edb2ac-5c20-4b0e-855a-023ab41f2a57","connIndex":0,"location":"ham01","ip":"198.41.192.67","protocol":"http2","time":"2024-04-05T15:18:55Z","message":"Registered tunnel connection"}
{"level":"info","event":0,"connection":"7b87644a-6f63-4776-b240-002be7e7e629","connIndex":1,"location":"ams01","ip":"198.41.200.233","protocol":"http2","time":"2024-04-05T15:18:55Z","message":"Registered tunnel connection"}
{"level":"info","event":0,"connection":"a9aec0e3-0373-4538-b4fb-2f13d29ae78a","connIndex":2,"location":"ham01","ip":"198.41.192.227","protocol":"http2","time":"2024-04-05T15:18:56Z","message":"Registered tunnel connection"}
{"level":"info","event":0,"connection":"47d3d61b-201b-4cb4-a140-b8ab39acf35b","connIndex":3,"location":"ams06","ip":"198.41.200.43","protocol":"http2","time":"2024-04-05T15:18:57Z","message":"Registered tunnel connection"}
{"level":"info","version":11,"config":"{\"ingress\":[{\"hostname\":\"*deleted*\", \"originRequest\":{}, \"service\":\"http://localhost:80\"}, {\"hostname\":\"*deleted*\", \"originRequest\":{}, \"service\":\"ssh://localhost:22\"}, {\"service\":\"http_status:404\"}], \"warp-routing\":{\"enabled\":false}}","time":"2024-04-05T15:18:58Z","message":"Updated to new configuration"}
```

Issue https://github.com/cloudflare/cloudflared/issues/1158 mentioned in that this  because occurs because cloudflared needs to be built with a go tool chain that supports post quantum curves. 

Please checkout https://github.com/cloudflare/cloudflared?tab=readme-ov-file#installing-cloudflared for how to install the go tool chain and build from source. Is it correct to assume that the Makefile needs to be adapted accordingly?

stokito, 2024-04-08:
Duplicate of #23596

alex9434, 2024-04-11:
Hi @stokito, apologies that I did not catch that. From reading https://github.com/openwrt/packages/issues/23596 I got the impression that you are not planning to use the modified toolchain. Therefore I suggest that you add a warning to the package description that "This package does not use the required toolchain and therefore misses core parts of the functionality with have significant performance impact." Maybe you should add that it is recommended to use the official docker image or alternative builds instead whenever possible because they will offer full functionality and better performance. I think a lot of users are not aware of this issue and should be given that information.

stokito, 2024-04-11:
Added to Wiki https://openwrt.org/docs/guide-user/services/vpn/cloudfare_tunnel#one_final_note

closed 2024-04-14 by alex9434

Tokisaki-Galaxy, 2024-05-18:
Thank you for your question. I came across your problem`CurvePreferences includes unsupported curve`, and finally solved the protocol from QUIC to HTTP2 through your proposal.
I suggest that you can write only use HTTP2 protocol in wiki, because QUIC protocol is used by default and cannot be switched automatically.

PunkFleet, 2024-12-23:
Is this problem solved? I'm installing cloudflared on my NAS, and setting up a VM as openwrt at the same time, and I'm trying to point both the NAS gateway and DNS to the openwrt address, but I'm running into the cloudflare error!
```
20024-12-23 04:45:38.253644+00:002024-12-23T04:45:38Z INF Starting tunnel tunnelID=d652c760-3ccf-45fd-b637-8be5cbf34702
2024-12-23 04:45:38.253683+00:002024-12-23T04:45:38Z INF Version 2024.12.2 (Checksum cb61fcb41380efff4e2ef0a1eeea23bcddff360e8703f99dbb5c24da1de937b9)
2024-12-23 04:45:38.253689+00:002024-12-23T04:45:38Z INF GOOS: linux, GOVersion: go1.22.5-devel-cf, GoArch: amd64
2024-12-23 04:45:38.253691+00:002024-12-23T04:45:38Z INF Settings: map[no-autoupdate:true]
2024-12-23 04:45:38.253696+00:002024-12-23T04:45:38Z INF Environmental variables map[TUNNEL_TOKEN:*****]
2024-12-23 04:45:38.253838+00:002024-12-23T04:45:38Z INF Generated Connector ID: cd8ce6e8-42f4-47ef-a353-165ef2189970
2024-12-23 04:45:39.045169+00:002024-12-23T04:45:39Z INF Initial protocol quic
2024-12-23 04:45:39.059692+00:002024-12-23T04:45:39Z INF ICMP proxy will use 172.16.5.2 as source for IPv4
2024-12-23 04:45:39.059853+00:002024-12-23T04:45:39Z INF ICMP proxy will use ::1 in zone lo as source for IPv6
2024-12-23 04:45:39.654816+00:002024-12-23T04:45:39Z INF ICMP proxy will use 172.16.5.2 as source for IPv4
2024-12-23 04:45:39.655078+00:002024-12-23T04:45:39Z INF ICMP proxy will use ::1 in zone lo as source for IPv6
2024-12-23 04:45:39.655223+00:002024-12-23T04:45:39Z INF Starting metrics server on 127.0.0.1:20241/metrics
2024-12-23 04:45:39.660388+00:002024/12/23 04:45:39 failed to sufficiently increase receive buffer size (was: 208 kiB, wanted: 7168 kiB, got: 416 kiB). See https://github.com/quic-go/quic-go/wiki/UDP-Buffer-Sizes for details.
2024-12-23 04:45:44.663102+00:002024-12-23T04:45:44Z ERR Failed to dial a quic connection error="failed to dial to edge with quic: timeout: no recent network activity" connIndex=0 event=0 ip=198.41.192.107
2024-12-23 04:45:44.663253+00:002024-12-23T04:45:44Z INF Retrying connection in up to 2s connIndex=0 event=0 ip=198.41.192.107
2024-12-23 04:45:50.933009+00:002024-12-23T04:45:50Z ERR Failed to dial a quic connection error="failed to dial to edge with quic: timeout: no recent network activity" connIndex=0 event=0 ip=198.41.192.37
2024-12-23 04:45:50.933142+00:002024-12-23T04:45:50Z INF Retrying connection in up to 4s connIndex=0 event=0 ip=198.41.192.37
2024-12-23 04:45:56.094671+00:002024-12-23T04:45:56Z ERR Failed to dial a quic connection error="failed to dial to edge with quic: timeout: no recent network activity" connIndex=0 event=0 ip=198.41.200.193
```

alex9434, 2024-12-30:
@PunkFleet, I am sorry but my impression from the interaction was that the maintainers have no intention to fix the issue. I would advise you to use the docker version instead. I personally believe this package could be excluded from OpenWrt because the functionality is severely limited.

PunkFleet, 2024-12-30:
> @PunkFleet, I am sorry but my impression from the interaction was that the maintainers have no intention to fix the issue. I would advise you to use the docker version instead. I personally believe this package could be excluded from OpenWrt because the functionality is severely limited.

I'm using cloudflared on nas, which should be the docker version.

> See the https://openwrt.org/docs/guide-user/services/vpn/cloudfare_tunnel#one_final_note mentioned above

Yes, even with the docker version of cloudflared, you run into the same problem.

So I ended up getting the same problem by adding to openwrt's dnsmasq the `/*. argotunnel.com/1.1.1.1` to the forwarding

alex9434, 2024-12-30:
> @alex9434 on which facts is based your impression? The package works and updated regularly and almost immoderately, it's well documented (so your problem is mentioned on wiki), everybody happy.

As already stated on my previous comment, "From reading https://github.com/openwrt/packages/issues/23596 I got the impression that you are not planning to use the modified toolchain as required by the authors of that tool." Is that interpretation correct? 

Do you have any plans to use the required toolchain to enable the full functionality of the tool? As stated on the other issue, the current version is not only slower but also "disables UDP and ICMP support. This means if you run tunnels over TLS only, only TCP packets would be allowed - all other packets directed to the tunnel will be dropped."

This makes (at least) me unhappy with the provided remaining functionality. I therefore chose to use the docker version which supports QUIC.

Does this address your question?