openwrt/packages #23857: add Xorg packages

esaaprillia, 2024-04-09:
is xorg acceptable here i want to make a pull request for the Xorg package

brada4, 2024-04-09:
Is there emerging generation of wifi routers with telescreens in sight?

esaaprillia, 2024-04-09:
I don't play OpenWRT with a router but I use x86 OpenWRT. If it's not accepted I don't do a pull request. thanks

feckert, 2024-04-10:
@esaaprillia Thanks for your request.
The main problem with your request is, that the **focus** using OpenWrt is on routers.
The most of the package and OpenWrt maintainers do not use X.org on there targets devices.
The X.Org packages must be maintained!

If at all, then it makes sense that this is maintained in a separate feed repository.
Just like it is already done with the [routing feed](https://github.com/openwrt/routing) or the [luci feed](https://github.com/openwrt/luci).

dhewg, 2024-04-10:
There's a video feed which would be the appropriate place to put it: https://github.com/openwrt/video
And it's already commented in the main repo's feeds.conf.default

feckert, 2024-04-10:
@dhewg Somehow I thought I'd seen this before. But unfortunately I couldn't remember. Thanks for the info.
@esaaprillia That would be the right way to go.

esaaprillia, 2024-04-10:
@dhewg 

I don't want to save it on video, many packages are damaged. there may be some damaged packages that need to be removed in the video

esaaprillia, 2024-04-10:
@feckert 

as an example makefile I made it like this. What do you think

```
PKG_BUILD_DIR:=$(BUILD_DIR)/Xorg/libXdmcp-$(PKG_VERSION)
PKG_INSTALL:=1

PKG_MAINTAINER:=Esa Aprilia Salsabila <esaapriliasalsabila@gmail.com>
PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=COPYING

PKG_BUILD_DEPENDS:=util-macros xorgproto

include $(INCLUDE_DIR)/package.mk

define Package/libxdmcp
	SECTION:=libs
	CATEGORY:=Libraries
	SUBMENU:=Xorg
	TITLE:=X Display Manager Control Protocol library
	URL:=http://xorg.freedesktop.org/
endef

define Package/libxdmcp/description
	libXdmcp - X Display Manager Control Protocol library
endef
```

esaaprillia, 2024-04-10:
if anyone is interested in the xorg package

https://github.com/esaaprillia/xorg

brada4, 2024-04-10:
Add xauth and check ssh -X ? ;-)

esaaprillia, 2024-04-10:
@brada4 

```
root@OpenWrt:/usr/lib# ssh -X
WARNING: Ignoring unknown option -X
Dropbear SSH client v2020.81 https://matt.ucc.asn.au/dropbear/dropbear.html
Usage: ssh [options] [user@]host[/port] [command]
-p <remoteport>
-l <username>
-t    Allocate a pty
-T    Don't allocate a pty
-N    Don't run a remote command
-f    Run in background after auth
-y    Always accept remote host key if unknown
-y -y Don't perform any remote host key checking (caution)
-s    Request a subsystem (use by external sftp)
-o option     Set option in OpenSSH-like format ('-o help' to list options)
-i <identityfile>   (multiple allowed, default .ssh/id_dropbear)
-A    Enable agent auth forwarding
-L <[listenaddress:]listenport:remotehost:remoteport> Local port forwarding
-g    Allow remote hosts to connect to forwarded ports
-R <[listenaddress:]listenport:remotehost:remoteport> Remote port forwarding
-W <receive_window_buffer> (default 24576, larger may be faster, max 1MB)
-K <keepalive>  (0 is never, default 0)
-I <idle_timeout>  (0 is never, default 0)
-J <proxy_program> Use program pipe rather than TCP connection
-c <cipher list> Specify preferred ciphers ('-c help' to list options)
-m <MAC list> Specify preferred MACs for packet verification (or '-m help')
-b    [bind_address][:bind_port]
-V    Version
```

```
root@OpenWrt:/usr/lib# xauth -h
usage:  xauth [-options ...] [command arg ...]
where options include:
    -f authfilename                name of authority file to use
    -v                             turn on extra messages
    -q                             turn off extra messages
    -i                             ignore locks on authority file
    -b                             break locks on authority file
    -n                             do not resolve host names in authority file
    -V                             show version number of xauth
and commands have the following syntax:
    add dpyname protoname hexkey   add entry
    exit                           save changes and exit program
    extract filename dpyname...    extract entries into file
    help [topic]                   print help
    info                           print information about entries
    list [dpyname...]              list entries
    merge filename...              merge entries from files
    nextract filename dpyname...   numerically extract entries
    nlist [dpyname...]             numerically list entries
    nmerge filename...             numerically merge entries
    quit                           abort changes and exit program
    remove dpyname...              remove entries
    source filename                read commands from file
    version                        show version number of xauth
    ?                              list available commands
    generate dpyname protoname [options]  use server to generate entry
    options are:
      timeout n    authorization expiration time in seconds
      trusted      clients using this entry are trusted
      untrusted    clients using this entry are untrusted
      group n      clients using this entry belong to application group n
      data hexkey  auth protocol specific data needed to generate the entry
A dash may be used with the "merge" and "source" to read from the
standard input.  Commands beginning with "n" use numeric format.
```

esaaprillia, 2024-04-13:
Mesa can be built by adding this in the mesa Makefile

```
define Meson/CreateNativeFile
	$(STAGING_DIR_HOST)/bin/sed \
		-e "s|@CC@|$(foreach BIN,$(HOSTCC),'$(BIN)',)|" \
		-e "s|@CXX@|$(foreach BIN,$(HOSTCXX),'$(BIN)',)|" \
		-e "s|@PKGCONFIG@|$(PKG_CONFIG)|" \
		-e "s|@CMAKE@|$(STAGING_DIR_HOST)/bin/cmake|" \
		-e "s|@PYTHON@|$(STAGING_DIR_HOSTPKG)/bin/python3|" \
		-e "s|@CFLAGS@|$(foreach FLAG,$(HOST_CFLAGS) $(HOST_CPPFLAGS),'$(FLAG)',)|" \
		-e "s|@CXXFLAGS@|$(foreach FLAG,$(HOST_CXXFLAGS) $(HOST_CPPFLAGS),'$(FLAG)',)|" \
		-e "s|@LDFLAGS@|$(foreach FLAG,$(HOST_LDFLAGS),'$(FLAG)',)|" \
		-e "s|@PREFIX@|$(HOST_BUILD_PREFIX)|" \
		< $(MESON_DIR)/openwrt-native.txt.in \
		> $(1)
endef

define Meson/CreateCrossFile
	$(STAGING_DIR_HOST)/bin/sed \
		-e "s|@CC@|$(foreach BIN,$(TARGET_CC),'$(BIN)',)|" \
		-e "s|@CXX@|$(foreach BIN,$(TARGET_CXX),'$(BIN)',)|" \
		-e "s|@LD@|$(foreach FLAG,$(TARGET_LINKER),'$(FLAG)',)|" \
		-e "s|@AR@|$(TARGET_AR)|" \
		-e "s|@STRIP@|$(TARGET_CROSS)strip|" \
		-e "s|@NM@|$(TARGET_NM)|" \
		-e "s|@PKGCONFIG@|$(PKG_CONFIG)|" \
		-e "s|@CMAKE@|$(STAGING_DIR_HOST)/bin/cmake|" \
		-e "s|@PYTHON@|$(STAGING_DIR_HOSTPKG)/bin/python3|" \
		-e "s|@CFLAGS@|$(foreach FLAG,$(TARGET_CFLAGS) $(EXTRA_CFLAGS) $(TARGET_CPPFLAGS) $(EXTRA_CPPFLAGS),'$(FLAG)',)|" \
		-e "s|@CXXFLAGS@|$(foreach FLAG,$(TARGET_CXXFLAGS) $(EXTRA_CXXFLAGS) $(TARGET_CPPFLAGS) $(EXTRA_CPPFLAGS),'$(FLAG)',)|" \
		-e "s|@LDFLAGS@|$(foreach FLAG,$(TARGET_LDFLAGS) $(EXTRA_LDFLAGS),'$(FLAG)',)|" \
		-e "s|@ARCH@|$(MESON_ARCH)|" \
		-e "s|@CPU@|$(MESON_CPU)|" \
		-e "s|@ENDIAN@|$(if $(CONFIG_BIG_ENDIAN),big,little)|" \
		< $(MESON_DIR)/openwrt-cross.txt.in \
		> $(1)
endef
```

esaaprillia, 2024-04-18:
hy
@feckert @dhewg @brada4 

can you see this problem

I added host in util-macros and in xorgproto but when xorgproto is built it seems xorgproto doesn't find xorg-macros.pc in ```staging_dir/hostpkg/share/```

xorg-macros.pc which is generated when util-macros is created as a host

```
prefix=/home/runner/work/mosdns/mosdns/sdk/staging_dir/hostpkg
exec_prefix=/home/runner/work/mosdns/mosdns/sdk/staging_dir/hostpkg
includedir=${prefix}/include
datarootdir=${prefix}/share
datadir=${datarootdir}
PACKAGE=util-macros
# Used by XORG_INSTALL
pkgdatadir=${datadir}/${PACKAGE}
# docdir is kept for backwards compatibility with XORG_INSTALL from
# util-macros 1.4
docdir=${pkgdatadir}

Name: X.Org Macros
Description: A set of autoconf project macros for X.Org modules
Version: 1.19.2
```

esaaprillia, 2024-04-20:
Looks like someone has asked this on the forum but there's no answer at all

https://forum.openwrt.org/t/pkg-config-fails-to-find-libraries-when-building-host-tools-from-makefile-hosted-within-a-package/30625

esaaprillia, 2024-05-05:
@feckert @dhewg 

I'm confused about how to do a pull request for an Xorg package. there are two options for adding Xorg

1. videos
2. package

see Xorg package libx11 has other package dependencies.

If the Xorg libx11 package is pushed then the libx11 dependencies must be added to the pull request.

for the packages in the video I have tried updating the packages and all can be built except the qt5 package I did not try to build the package