openwrt/packages #23868: https-dns-proxy creates error on overview page after upgrade to 23.05.3

bassopt, 2024-04-10:
https-dns-proxy :

Environment: Openwrt 23.05.3 x86/64

Descriptioon :
After upgrading from version 23.05.2 to 23.05.3 https-dns-proxy created an error on the overview page. 
TypeError
  can't convert undefined to object
Upgrade was done by saving a backup of configuration. Upgrade firmware and restorethe backup file. Also  using a custom build from image builder using the latest stable. 
Uninstalling and Deleting https-dns-proxy config and reinstalling fixes it but it's still annoying. Fortunately it's not hard to reconfigure

bassopt, 2024-04-10:
Sorry I don’t know exactly where to report this. 

The config file was the version I was using on 23.05.02 ( imported from a backup file ) don’t know the package version exactly

stangri, 2024-04-10:
Not much I can do without the config file.

closed 2024-04-10 by stangri

bassopt, 2024-04-10:
I know you closed this, but here are the contents of the config file

`
config main 'config'
	option canary_domains_icloud '1'
	option canary_domains_mozilla '1'
	option dnsmasq_config_update '*'
	option force_dns '1'
	list force_dns_port '53'
	list force_dns_port '853'
	option procd_trigger_wan6 '0'
`

stangri, 2024-04-10:
@bassopt thank you for providing a sample config file which lead to unexpected behaviuor, the fix is on the way: https://github.com/openwrt/luci/pull/7052