openwrt/packages #23914: iperf3: enable sctp

nshopik, 2024-04-12:
Maintainer: @jonasjelonek

Environment: (ARMv8, Xiaomi Redmi Router AX6S, OpenWrt 23.05.3 r23809-234f1a2efa)

Description:
Maybe it configured  `--without-sctp`, it would be great to have sctp available.
```
iperf3 --sctp -c 127.0.0.1
iperf3: unrecognized option: sctp
```

jonasjelonek, 2024-04-12:
(I'm not the maintainer of the package. Please always have a look into the Makefile for the current maintainer instead of who authored the last commit)

In the Makefile no configure option regarding SCTP is set. Looking at the verbose build output for iperf3 shows that autoreconf checks for the header `netinet/sctp.h` but can't find it and thus disables SCTP support.

nshopik, 2024-04-13:
Oh, apologies Maintainer would be @nbd168

brada4, 2024-04-13:
netinet/sctp.h comes from libsctp, so another package flavour to not beef up "normal" package size.

nshopik, 2024-04-15:
I tried to build manually iperf3 --without-sctp on x86 with `./configure --without-openssl --without-sctp` and then `./configure --without-openssl` got same binary size - 35832 bytes. I made sure libsctp available before hand.