openwrt/packages #23924: Add Snapcast packages (snapserver, snapclient)

xabolcs, 2024-04-13:
**badaix** is done with SnapOs - he archived the repository on April 12, 2024 and [removed references to SnapOs from the Snapcast repository](https://github.com/badaix/snapcast/commit/b6c11930127d1ea6efe48de93b4add8ffe1fd0e9) at end of March 2024.

So I thought it's time to ask for inclusion into OpenWrt Packages.

----

**S**y**n**chronous **a**udio **p**layer
> [Snapcast](https://github.com/badaix/snapcast) is a multiroom client-server audio player, where all clients are time synchronized with the server to play perfectly synced audio. It's not a standalone player, but an extension that turns your existing audio player into a Sonos-like multiroom solution.
> Audio is captured by the server and routed to the connected clients. Several players can feed audio to the server in parallel and clients can be grouped to play the same audio stream.
> One of the most generic ways to use Snapcast is in conjunction with the music player daemon ([MPD](http://www.musicpd.org/)) or [Mopidy](https://www.mopidy.com/).

It has a [nice documentation](https://github.com/badaix/snapos/tree/master/openwrt) how to build for OpenWrt.
I use it on mini routers like HT-TM02 (32/8, stucked at OpenWrt 19.07 and Snapcast 0.26.0) or GL-AR300M, all good! :ok_hand:

brada4, 2024-04-13:
PR welcome

davidandreoletti, 2024-08-18:
@xabolcs  I am interested in this and already compile snapcast (server) on my end with OpenWRT snapshot.

commit eeb8d131 (2025-09-09): snapcast: add package snapserver and snapclient

Snapcast [1] is a multiroom client-server audio player, where all
clients are time synchronized with the server to play perfectly synced
audio.

It's not a standalone player, but an extension that turns your existing
audio player into a Sonos-like multiroom solution.
Audio is captured by the server and routed to the connected clients.
Several players can feed audio to the server in parallel and clients can
be grouped to play the same audio stream.
One of the most generic ways to use Snapcast is in conjunction with the
music player daemon (MPD) [ [... cut]
closed 2025-09-09 by 1715173329