openwrt/packages #23934: tini: Fails to compile with docker and latest OpenWRT snapshot

schuettecarsten, 2024-04-14:
```
FAILED: CMakeFiles/tini.dir/src/tini.c.o
/home/cschuette/openwrt/snapshot/openwrt/staging_dir/host/bin/ccache /home/cschuette/openwrt/snapshot/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-13.2.0_mu
sl/bin/aarch64-openwrt-linux-musl-gcc  -I/home/cschuette/openwrt/snapshot/openwrt/build_dir/target-aarch64_cortex-a72_musl/tini-0.19.0 -Os -pipe -g3 -fno-caller-sav
es -fno-plt -fhonour-copts -ffile-prefix-map=/home/cschuette/openwrt/snapshot/openwrt/build_dir/target-aarch64_cortex-a72_musl/tini-0.19.0=tini-0.19.0 -ffunction-se
ctions -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99 -We
rror -Wextra -Wall -pedantic-errors -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -DNDEBUG -MD -MT CMakeFiles/tini.dir/src/tini.c.o -MF CMakeFiles/tini.dir/src/tini.c.o.d -o CMakeFiles/tini.dir/src/tini.c.o -c /home/cschuette/openwrt/snapshot/openwrt/build_dir/target-aarch64_cortex-a72_musl/tini-0.19.0/src/tini.c
src/tini.c: In function 'print_usage':
src/tini.c:227:36: error: implicit declaration of function 'basename' [-Wimplicit-function-declaration]
  227 |         fprintf(file, "%s (%s)\n", basename(name), TINI_VERSION_STRING);
      |                                    ^~~~~~~~
src/tini.c:227:25: error: format '%s' expects argument of type 'char *', but argument 3 has type 'int' [-Werror=format=]
  227 |         fprintf(file, "%s (%s)\n", basename(name), TINI_VERSION_STRING);
      |                        ~^          ~~~~~~~~~~~~~~
      |                         |          |
      |                         char *     int
      |                        %d
src/tini.c:232:32: error: format '%s' expects argument of type 'char *', but argument 3 has type 'int' [-Werror=format=]
  232 |         fprintf(file, "Usage: %s [OPTIONS] PROGRAM -- [ARGS] | --version\n\n", basename(name));
      |                               ~^                                               ~~~~~~~~~~~~~~
      |                                |                                               |
      |                                char *                                          int
      |                               %d
src/tini.c:234:90: error: format '%s' expects argument of type 'char *', but argument 3 has type 'int' [-Werror=format=]
  234 |         fprintf(file, "Execute a program under the supervision of a valid init process (%s)\n\n", basename(name));
      |                                                                                         ~^        ~~~~~~~~~~~~~~
      |                                                                                          |        |
      |                                                                                          char *   int
      |                                                                                         %d
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:46: /home/cschuette/openwrt/snapshot/openwrt/build_dir/target-aarch64_cortex-a72_musl/tini-0.19.0/.built] Error 1
make[2]: Leaving directory '/home/cschuette/openwrt/snapshot/openwrt/feeds/packages/utils/tini'
time: package/feeds/packages/tini/compile#0.16#0.03#0.18
    ERROR: package/feeds/packages/tini failed to build.
make[1]: *** [package/Makefile:129: package/feeds/packages/tini/compile] Error 1
```

pesa1234, 2024-04-15:
I have same issue...

schuettecarsten, 2024-04-15:
#23929

pesa1234, 2024-04-15:
> #23929

Thanks

cyqtmxk, 2024-04-17:
same problem，thanks ！

anonymous-one, 2024-04-19:
bump! could we the the patch approved so all those (...many of us) using docker can go back to using the latest snapshots :D

thanks!

anonymous-one, 2024-04-20:
what needs to be done for this to get merged? docker fails to install due to the missing dep and has so far close to 2 weeks now :/

anonymous-one, 2024-04-21:
looks like we can (probably) close this.

https://github.com/openwrt/packages/commit/ddd379416ed01b541101644d3880dd9ece3de4d3

should start seeing this in the builds in a day or two.

schuettecarsten, 2024-04-24:
Fixed by https://github.com/openwrt/packages/commit/ddd379416ed01b541101644d3880dd9ece3de4d3

closed 2024-04-24 by schuettecarsten