openwrt/packages #23957: lang/python/python3: compile error: implicit declaration of function 'pwritev2'

aleks-mariusz, 2024-04-18:
# Issue template (remove lines from top till here)

Maintainer: @jefferyto 
Target Environment: target: mvebu-cortexa9, linksys wrt32x, OpenWrt build rev [2116073](https://github.com/openwrt/openwrt/tree/2116073f981ead48f5fe0af213cbdfe013288fb9) / package rev [6a400cc](https://github.com/openwrt/packages/tree/6a400cce62db170fd7e06179cf5a27eca482213f)
Build Environment: dockererized [container](https://github.com/aleks-mariusz/docker-openwrt-builder/blob/alpine/Dockerfile) running Alpine (i am contributor to the official build-system setup [page section](https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem#alpine:~:text=be%20your%20default.-,Alpine,-apk%20add%20argp) for Alpine).

Description:

I regularly recompile the openwrt firmware images every few weeks, however this latest attempt I'm having a new issue now where there's a failure happening during compiling lang/python/python3 due to what appears to be a issue detecting the syscall pwritev2 being available, when it seems it's not in my build environment

```
make[3]: Entering directory '/home/user/workdir/openwrt/feeds/packages/lang/python/python3'
rm -f /home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7/.built
touch /home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7/.built_check
CFLAGS="-O2 -pipe -mthumb -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -ffile-prefix-map=/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7=Python-3.11.7 ;
make[4]: Entering directory '/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7'
Running code to generate profile data (this can take a while):
# First, we need to create a clean build with profile generation
# enabled.
make profile-gen-stamp
make[5]: Entering directory '/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7'
Building with support for profile generation:
make build_all_generate_profile
make[6]: Entering directory '/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7'
make build_all CFLAGS_NODIST=" -fprofile-generate" LDFLAGS_NODIST=" -fprofile-generate" LIBS="-ldl  -lpthread"
make[7]: Entering directory '/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7'
arm-openwrt-linux-muslgnueabi-gcc  -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -O2 -pipe -mthumb -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -ffile-prefix-map=/home/user/workdir/openwrt/build_do
cc1: note: someone does not honour COPTS correctly, passed 2 times
./Modules/posixmodule.c: In function 'os_preadv_impl':
./Modules/posixmodule.c:9882:13: error: implicit declaration of function 'preadv2'; did you mean 'preadv'? [-Werror=implicit-function-declaration]
 9882 |         n = preadv2(fd, iov, cnt, offset, flags);
      |             ^~~~~~~
      |             preadv
./Modules/posixmodule.c: In function 'os_pwritev_impl':
./Modules/posixmodule.c:10528:18: error: implicit declaration of function 'pwritev2'; did you mean 'pwritev'? [-Werror=implicit-function-declaration]
10528 |         result = pwritev2(fd, iov, cnt, offset, flags);
      |                  ^~~~~~~~
      |                  pwritev
cc1: some warnings being treated as errors
make[7]: *** [Makefile:2743: Modules/posixmodule.o] Error 1
make[7]: Leaving directory '/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7'
make[6]: *** [Makefile:767: build_all_generate_profile] Error 2
make[6]: Leaving directory '/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7'
make[5]: *** [Makefile:743: profile-gen-stamp] Error 2
make[5]: Leaving directory '/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7'
make[4]: *** [Makefile:755: profile-run-stamp] Error 2
make[4]: Leaving directory '/home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7'
make[3]: *** [Makefile:364: /home/user/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/Python-3.11.7/.built] Error 2
make[3]: Leaving directory '/home/user/workdir/openwrt/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/compile#30.93#3.06#33.08
    ERROR: package/feeds/packages/python3 failed to build.
```

Any advice what i might be missing or what to check for?  I would have thought the ./configure script should have correctly detected pvread2 is not available and [this macro](https://github.com/python/cpython/blob/cd3e2d3a6ca4bfcc8c99eb50e92a2ce0b5b1ddf6/Modules/posixmodule.c#L9878-L9886) should have the `HAVE_PVREAD2` variable set to false, but i'm not super familiar with how that's detected or if that's even the cause so looking for ideas on what to check and what to try (ideally if there's just a package missing from the build environment i need to install).

yichya, 2024-04-18:
Did you try `make dirclean` and build again?

aleks-mariusz, 2024-04-18:
> Did you try `make dirclean` and build again?

This seems to have fixed it, it now compiles cleanly fully.  No idea how it got into this situation, but thank you for the idea!

closed 2024-04-18 by aleks-mariusz