openwrt/packages #23968: Not detecting a lack of Internet

RChadwick7, 2024-04-20:
Using the latest Travelmate, it will stay on an XFinity hotspot even if there is no Internet. For whatever reason, Xfinity keeps deleting this hotspot and the generic portal script doesn't work. There are other, less desirable yet mostly working hotspots available but Travelmate stays on the non working access point.
```

dibdot, 2024-04-20:
Check the online available readme (linked in the LuCI app), esp. the 'trm_netcheck' option.

closed 2024-04-20 by dibdot

hackerb9, 2025-01-12:
> Check the online available readme (linked in the LuCI app), esp. the 'trm_netcheck' option.

I think a better solution is to add a bad access point like "xfinity" to the list of stations manually and click "On/Off" to disable it. (Presuming you don't actually use xfinity).

Netcheck is dangerous to enable since it completely disables stations which lack Internet instead of marking them to be checked again later. That means a good station which had a temporary network outage will not be used even when it does have Internet. Undoing the problem requires the user to login to OpenWRT to re-enable the station manually.

@dibdot: Do I have it correct how `trm_netcheck` works?