openwrt/packages #23989: bandwidth monitor

rticzas, 2024-04-22:
The router overstates statistics. Windows downloaded 49 GB in 7 days and the router shows 566 GB for 6 days

[image]

[image]

OpenWrt 23.05-SNAPSHOT r23821-77b7c1088

brada4, 2024-04-24:
it is ok 50G in 7 days matches well with other picture with 566GB in 22 days
That is some kind of vendor application, this is to report bugs with OpenWRT produce, not vendor forks, like mentioning affected package FROM HERE NOT RANDOM VENDOR

andris155, 2026-08-15:
This error was caused for me when I enabled the hardware flow offloading.