openwrt/packages #23990: lvm2: too many dependencies?

Lanchon, 2024-04-22:
Maintainer: @dangowrt 
Environment: 23.05.3

`lvm2` pulls in `libdevmapper-selinux` instead of `libdevmapper`, which pulls in `libselinux` and deps. is this ok?

also, both `libdevmapper` and `libdevmapper-selinux` pull in tons of crypto and TPM stuff. this seems to high a cost to pay for simple device mapper usage. could crypto be separated?

dangowrt, 2024-04-22:
I don't see any selinux libraries installed **at run-time** for having `lvm2` installed.

> also, both `libdevmapper` and `libdevmapper-selinux` pull in tons of crypto and TPM stuff.

Can you elaborate please which libraries you are talking about and how they are related to *"TPM stuff"*?

That's the **run time** info for the packages you have mentioned:

```
...
Package: libdevmapper
Depends: libc, kmod-dm, libpthread, libuuid1, librt, libblkid1
...
Package: lvm2
Depends: libc, libreadline8, libncurses6, libaio, libdevmapper
...
```

So as you can see the dependencies are quite modest and really just what is needed to deal with block devices (**libblkid, libuuid**) and some small generic helpers (**libpthread, librt**), a helper for async IO (**libaio**) and some console libs for the lvm CLI (**libreadline, libncurses**).

Maybe you are talking about **build time** dependencies, and well, those are not optimized because they don't matter at run time. Feel free to open a PR to optimize build-time dependencies if you have time for that (I don't).

Lanchon, 2024-04-22:
well, i'm just trying to install lvm2 and this is what i get:
[image]

i'm assuming this means `libdevmapper-selinux` will be installed with all its deps including `libselinux` and crypto (incl `kmod-tpm`). am i mistaken?

dangowrt, 2024-04-22:
It is quite strange (but well, opkg...) that `libdevmapper-selinux` is preferred over `libdevmapper`. Try manually installing `libdevmapper` first, that should resolve it. The `-selinux` variants are only needed on SELinux-enabled installations and don't make any sense unless you are planning to use `selinux-policy` (and a SELinux-enabled kernel).

The kernel module dependencies are unavoidable thought (dependencies of `kmod-dm` which includes `kmod-keys-encrypted` and goes all the way down to `kmod-asn1-encoder`), that's the monolithic nature of Linux and hence limitation by design. Some of those dependencies could probably be avoided if building a kernel with those features entirely missing -- but then you also could not install them as a module...

Lanchon, 2024-04-22:
i understand about the kernel config issue. i wonder what would happen if the modules are not there; as in, would any code try to load them?

but anyways, the `libdevmapper-selinux` thing is what made me create this issue. if all is right with this package and its an issue outside of its control, feel free to close.

thanks for taking the time to explain.

dangowrt, 2024-04-23:
> i understand about the kernel config issue. i wonder what would happen if the modules are not there; as in, would any code try to load them?

If the kernel was *built* with support for `keys-encrypted` enabled, the result is that the `dm` module has a hard dependency on the `keys-encrypted` module. There are, of course, possible structural improvements which can avoid this hard dependency even on a monolithic kernel (the actual crypto algorithms also built as modules are an example for that). However, I assume that nobody implemented such mechanism with respect to `dm`'s use of `keys-encrypted` which hence results in a hard dependency.

> but anyways, the `libdevmapper-selinux` thing is what made me create this issue. if all is right with this package and its an issue outside of its control, feel free to close.

I don't think there is much we can do about it at this point. Unfortunately `opkg` doesn't have a way to express priorities for `PROVIDES` or any mechanism other than prior user selection when it comes to which (`libdevmapper` or `libdevmapper-selinux`) to chose by default. I hope this can be handled better once we [switch to `apk`](https://github.com/openwrt/openwrt/pull/4294) instead of `opkg`.

closed 2024-04-23 by dangowrt