openwrt/packages #24010: <muonsocks>: microsocks enhancement package request

lars18th, 2024-04-25:
The microsocks package already exists: https://pkgs.staging.openwrt.org/package/master/packages/aarch64_cortex-a76/microsocks

But it only supports Socks v5. For v4a support the we need to use "muonsocks" that it's a fork: https://github.com/niklata/muonsocks

I don't recommend to "replace" the original microsocks package, but to "add" the new one. To compile the fork we can reuse the receipt for the original package.

lars18th, 2024-05-24:
Hi, I need help because "microsocks" is V5 only, and "muonsocks" is a fork that support V4a as well.

lars18th, 2024-05-28:
Hi @matkor ,

> Package : microsocks
> Maintainer : Mateusz Korniak

Since you are listed as maintainer of the `microsocks` package I ask if you can: either replace it with the `mounsocks` code or create a new package. Please comment. Thank you.

matkor, 2024-05-29:
Seems mounsocks functionality is larger, also it already has different name, so IMHO it is better to add muonsocks package and let users choose.

lars18th, 2024-05-30:
> Seems mounsocks functionality is larger, also it already has different name, so IMHO it is better to add muonsocks package and let users choose.

What you say sounds perfect to me. However, I don't have enough skills to create a PR for a new package. And since you are the maintainer of `microsocks` then the question is if you can generate a new one for `muonsocks`. Please. 😉

vellrya, 2024-09-10:
@matkor, сan you please update microsocks to version 1.0.4 so that it is possible to use the -q flag? Thanks

lars18th, 2024-10-01:
Hi @vellrya,

> @matkor, сan you please update microsocks to version 1.0.4 so that it is possible to use the -q flag? Thanks

The version 1.0.4 is already updated in the repo: https://github.com/openwrt/packages/blob/master/net/microsocks/Makefile

However, I don't know why with the last 23.05.5 version in my router, in the packages only the version 1.0.3 is listed. Any idea?

And @matkor , could you add `muonsocks` as a new package please? I think it will be easy for you, right?