openwrt/packages #24015: mdadm: segfaults on creation, assembly of raid member

BigSleezy00, 2024-04-26:
Using snapshot on x86/64. Newest build- SNAPSHOT (r26043-1991bfb814). 

Whenever I try to build a new raid member or reassemble one from another OS, I always get a segfault. Verified drives are good, with finnix, where I built a new array and then tried mdadm on openwrt. No joy. I'm not sure what direction to go. Ideas?

```
mdadm --create /dev/md0 --chunk=64 -l 0 -n 2 /dev/sda /dev/sdc
[ 1303.479034] mdadm[4550]: segfault at 366fef3c IP 00007f98c6f3f252.... error 4 in libc.so []7f98c6efa000+54000] likely on CPU 7 (core 3, socket 0)
...
segmentation fault

```

neheb, 2024-05-18:
If we look at https://git.alpinelinux.org/aports/tree/main/mdadm?h=master , we see that time_t bugs are patched. That might be the reason it's crashing actually...

neheb, 2024-06-08:
Merged. Closing.

closed 2024-06-08 by neheb