openwrt/packages #24026: xtables-addons: error compiling with test kernel 6.6

Fil010, 2024-04-28:
Maintainer: @jow-
Environment: (x86/64, Default string Default string, OpenWrt SNAPSHOT)

Description:

Good day to all

Compiling an image for x86/64 the xtables-addons package presents the following error when I want to use the 6.6 test kernel

Gratitude to all

```
hecking Xtables module directory... /usr/lib/iptables
checking kernel version that we will build against... 6.6.28.0 in /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.28
WARNING: That kernel version is not officially supported yet. Continue at own luck.
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating Makefile.iptrules
config.status: creating Makefile.mans
config.status: creating extensions/Makefile
config.status: creating extensions/ACCOUNT/Makefile
config.status: creating extensions/pknock/Makefile
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
configure: WARNING: unrecognized options: --disable-nls
touch /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/.built
touch /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/.built_check
make  -C /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24 KCFLAGS="-fmacro-prefix-map=/home/fil/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/fil/openwrt/staging_dir/host/include -I/home/fil/openwrt/staging_dir/hostpkg/include -I/home/fil/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Apr 27 21:44:48 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/fil/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers" CC="x86_64-openwrt-linux-musl-gcc" KERNELRELEASE=6.6.28 DESTDIR="/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/ipkg-install" DEPMOD="/bin/true" all
make[3]: Entering directory '/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
make  all-recursive
make[4]: Entering directory '/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
Making all in extensions
make[5]: Entering directory '/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions'
Xtables-addons 3.24 - Linux 6.6.28
if [ -n "/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.28" ]; then make -C /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.28 M=/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions modules; fi;
make[6]: Entering directory '/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.28'
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/ACCOUNT/xt_ACCOUNT.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/pknock/xt_pknock.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/rtsp/nf_nat_rtsp.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/rtsp/nf_conntrack_rtsp.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/xt_LUA_target.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/nf_lua.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_helpers.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/byte_array.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/controller.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_ethernet.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_icmp.o
  CC [M]  /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_ip.o
In file included from /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_ip.c:20:
./include/net/checksum.h: In function 'csum_shift':
./include/net/checksum.h:90:40: error: implicit declaration of function 'ror32' [-Werror=implicit-function-declaration]
   90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
      |                                        ^~~~~
In file included from ./include/linux/kernel.h:22,
                 from ./include/linux/skbuff.h:13,
                 from ./include/linux/tcp.h:17,
                 from ./include/net/tcp.h:20,
                 from /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_ip.c:21:
./include/linux/bitops.h: At top level:
./include/linux/bitops.h:134:21: error: conflicting types for 'ror32'; have '__u32(__u32,  unsigned int)' {aka 'unsigned int(unsigned int,  unsigned int)'}
  134 | static inline __u32 ror32(__u32 word, unsigned int shift)
      |                     ^~~~~
./include/net/checksum.h:90:40: note: previous implicit declaration of 'ror32' with type 'int()'
   90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
      |                                        ^~~~~
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:243: /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_ip.o] Error 1
make[8]: *** [scripts/Makefile.build:480: /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA] Error 2
make[7]: *** [/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.28/Makefile:1913: /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions] Error 2
make[6]: *** [Makefile:234: __sub-make] Error 2
make[6]: Leaving directory '/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.28'
make[5]: *** [Makefile:468: modules] Error 2
make[5]: Leaving directory '/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions'
make[4]: *** [Makefile:625: all-recursive] Error 1
make[4]: Leaving directory '/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
make[3]: *** [Makefile:397: all] Error 2
make[3]: Leaving directory '/home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
make[2]: *** [Makefile:203: /home/fil/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/.built] Error 2
make[2]: Leaving directory '/home/fil/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#22.55#5.02#28.02
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[1]: *** [package/Makefile:129: package/feeds/packages/xtables-addons/compile] Error 1
make[1]: Leaving directory '/home/fil/openwrt'
make: *** [/home/fil/openwrt/include/toplevel.mk:233: package/xtables-addons/compile] Error 2

```

brada4, 2024-04-28:
Can you try to bump upstream version locally?
https://inai.de/projects/xtables-addons/

ntzb, 2024-05-30:
same issue (x86, kernel 6.6), same error.
tried upstream 3.26, same result
```
In file included from /home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26/extensions/LUA/prot_buf_ip.c:20:
./include/net/checksum.h: In function 'csum_shift':
./include/net/checksum.h:90:40: error: implicit declaration of function 'ror32' [-Werror=implicit-function-declaration]
   90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
      |                                        ^~~~~
In file included from ./include/linux/kernel.h:22,
                 from ./include/linux/skbuff.h:13,
                 from ./include/linux/tcp.h:17,
                 from ./include/net/tcp.h:20,
                 from /home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26/extensions/LUA/prot_buf_ip.c:21:
./include/linux/bitops.h: At top level:
./include/linux/bitops.h:134:21: error: conflicting types for 'ror32'; have '__u32(__u32,  unsigned int)' {aka 'unsigned int(unsigned int,  unsigned int)'}
  134 | static inline __u32 ror32(__u32 word, unsigned int shift)
      |                     ^~~~~
./include/net/checksum.h:90:40: note: previous implicit declaration of 'ror32' with type 'int()'
   90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
      |                                        ^~~~~
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:243: /home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26/extensions/LUA/prot_buf_ip.o] Error 1
make[8]: *** [scripts/Makefile.build:480: /home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26/extensions/LUA] Error 2
make[7]: *** [/home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.30/Makefile:1913: /home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26/extensions] Error 2
make[6]: *** [Makefile:234: __sub-make] Error 2
make[6]: Leaving directory '/home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.30'
make[5]: *** [Makefile:460: modules] Error 2
make[5]: Leaving directory '/home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26/extensions'
make[4]: *** [Makefile:625: all-recursive] Error 1
make[4]: Leaving directory '/home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26'
make[3]: *** [Makefile:397: all] Error 2
make[3]: Leaving directory '/home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26'
make[2]: *** [Makefile:203: /home/user/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.26/.built] Error 2
make[2]: Leaving directory '/home/user/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#15.75#10.73#27.93
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[1]: *** [package/Makefile:173: package/feeds/packages/xtables-addons/compile] Error 1
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:233: package/feeds/packages/xtables-addons/compile] Error 2
```

ntzb, 2024-05-30:
this seems to help:
https://github.com/coolsnowwolf/lede/issues/11824#issuecomment-1911438167

Fil010, 2024-07-10:
Solved with the following [[24535](https://github.com/openwrt/packages/pull/24535)] , closed

closed 2024-07-10 by Fil010