openwrt/packages #24027: nft-qos: has misleading IPv6 support

asdfjkluiop, 2024-04-28:
# Issue template (remove lines from top till here)

Maintainer: @rosysong 
Environment: x86_64 VM OpenWRT 23.05.3

Description:

IPv6 is not supported despite the fact that you can configure it. While this package does detect the presence of IPv6 and switch to the `inet` family that's as far as its support goes. It continues to add rules with `ip` instead of `ip6` and so no IPv6 rules ever get inserted, it also relies on DHCP for dynamic QoS which is problematic given how DHCP is not universal like it is with IPv4 and even when used isn't used by all devices. Even then it uses the wrong DHCP file as its reference for clients that are connected and so never adjusts QoS values.

brada4, 2024-04-28:
Could you show `nft list ruleset`
sed -i s+abcf:efgh+x:x+g output.txt
same with mac and ip4, just leaving last group telling 2 machines from eachother.
I am not maintainer, but at least can remove misleading options.

asdfjkluiop, 2024-05-04:
I'm not entirely sure what you're asking for? What's the sed for? I figured you want me to run sed against the output of `nft list ruleset` but that doesn't seem to do anything?

brada4, 2024-05-04:
`nft list ruleset` would show whole generated firewall ruleset.
IF it contains your public IP or other private data you may replace that using sed command.

asdfjkluiop, 2024-05-31:
Sorry for the delay, I don't even need to filter the rule list because no v6 entries are even created for QoS. For reference I've also provided my QoS config
[nft-rules.txt](https://github.com/user-attachments/files/15507588/nft-rules.txt)
[qos-cfg.txt](https://github.com/user-attachments/files/15507589/qos-cfg.txt)