openwrt/packages #24030: frr: multiple definition of `xmalloc'

trippleflux, 2024-04-28:
Maintainer: @lucize 
Environment: x86_64, generic

Description:

Recently I am getting the following errors when frr compilation for SNAPSHOT 6e5edec159e2dc468607a5b9179f722857a38421
```
ccache /home/username/works/openwrt/staging_dir/host/bin/gcc -DHAVE_CONFIG_H -DSYSCONFDIR=\"/home/username/works/openwrt/staging_dir/hostpkg/etc/\" -DCONFDATE=0 -I.  -I. -I./include -I./lib -I.   -D_GNU_SOURCE -DBUILDING_CLIPPY -I/home/username/works/openwrt/staging_dir/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/include -I/home/username/works/openwrt/staging_dir/host/include/libelf -fms-extensions -fno-omit-frame-pointer -funwind-tables -Wall -Wextra -Wformat-nonliteral -Wformat-security -Wswitch-enum -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wundef -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers -pthread -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.11 -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.11 -O2 -I/home/username/works/openwrt/staging_dir/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/include -c -o lib/clippy-typesafe.o `test -f 'lib/typesafe.c' || echo './'`lib/typesafe.c
ccache /home/username/works/openwrt/staging_dir/host/bin/gcc -DHAVE_CONFIG_H -DSYSCONFDIR=\"/home/username/works/openwrt/staging_dir/hostpkg/etc/\" -DCONFDATE=0 -I.  -I. -I./include -I./lib -I.   -D_GNU_SOURCE -DBUILDING_CLIPPY -I/home/username/works/openwrt/staging_dir/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/include -I/home/username/works/openwrt/staging_dir/host/include/libelf -fms-extensions -fno-omit-frame-pointer -funwind-tables -Wall -Wextra -Wformat-nonliteral -Wformat-security -Wswitch-enum -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wundef -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers -pthread -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.11 -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.11 -O2 -I/home/username/works/openwrt/staging_dir/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/include -c -o lib/clippy-vector.o `test -f 'lib/vector.c' || echo './'`lib/vector.c
/usr/bin/env bash ./libtool  --tag=CC   --mode=link ccache /home/username/works/openwrt/staging_dir/host/bin/gcc -fms-extensions -fno-omit-frame-pointer -funwind-tables -Wall -Wextra -Wformat-nonliteral -Wformat-security -Wswitch-enum -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wundef -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers -pthread -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.11 -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.11 -O2 -I/home/username/works/openwrt/staging_dir/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/include -export-dynamic   -L/home/username/works/openwrt/staging_dir/host/lib -L/home/username/works/openwrt/staging_dir/hostpkg/lib -L/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib -g -o lib/clippy lib/clippy-jhash.o lib/clippy-clippy.o lib/clippy-command_graph.o lib/clippy-command_lex.o lib/clippy-command_parse.o lib/clippy-command_py.o lib/clippy-defun_lex.o lib/clippy-elf_py.o lib/clippy-graph.o lib/clippy-libfrr_trace.o lib/clippy-memory.o lib/clippy-typesafe.o lib/clippy-vector.o -L/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.11/config-3.11-x86_64-linux-gnu -L/home/username/works/openwrt/staging_dir/hostpkg/lib -lpython3.11 -ldl  -lm   -lelf -lpthread  -lrt
OpenWrt-libtool: link: ccache /home/username/works/openwrt/staging_dir/host/bin/gcc -fms-extensions -fno-omit-frame-pointer -funwind-tables -Wall -Wextra -Wformat-nonliteral -Wformat-security -Wswitch-enum -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wundef -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.11 -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.11 -O2 -I/home/username/works/openwrt/staging_dir/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/include -g -o lib/clippy lib/clippy-jhash.o lib/clippy-clippy.o lib/clippy-command_graph.o lib/clippy-command_lex.o lib/clippy-command_parse.o lib/clippy-command_py.o lib/clippy-defun_lex.o lib/clippy-elf_py.o lib/clippy-graph.o lib/clippy-libfrr_trace.o lib/clippy-memory.o lib/clippy-typesafe.o lib/clippy-vector.o -Wl,--export-dynamic  -L/home/username/works/openwrt/staging_dir/host/lib -L/home/username/works/openwrt/staging_dir/hostpkg/lib -L/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib -L/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.11/config-3.11-x86_64-linux-gnu -lpython3.11 -ldl -lm -Wl,--whole-archive /home/username/works/openwrt/staging_dir/host/lib/libelf.a -Wl,--no-whole-archive -lz -lpthread -lrt -pthread
/usr/bin/ld: /home/username/works/openwrt/staging_dir/host/lib/libelf.a(xmalloc.o): in function `xmalloc':
xmalloc.c:(.text+0x0): multiple definition of `xmalloc'; /home/username/works/openwrt/staging_dir/host/lib/libelf.a(libgnu_la-xmalloc.o):xmalloc.c:(.text+0x0): first defined here
/usr/bin/ld: /home/username/works/openwrt/staging_dir/host/lib/libelf.a(xmalloc.o): in function `xcalloc':
xmalloc.c:(.text+0x30): multiple definition of `xcalloc'; /home/username/works/openwrt/staging_dir/host/lib/libelf.a(libgnu_la-xmalloc.o):xmalloc.c:(.text+0x180): first defined here
/usr/bin/ld: /home/username/works/openwrt/staging_dir/host/lib/libelf.a(xmalloc.o): in function `xrealloc':
xmalloc.c:(.text+0x60): multiple definition of `xrealloc'; /home/username/works/openwrt/staging_dir/host/lib/libelf.a(libgnu_la-xmalloc.o):xmalloc.c:(.text+0x40): first defined here
/usr/bin/ld: /home/username/works/openwrt/staging_dir/host/lib/libelf.a(xstrdup.o): in function `xstrdup':
xstrdup.c:(.text+0x0): multiple definition of `xstrdup'; /home/username/works/openwrt/staging_dir/host/lib/libelf.a(libgnu_la-xmalloc.o):xmalloc.c:(.text+0x270): first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:8134: lib/clippy] Error 1
make[3]: Leaving directory '/home/username/works/openwrt/build_dir/hostpkg/frr-9852228d1e87bdbad13e0fed8313203f00bf26af'
make[2]: *** [Makefile:244: /home/username/works/openwrt/build_dir/hostpkg/frr-9852228d1e87bdbad13e0fed8313203f00bf26af/.built] Error 2
make[2]: Leaving directory '/home/username/works/openwrt/feeds/packages/net/frr'
time: package/feeds/packages/frr/host-compile#41.68#15.70#58.75
    ERROR: package/feeds/packages/frr [host] failed to build.
make[1]: *** [package/Makefile:127: package/feeds/packages/frr/host/compile] Error 1
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:233: package/frr/compile] Error 2
```

brada4, 2024-04-28:
It fails building mandatory host tools. What is your host os and how far you get cleaning and rebuilding them without ccache?

trippleflux, 2024-04-28:
This frr building error is just happen recently, It didn't happened before even with ccache, I am sure it also has something to do with recent elfutils related upgrade.

My host os is Debian SID, I didn't do any recent changes to my host os, i just removed old OpenWrt folder and redo compilation as always.

trippleflux, 2024-04-29:
Turned off ccache and recompiling doesn't do different things, the same errors.

trippleflux, 2024-04-29:
Possible temporary and really bad solution : 
999-hack-for-clippy.patch
```
diff -Naur a/lib/subdir.am b/lib/subdir.am
--- a/lib/subdir.am
+++ b/lib/subdir.am
@@ -456,7 +456,7 @@
 	lib/libfrr.la
 
 lib_clippy_CPPFLAGS = $(CPPFLAGS_BASE) -D_GNU_SOURCE -DBUILDING_CLIPPY
-lib_clippy_CFLAGS = $(AC_CFLAGS) $(PYTHON_CFLAGS)
+lib_clippy_CFLAGS = $(AC_CFLAGS) $(PYTHON_CFLAGS) -Wl,--allow-multiple-definition
 lib_clippy_LDADD = $(PYTHON_LIBS) $(UST_LIBS) -lelf
 # no $(SAN_FLAGS) here
 lib_clippy_LDFLAGS = -export-dynamic $(AC_LDFLAGS) $(AC_LDFLAGS_EXEC)
```

httpstorm, 2024-04-29:
@robimarko 
I compiled a full image, only to discover that `package/feeds/packages/frr/host` is not enabled in my configuration. I'm not sure what selects it, so I tried building it manually on Ubuntu 22.04.4 LTS and I reproduced the error which @trippleflux reported. It also fails on macOS but due to a different error. Logs below.
```
make package/feeds/packages/frr/host/{clean,compile} -j 1 V=s
```
[x64.2024-04-29.01-frr-ubuntu.txt](https://httpstorm.com/share/.openwrt/test/2024-03-12_xcode-15.3/x64.2024-04-29.01-frr-ubuntu.txt)
[x64.2024-04-29.02-frr-macos.txt](https://httpstorm.com/share/.openwrt/test/2024-03-12_xcode-15.3/x64.2024-04-29.02-frr-macos.txt)

trippleflux, 2024-04-29:
@httpstorm 
Thanks for joining in, could it really be related with recent elfutils upgrade?, as the linker when frr invoking is mentioning about libelf.

robimarko, 2024-04-29:
Yeah, indeed they are both using exporting the same symbols and then frr is trying to link against libelf so they conflict.

I checked out the tools at the ZSTD fPIC commit and then frr compiles

httpstorm, 2024-04-29:
@robimarko @mpratt14 @guidosarducci 
These functions are defined in `build_dir/host/elfutils/libgnu/xmalloc.c`. Files in `libgnu` are copied from `build_dir/host/gnulib/lib/`. If I disable the duplicate functions via `#if 0` in `gnulib` (the same patched file appears in `elfutils/libgnu/xmalloc.c`), then `frr` compiles. So pretty much any project that tries to link agains both and use any of these functions is expected to fail. I wonder if we need those copies of C files? If `elfutils` needs them, it should instead link to `gnulib`. At this commit https://github.com/openwrt/openwrt/commit/b6f025b42429ec809afeb0956b35535c248b6c1b `frr/host` is broken, but if I go one step back, `elfutils` does not build. Before the PR everything builds fine.
The macOS error is unrelated and was present before the PR so I can't test there. The Ubuntu computer is 2010 slow.

robimarko, 2024-04-29:
@httpstorm Well, for me the question is what exact change made it statically link those symbols into libelf as it all worked with 0.189

robimarko, 2024-04-29:
BTW, I tried 0.190 as well and it has the same issue, so it should be something in 0.190

httpstorm, 2024-04-29:
@robimarko 
- [tools/missing-macros: add symlinks to makeinfo](https://github.com/openwrt/openwrt/commit/cdd56fc8934142a7df68118830611617ddb5e8fc) last good state
- [tools/gnulib: update to branch stable-202401](https://github.com/openwrt/openwrt/commit/4de8c0e1d82275108ced490e6cb02e62c0ef2da5): `elfutils` breaks
- 6 commits
- - [tools/gnulib: add macros to skip reallocarray() functions](https://github.com/openwrt/openwrt/commit/5ade7ee60ef726869d421673afc9aee97e851772)
- - [tools/gnulib: make tdestroy() fully portable](https://github.com/openwrt/openwrt/commit/eb726c90be67b8aa3638e2b9e988cd3865ed3b49)
- - [gettext-full: update to 0.22.5](https://github.com/openwrt/openwrt/commit/098bde1f3ebee9b5f302924474a99c00d6cc0242)
- - [tools/bison: hardcode path to m4 with STAGING_DIR_HOST](https://github.com/openwrt/openwrt/commit/41bc16dcc4594cd85fb87942032dcbec0ade7068)
- - [tools/coreutils: update to 9.4](https://github.com/openwrt/openwrt/commit/0489436506f20207dc2177fc30e84f7fb0cc9d07)
- - https://github.com/openwrt/openwrt/commit/44625e9d95882a3d868df2c83d304fc32cc5afc0
- [tools/elfutils: update to 1.91](https://github.com/openwrt/openwrt/commit/b6f025b42429ec809afeb0956b35535c248b6c1b) fixes `elfutils`: `frr` is broken

We could try to build without some of the 6 commits in between (could be one of them) or diff the two `elfutil` versions.
Well if you suspect 0.189 -> 0.190 that narrows it down. I'll check the changes.

robimarko, 2024-04-29:
I tried downgrading to 0.190 and it fails in the same way.
I am not sure if gnulib could have anything with it since xmalloc and the rest are implemented directly in elfutils.

httpstorm, 2024-04-29:
I have an interesting idea that will allow me to focus on `libelf` without interferences from the rest of the commits.
Here's my implementation for upgrading `gnulib` `coreutils` and `elfutils`. I'll test `frr` there.
https://github.com/httpstorm/openwrt/commits/elfutils-0.191/

robimarko, 2024-04-29:
Ok, so the error seems to be not directly in elfutils.
I downgraded it to 0.189 manually, keeping everything else the same and the same error is still present.

robimarko, 2024-04-29:
Well, reverting the gnulib update to 202401 and the two follow-up realloc and tdestroy commits doesn't help with 0.191

httpstorm, 2024-04-29:
Reproduced on my branch. That narrows it to `elfutils` or `gnulib` and excludes those 6 commits. Good night!

httpstorm, 2024-04-30:
On my branch I can revert back to the old `gnulib`. `elfutils-0.191` still compiles, `frr` error is reproduced. That narrows it to `elfutils`. Then I compiled 0.190, same error in `frr`. So either 0.189 -> 0.190 or the changes to Makefile and portability patch.

robimarko, 2024-04-30:
@httpstorm Yeah, without `dirname` 0.191 will fail to build since the header was added in 0.191.

BTW, I really dislike the amount of macOS fixes that are just showed regularly to various tooling and then nobody knows what it does nor how its supposed to work

robimarko, 2024-04-30:
Dropped dirname and `dirname.h` includes on macOS 14.4.1 in elfutils and it still compiled just fine.

Building the rest to see if frr will compile fine, if so we can just drop our hack since elfutils has a basename helper now.

httpstorm, 2024-04-30:
I fixed it on my branch. I'm not sure if it still builds on macOS.
I'll try to apply this on top of `main`, and test macOS.
https://github.com/httpstorm/openwrt/commits/elfutils-0.191/

``` diff
diff --git a/tools/elfutils/Makefile b/tools/elfutils/Makefile
index adf8770783..c8dd6ba98e 100644
--- a/tools/elfutils/Makefile
+++ b/tools/elfutils/Makefile
@@ -43,7 +43,7 @@ endif
 
 Hooks/HostConfigure/Pre := Host/Gnulib $(Hooks/HostConfigure/Pre)
 define Host/Gnulib
-	cd $(HOST_BUILD_DIR); $(STAGING_DIR_HOST)/bin/gnulib-tool --libtool --source-base=libgnu --import argp dirname fts obstack progname strchrnul tsearch;
+	cd $(HOST_BUILD_DIR); $(STAGING_DIR_HOST)/bin/gnulib-tool --libtool --source-base=libgnu --import argp fts obstack progname strchrnul tsearch;
 	ln -sf ../lib/eu-config.h $(HOST_BUILD_DIR)/libgnu/;
 endef
 
diff --git a/tools/elfutils/patches/100-portability.patch b/tools/elfutils/patches/100-portability.patch
index c557e5043f..a7c9d5bbe4 100644
--- a/tools/elfutils/patches/100-portability.patch
+++ b/tools/elfutils/patches/100-portability.patch
@@ -375,17 +375,6 @@
 +#endif
 +
  #endif
---- a/libdwfl/libdwflP.h
-+++ b/libdwfl/libdwflP.h
-@@ -31,6 +31,8 @@
- 
- #include <libdwfl.h>
- #include <libebl.h>
-+#include <libeu.h>
-+#include <dirname.h>
- #include <assert.h>
- #include <dirent.h>
- #include <errno.h>
 --- /dev/null
 +++ b/lib/stdio_ext.h
 @@ -0,0 +1,6 @@
@@ -395,19 +384,6 @@
 +#else
 +#define __fsetlocking(...) 0
 +#endif
---- a/libdw/libdwP.h
-+++ b/libdw/libdwP.h
-@@ -32,8 +32,10 @@
- #include <stdbool.h>
- #include <pthread.h>
- 
-+#include <libeu.h>
- #include <libdw.h>
- #include <dwarf.h>
-+#include <dirname.h>
- 
- 
- /* Known location expressions already decoded.  */
 --- a/libdw/Makefile.am
 +++ b/libdw/Makefile.am
 @@ -34,14 +34,12 @@ endif
-- 
2.34.1
```

robimarko, 2024-04-30:
@httpstorm Yeah, that is what I essentially did: https://github.com/openwrt/openwrt/pull/15337

robimarko, 2024-04-30:
@httpstorm I dug and refresh this patch to make it host compile on macOS:
```
From ad26e0926d1fa20bfdd9b5f63cec4db9837f880a Mon Sep 17 00:00:00 2001
From: Ruben Kerkhof <ruben@rubenkerkhof.com>
Date: Wed, 18 Mar 2020 15:40:39 +0100
Subject: [PATCH] lib: fix error on MacOS

Sections use a different syntax for Mach-O executables.

Fixes:

lib/bfd.c:35:1: error: argument to 'section' attribute is not valid for this target: mach-o section specifier requires a segment and section separated by a
      comma
DEFINE_MTYPE_STATIC(LIB, BFD_INFO, "BFD info")
^
./lib/memory.h:140:2: note: expanded from macro 'DEFINE_MTYPE_STATIC'
        DEFINE_MTYPE_ATTR(group, name, static, desc)                           \
        ^
./lib/memory.h:110:26: note: expanded from macro 'DEFINE_MTYPE_ATTR'
                __attribute__((section(".data.mtypes"))) = { {                 \
                                       ^
1 error generated.

---
 lib/memory.h | 43 +++++++++++++++++++++++++++----------------
 1 file changed, 27 insertions(+), 16 deletions(-)

--- a/lib/memory.h
+++ b/lib/memory.h
@@ -67,17 +67,21 @@ struct memgroup {
  *         but MGROUP_* aren't.
  */
 
+#ifdef __MACH__
+#define ATTRIBUTE_MGROUPS __attribute__((section("__DATA,mgroups")))
+#else
+#define ATTRIBUTE_MGROUPS __attribute__((section(".data.mgroups")))
+#endif
+
 #define DECLARE_MGROUP(name) extern struct memgroup _mg_##name
 #define _DEFINE_MGROUP(mname, desc, ...)                                       \
-	struct memgroup _mg_##mname                                            \
-		__attribute__((section(".data.mgroups"))) = {                  \
-			.name = desc,                                          \
-			.types = NULL,                                         \
-			.next = NULL,                                          \
-			.insert = NULL,                                        \
-			.ref = NULL,                                           \
-			__VA_ARGS__                                            \
-	};                                                                     \
+	struct memgroup _mg_##mname ATTRIBUTE_MGROUPS = {                      \
+		.name = desc,                                                  \
+		.types = NULL,                                                 \
+		.next = NULL,                                                  \
+		.insert = NULL,                                                \
+		.ref = NULL,                                                   \
+ 	};                                                                    \
 	static void _mginit_##mname(void) __attribute__((_CONSTRUCTOR(1000))); \
 	static void _mginit_##mname(void)                                      \
 	{                                                                      \
@@ -95,6 +99,12 @@ struct memgroup {
 	}                                                                      \
 	MACRO_REQUIRE_SEMICOLON() /* end */
 
+#ifdef __MACH__
+#define ATTRIBUTE_MTYPES __attribute__((section("__DATA,mtypes")))
+#else
+#define ATTRIBUTE_MTYPES __attribute__((section(".data.mtypes")))
+#endif
+
 #define DEFINE_MGROUP(mname, desc) \
 	_DEFINE_MGROUP(mname, desc, )
 #define DEFINE_MGROUP_ACTIVEATEXIT(mname, desc) \
@@ -105,14 +115,13 @@ struct memgroup {
 	/* end */
 
 #define DEFINE_MTYPE_ATTR(group, mname, attr, desc)                            \
-	attr struct memtype MTYPE_##mname[1]                                   \
-		__attribute__((section(".data.mtypes"))) = { {                 \
-			.name = desc,                                          \
-			.next = NULL,                                          \
-			.n_alloc = 0,                                          \
-			.size = 0,                                             \
-			.ref = NULL,                                           \
-	} };                                                                   \
+	attr struct memtype MTYPE_##mname[1] ATTRIBUTE_MTYPES = {{             \
+		.name = desc,                                                  \
+		.next = NULL,                                                  \
+		.n_alloc = 0,                                                  \
+		.size = 0,                                                     \
+		.ref = NULL,                                                   \
+	}};                                                                    \
 	static void _mtinit_##mname(void) __attribute__((_CONSTRUCTOR(1001))); \
 	static void _mtinit_##mname(void)                                      \
 	{                                                                      \
@@ -120,7 +129,7 @@ struct memgroup {
 			_mg_##group.insert = &_mg_##group.types;               \
 		MTYPE_##mname->ref = _mg_##group.insert;                       \
 		*_mg_##group.insert = MTYPE_##mname;                           \
-		_mg_##group.insert = &MTYPE_##mname->next;                      \
+		_mg_##group.insert = &MTYPE_##mname->next;                     \
 	}                                                                      \
 	static void _mtfini_##mname(void) __attribute__((_DESTRUCTOR(1001)));  \
 	static void _mtfini_##mname(void)                                      \

```

trippleflux, 2024-04-30:
Confirmed fixed, once again thanks @robimarko & @httpstorm .

closed 2024-04-30 by trippleflux