openwrt/packages #24050: make ovpn-dco ready for openvpn

tbsky, 2024-04-30:
Hi:
    ovpn-dco was in openwrt since 22.03. but openvpn is too old so seems nobody had test it.
I tried to compile/test ovpn-dco and openvpn 2.6.10 under 23.05, and found ovpn-dco module need some adjusting  so openvpn can really use it.

1. need to compile with "CONFIG_MODULE_STRIPPED=n" for the necessary file " /sys/module/ovpn_dco_v2/version"
2. need to depend module CONFIG_CRYPTO_CHACHA20POLY1305" "CONFIG_CRYPTO_LIB_POLY1305" and "CONFIG_CRYPTO_LIB_CHACHA" to use chacha20poly1035.

Spudz76, 2024-05-21:
Pivoting to this thread since spamming the [main kernel-6.6-for-ramips thread](https://github.com/openwrt/openwrt/pull/15369) is not really the right place anymore.

`+kmod-crypto-chacha20poly1305` by itself already pulls in `+kmod-crypto-lib-chacha20 +kmod-crypto-lib-poly1305` (and `+kmod-crypto-aead` which can thus be removed)

Working on those dep fixes, currently testing a build with adjustments.  Thing is kernel modules build keeps crashing on missing `chacha-mips.ko` but the deps are now the same as `kmod-wireguard` which builds everything fine (confusing).  Everything in the makefiles should be pulling that module in.  I have done just about every clean other than distclean.

Spudz76, 2024-05-21:
Okay I think the problem is that since `kmod-wireguard` is in-tree it can do all the cool submodule deps weaving from `drivers/net/Kconfig`, whilst and out-of-tree module build seems like it can't trigger all the same Kconfig things.  Trying to figure out a way around that.

Still builds fine as long as `kmod-wireguard` is enabled so it can pull in the deps.

Xeon64, 2024-05-27:
A user wanted me to add my current bug report / ticket to this one so that's what I'm doing..

https://github.com/openwrt/openwrt/issues/15491

It seems to be a common issue from distro to distro of DCO not working.. working then not working again.. people maintaining this package don't seem to correctly test  before releasing a issue.

Even the latest Ubuntu non-LTS (noble) is not immune.. while it was working with their previous versions perfectly fine.. DCO is not working with the latest 6.8 kernel for some reason.. even self-compiled DCO modules from github fails.

Torbins, 2024-12-10:
What about openvpn executable itself, does standard one work, or you need to build it manually?

Spudz76, 2024-12-10:
> What about openvpn executable itself, does standard one work, or you need to build it manually?

`OPENVPN_<ssl-flavor>_ENABLE_DCO` defaults to `n`, and depends on `OPENVPN_<ssl-flavor>_ENABLE_IPROUTE2` which also defaults to `n`

So I expect the prebuilds do not have those set `y` unless overridden elsewhere.

closed 2025-06-18 by BKPepe