openwrt/packages #24063: frr: fails to compile when submodule frr-libfrr is not enabled

httpstorm, 2024-05-02:
`frr`: fails to compile when submodule `frr-libfrr` is not enabled.

Maintainer: @lucize 
Environment: (x86_64, OpenWrt main https://github.com/openwrt/openwrt/commit/9e86e0b33bfa1a81cfc92647def72f151da2e5b7)
Reproduced on: macOS 14.4.1 and Ubuntu 22.04.4 LTS

Description:

``` Makefile
./scripts/diffconfig.sh

CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_FRR_INTERNAL=y
CONFIG_FRR_NO_SNMP=y
CONFIG_MOLD=y
CONFIG_PACKAGE_frr=y
CONFIG_PACKAGE_frr-libfrr=y
CONFIG_PACKAGE_frr-watchfrr=y
CONFIG_PACKAGE_frr-zebra=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libprotobuf-c=y
CONFIG_PACKAGE_libyang=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_MOLD=y
# CONFIG_USE_MOLD is not set
```

``` bash
make package/feeds/packages/frr/{clean,compile} -j 1 V=s
…
lib/vty.c: In function 'vty_mgmt_resume_response':
lib/vty.c:195:27: error: 'VTYSH_READ' undeclared (first use in this function); did you mean 'VTY_READ'?
  195 |                 vty_event(VTYSH_READ, vty);
      |                           ^~~~~~~~~~
      |                           VTY_READ
```

``` C
// lib/vty.c
/* Vty events */
enum vty_event {
	VTY_SERV,
	VTY_READ,
	VTY_WRITE,
	VTY_TIMEOUT_RESET,
#ifdef VTYSH
	VTYSH_SERV,
	VTYSH_READ,
	VTYSH_WRITE
#endif /* VTYSH */
};
```

The error is a bug in `frr`: not all use cases of the `VTYSH_*` enums are guarded by `#ifdef VTYSH`. These enums are enabled by the `VTYSH` macro, which is defined if sub package `frr-vtysh` is enabled in `menuconfig`. According to the following support ticket https://github.com/FRRouting/frr/issues/15752#issuecomment-2059328993 , building without `frr-vtysh` is no longer supported.
> vtysh is a required component. We should disable the ability to turn this off

One way to solve this is to enable `frr-vtysh` by default:
``` diff
# net/frr/Makefile
 define Package/frr-vtysh
   $(call Package/frr/Default)
   DEPENDS+=+frr-libfrr +libreadline +libncurses +more
   TITLE:=integrated shell for frr routing software
   CONFLICTS:=quagga-vtysh
+  DEFAULT:=y if PACKAGE_frr
 endef
```

I don't see a package `quagga` (`frr` is a fork of that), so we shouldn't run into conflicts.
Though it would be a better choice to make it a hard-requirement, so that it cannot be disabled.

Later if we want to move forward to version 10 or master, it gets a bit more complicated.
First we need to update to `libyang >= 2.1.128`, I tested `libyang-2.1.148`, which is the latest 2.1.*. Likely safe, but we need to test a few packages that depend on it (list below). Then `frr-10` and master both build fine.

However if we update to the current release `libyang-2.2.8`, there is an API change in `struct ly_err_item`, defined in `libyang/log.h`, which results in another build error:
https://github.com/CESNET/libyang/commit/7a266775af9369e22065920a5c053fd78c432178#diff-67a354ecb35f341abf57f84bd0d714b05280fddb1063b90a31f0d3ef67675bd7
``` C
lib/vty.c:3676:28: error: 'struct ly_err_item' has no member named 'path'
 3676 |         bool have_path = ei->path && ei->path[0] != 0;
      |                            ^~
```
That would make it more complicated, considering the following packages also depend on `libyang`:
```
frr-libfrr frr libnetconf2 libsysrepo netopeer2-cli netopeer2-server yanglint
```

Note: the current support ticket is not related to PR https://github.com/openwrt/packages/pull/24053 , where we first started discussion on the issue.

@robimarko @neheb @trippleflux

trippleflux, 2024-05-02:
I also agree to enable frr-vtysh as default, and better be to bump up libyang (2.1.148) & netopeer2 (2.2.13) version.

httpstorm, 2024-05-02:
@robimarko 
Hey Robi, the following branch fixes the `frr-vtysh` dependency using my original fix. Indeed your idea is better. Please let me know how to implement the changes you suggested. Then the following commits update `libyang` and `frr`. At this point the existing version of `netopeer2` still builds fine.
https://github.com/httpstorm/openwrt.packages/commits/frr-enable-vtysh-by-default.2024-05-02/

Now about `netopeer2`: this one is more complicated. First we need to update to `libnetconf2-3.0.8`, then `sysrepo-2.2.150` which requires changes to `define Package/libsysrepo/install` because all files have `@date` postfix. And I'm not sure if I should keep or remove the postfix in the destination. Currently I removed it. These two compile, but then `netopeer2-2.2.13` fails:

``` text
-- Installing missing sysrepo modules...
/Volumes/test/openwrt/build_dir/target-x86_64_musl/netopeer2-2.2.13/ipkg-install/usr/share/netopeer2/scripts/setup.sh: line 125: /Volumes/test/openwrt/staging_dir/target-x86_64_musl/usr/bin/sysrepoctl: cannot execute binary file: Exec format error
/Volumes/test/openwrt/build_dir/target-x86_64_musl/netopeer2-2.2.13/ipkg-install/usr/share/netopeer2/scripts/setup.sh: line 137: /Volumes/test/openwrt/staging_dir/target-x86_64_musl/usr/bin/sysrepoctl: cannot execute binary file: Exec format error
CMake Error at cmake_install.cmake:111 (message):
   scripts/setup.sh failed: 126
```
That's my progress so far.

robimarko, 2024-05-02:
Well, removing libfrr and vtysh as separate packages requires quite a big rework of the makefile.

Maybe just hide vtysh and default to selected on it

lucize, 2024-05-02:
don't think is a good option to get rid of vtysh, but default it on watch-frr selection I guess it's an ok trade out

robimarko, 2024-05-03:
@httpstorm That is what I was thinking, just always set it as enabled for the build option, but libfrr needs to be converted so their respective packages are removed as it doesn't compile without them at all.

httpstorm, 2024-05-05:
@robimarko 
I merged `frr-libfrr` and `frr-vtysh` into `frr` https://github.com/openwrt/packages/pull/24083 . I intentionally left it as a separate commit. Robi, I will be on a vacation for a few days, maybe more. Please review the changes in the PR and feel free to take it from here. Best of luck!

closed 2024-07-20 by httpstorm