openwrt/packages #24066: `shadowsocks-libev` Compiler error!

zxlhhyccc, 2024-05-02:
```
checking whether a program can dlopen itself... cross
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... no
checking whether to build static libraries... yes
checking for pcre2-config... pcre2-config
checking for pcre2 headers in /home/lin/ax6-6.1/staging_dir/target-aarch64_cortex-a53_musl/usr/include... ok
configure: adding /home/lin/ax6-6.1/staging_dir/target-aarch64_cortex-a53_musl/usr/lib  to RPATH
checking for library containing pcre2_match_8... -lpcre2-8
checking pcre2.h... ok
checking pcre2/pcre2.h... not found
checking for aarch64-openwrt-linux-gcc... (cached) aarch64-openwrt-linux-musl-gcc
checking whether the compiler supports GNU C... (cached) yes
checking whether aarch64-openwrt-linux-musl-gcc accepts -g... (cached) yes
checking for aarch64-openwrt-linux-musl-gcc option to enable C11 features... (cached) none needed
checking whether aarch64-openwrt-linux-musl-gcc understands -c and -o together... (cached) yes
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for thread local storage (TLS) class... __thread
checking for mbedtls_cipher_setup in -lmbedcrypto... yes
checking whether mbedtls supports Cipher Feedback mode or not... configure: error: MBEDTLS_CIPHER_MODE_CFB required
Makefile:132: recipe for target '/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/shadowsocks-libev-3.3.5/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[2]: *** [/home/lin/ax6-6.1/build_dir/target-aarch64_cortex-a53_musl/shadowsocks-libev-3.3.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/home/lin/ax6-6.1/feeds/packages/net/shadowsocks-libev'
time: package/feeds/packages/shadowsocks-libev/compile#26.37#2.79#29.48
    ERROR: package/feeds/packages/shadowsocks-libev failed to build.
package/Makefile:127: recipe for target 'package/feeds/packages/shadowsocks-libev/compile' failed
make[1]: *** [package/feeds/packages/shadowsocks-libev/compile] Error 1
make[1]: Leaving directory '/home/lin/ax6-6.1'
/home/lin/ax6-6.1/include/toplevel.mk:231: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2
`````

microdockman, 2024-05-05:
https://github.com/coolsnowwolf/lede/tree/master/package/libs

microdockman, 2024-05-05:
https://github.com/coolsnowwolf/lede/tree/master/package/libs

Replace the corresponding directory
https://github.com/openwrt/openwrt/tree/main/package/libs 
mbedtls   ustream-ssl  uclient

zxlhhyccc, 2024-05-05:
@microdockman You're demoting.