openwrt/packages #24075: gptfdisk: Unable to generate a proper UUID

fanck0605, 2024-05-04:
Maintainer: unknown
Environment: (put here arch, model, OpenWrt version)

arch: x86_64
version: v23.05.3

Description:

I want to recreate the partition by **`sgdisk`** command, but it print the warning log.

```
# sgdisk -og -N1 -p /dev/sdb
Warning! Unable to generate a proper UUID! Creating an improper one as a last
resort! Windows 7 may crash if you save this partition table!
```

It seems that the uuid module does not works well. Also i don't know weather there will be any issues in my created partition.
Is there anyone can fix it. Thanks!

krant, 2024-05-05:
@fanck0605 gptfdisk 1.0.10 has it fixed. If you want it to become available on OpenWrt 23.05 ASAP - I'd suggest you to create PR with https://github.com/openwrt/packages/commit/ad6344d0455038e374f57a7fb15d3d1ace8d889b cherry-picked to openwrt-23.05 branch.

neheb, 2024-05-18:
backported.

closed 2024-05-18 by neheb