openwrt/packages #24106: <nut-upsmon> v. 2.8.1-r1 bug

grifos, 2024-05-09:
Please note: I'm really new to GitHub, I apologize for any mistakes.

Maintainer: I'm not sure if there is an official maintainer, I'm adding @ruralroots who submitted the update to v. 2.8.1

Environment: Belkin RT1800 (MT7621, mipsel_24kc), running OpenWrt snapshot r26235-0d74b2a1e5 from 2024-05-08.

nut-upsmon v. 2.8.1-r1 

Description:
I've also reported this on the forum https://forum.openwrt.org/t/bug-in-nut-upsmon-v-2-8-1-r1/197258

It looks like there is a typo for path /var/run/upsmon.pid (the /var is missing), causing multiple instances of nut-upsmon to spawn. See the below logs from nut-upsmon v. 2.8.1-r1 vs. v. 2.8.0-3:

OpenWrt snapshot r26235-0d74b2a1e5 from 2024-05-08 - nut-upsmon v. 2.8.1-r1 (upsmon spawns multiple processes):

```
Wed May  8 02:55:21 2024 daemon.err upsmon[7155]:    0.000005	fopen /run/upsmon.pid: No such file or directory
Wed May  8 02:55:21 2024 daemon.err upsmon[7155]:    0.000371	Could not find PID file to see if previous upsmon instance is already running!
Wed May  8 02:55:21 2024 daemon.err upsmon[7155]:    0.000989	Using power down flag file /var/run/killpower
Wed May  8 02:55:21 2024 daemon.err upsmon[7155]:    0.001812	UPS: qnapups@localhost (primary) (power value 1)
Wed May  8 02:55:21 2024 daemon.err upsmon[7155]:    0.002083	[D1] debug level is '1'
Wed May  8 02:55:21 2024 daemon.err upsmon[7155]:    0.006400	writepid: fopen /run/upsmon.pid: No such file or directory
```

OpenWrt 23.05.3 - nut-upsmon v. 2.8.0-3 (upsmon is working fine):

```
Wed May  8 11:25:35 2024 daemon.err upsmon[7159]:    0.000000	fopen /var/run/upsmon.pid: No such file or directory
Wed May  8 11:25:35 2024 daemon.err upsmon[7159]:    0.000204	Could not find PID file to see if previous upsmon instance is already running!
Wed May  8 11:25:35 2024 daemon.err upsmon[7159]:
Wed May  8 11:25:35 2024 daemon.err upsmon[7159]:    0.000829	Using power down flag file /var/run/killpower
Wed May  8 11:25:35 2024 daemon.err upsmon[7159]:    0.001554	UPS: qnapups@localhost (primary) (power value 1)
Wed May  8 11:25:35 2024 daemon.err upsmon[7159]:    0.001890	[D1] debug level is '1'
Wed May  8 11:25:35 2024 daemon.err upsmon[7159]:    0.002058	[D1] debug level is '1'
Wed May  8 11:25:35 2024 daemon.err upsmon[7159]:    0.007170	[D1] Saving PID 7199 into /var/run/upsmon.pid
```

tofurky, 2024-08-23:
Seeing the same. I did not encounter the issue until I moved from 23.05.x (2.8.0) to master (2.8.1).

Checking changes from 2.8.0 -> 2.8.1, it looks like if the pid path isn't explicitly passed when running ./configure, it uses /run if /run exists on the build system:

https://github.com/networkupstools/nut/commit/1e790b44dc78673948740e9288a1e47e3b41c95f#diff-49473dca262eeab3b4a43002adb08b4db31020d190caaad1594b47f1d5daa810R156-R172

```
dnl Historically this refers to *system location* for PID files,
dnl and more specifically that for `upsmon` (running as `root`).
dnl See also ALTPIDPATH (defaulted to STATEPATH) setting below
dnl for the unprivileged daemons (`upsd`, drivers).
PIDPATH="/var/run"
dnl Honour new LFS recommendations if applied on the build system:
AS_IF([test -d "/run"], [PIDPATH="/run"])
```
So we probably just need to specify that when building.

```
  --with-altpidpath=PATH  path for NUT driver/upsd .pid files not running as
                          root (<statepath>)
  --with-pidpath=PATH     Path for root-owned .pid files (${PIDPATH},
                          typically /var/run)
```

tofurky, 2024-08-23:
I opened https://github.com/openwrt/packages/pull/24874 which specifes `--with-pidpath=/var/run/nut` and (in testing) fixes the issue.

closed 2024-10-02 by neheb