openwrt/packages #24114: Unbound: cannot offload log from syslog

scottsteely, 2024-05-10:
Maintainer: @
Environment: ARM, PI-4  OpenWrt 23.05.3

Description:
Hello, I am trying to move the output of unbound to /var/lib/unbound.log but haven't been successful. 
I have SET the UCI parameters:

```
    option log-replies 'yes'
    option log-tag-queryreply 'yes'
    option log-local-actions 'yes'
    option logfile '/var/lib/unbound/unbound.log'
    option use-syslog: 'no'
```
After setting these it appears that  use-syslog is hardcoded in at /usr/lib/unbound/unbound.sh : 

"""
unbound_conf() {
.
.
.
 # Logging
    echo "  use-syslog: yes"
"""

I have also manual changed that but still no log in /var/lib/unbound. 

I understand I can completely remove UCI from unbound, there are some dynamic settings that I don't want to break.

brada4, 2024-05-10:
I think need to remove colon

dave14305, 2024-05-11:
Valid UCI options:
https://github.com/openwrt/packages/blob/master/net/unbound/files/README.md#complete-uci

The hard-coded `use-syslog` will be overridden if you add `logfile: "/var/lib/unbound/unbound.log"` in unbound_srv.conf. Just don’t mix UCI with native Unbound options in the .conf.

scottsteely, 2024-05-11:
Thanks clearing up some misconceptions I had about UCI options. I was ably to get it to work, but I changed so many config files I'm not even sure which one did the trick.

closed 2024-05-11 by scottsteely