openwrt/packages #24116: iperf3 segmentation fault on ath79-generic

Djfe, 2024-05-11:
Description:
iperf3 runs into a segmentation fault on ath79-generic when trying to calculcate the SUM at the end of a tcp run.
I reproduced this on several devices. (3 UniFi AP Outdoor+ and 1 UniFi AP Pro)
ramips-mt7621 doesn't appear to be affected (Zyxel-NWA55AXE)
x86-64 isn't affected

I could test this on further devices if required or if you can't reproduce the issue on your own devices right away.

Shortened output:
```
~# iperf3 -c  freifunkm01.ring.nlnog.net -p
5202 -R  -P10 -t15
Connecting to host freifunkm01.ring.nlnog.net, port 5202
Reverse mode, remote host freifunkm01.ring.nlnog.net is sending
[...]
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]  14.00-15.00  sec   768 KBytes  6.27 Mbits/sec
[  7]  14.00-15.11  sec   640 KBytes  4.73 Mbits/sec
[  9]  14.00-15.11  sec   896 KBytes  6.62 Mbits/sec
[ 11]  14.17-15.11  sec   384 KBytes  3.35 Mbits/sec
[ 13]  14.17-15.11  sec   256 KBytes  2.23 Mbits/sec
[ 15]  14.17-15.11  sec   384 KBytes  3.35 Mbits/sec
[ 17]  14.17-15.18  sec   640 KBytes  5.18 Mbits/sec
[ 19]  14.17-15.18  sec   640 KBytes  5.18 Mbits/sec
[ 21]  14.27-15.18  sec   512 KBytes  4.57 Mbits/sec
[ 23]  14.27-15.18  sec   768 KBytes  6.86 Mbits/sec
[SUM]  14.00-15.00  sec  5.75 MBytes  48.1 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-15.28  sec  14.4 MBytes  7.92 Mbits/sec  109             sender
[  5]   0.00-15.00  sec  13.2 MBytes  7.41 Mbits/sec                  receiver
[  7]   0.00-15.28  sec  6.73 MBytes  3.70 Mbits/sec   37             sender
[  7]   0.00-15.00  sec  6.50 MBytes  3.63 Mbits/sec                  receiver
[  9]   0.00-15.28  sec  15.1 MBytes  8.29 Mbits/sec  163             sender
[  9]   0.00-15.00  sec  13.5 MBytes  7.55 Mbits/sec                  receiver
[ 11]   0.00-15.28  sec  10.3 MBytes  5.68 Mbits/sec   82             sender
[ 11]   0.00-15.00  sec  9.75 MBytes  5.45 Mbits/sec                  receiver
[ 13]   0.00-15.28  sec  5.18 MBytes  2.85 Mbits/sec   58             sender
[ 13]   0.00-15.00  sec  4.88 MBytes  2.73 Mbits/sec                  receiver
[ 15]   0.00-15.28  sec  11.7 MBytes  6.40 Mbits/sec  109             sender
[ 15]   0.00-15.00  sec  10.9 MBytes  6.08 Mbits/sec                  receiver
[ 17]   0.00-15.28  sec  11.2 MBytes  6.14 Mbits/sec   89             sender
[ 17]   0.00-15.00  sec  10.8 MBytes  6.01 Mbits/sec                  receiver
[ 19]   0.00-15.28  sec  12.2 MBytes  6.69 Mbits/sec  121             sender
[ 19]   0.00-15.00  sec  11.6 MBytes  6.50 Mbits/sec                  receiver
[ 21]   0.00-15.28  sec  8.15 MBytes  4.48 Mbits/sec   24             sender
[ 21]   0.00-15.00  sec  7.88 MBytes  4.40 Mbits/sec                  receiver
[ 23]   0.00-15.28  sec  16.0 MBytes  8.77 Mbits/sec   57             sender
[ 23]   0.00-15.00  sec  15.2 MBytes  8.53 Mbits/sec                  receiver
Segmentation fault
```

Djfe, 2024-05-11:
This is also annoying when I run iperf3 against ath79-generic as a server.
(Client doesn't run in reverse -R)
because it will crash the server application that should keep on running.

brada4, 2024-05-11:
Whats in dmesg?

Djfe, 2024-05-12:
```
[83837.386363] do_page_fault(): sending SIGSEGV to iperf3 for invalid read access from 00000212
[83837.394845] epc = 77e6adc0 in libc.so[77dec000+a9000]
[83837.400021] ra  = 77e6cb04 in libc.so[77dec000+a9000]
```

brada4, 2024-05-12:
It is aslr changig memory layout, see the file offset does not change.

Djfe, 2024-05-12:
do you have an idea where to start? is ASLR not working correctly for the target?

neheb, 2024-05-12:
gdb would be useful to figure out where it's crashing.

could also try compiling without mips16.

brada4, 2024-05-13:
Aslr is ok, just explanation to OP that two code points are same.

neheb, 2024-06-08:
```
Program received signal SIGSEGV, Segmentation fault.
0x77f88624 in freeaddrinfo () from /lib/ld-musl-mips-sf.so.1
(gdb) bt
#0  0x77f88624 in freeaddrinfo () from /lib/ld-musl-mips-sf.so.1
#1  0x77f10cd1 in create_socket () from /usr/lib/libiperf.so.0
```

neheb, 2024-06-08:
Can someone try:

```
--- a/src/iperf_api.c
+++ b/src/iperf_api.c
@@ -4056,7 +4056,7 @@ iperf_print_results(struct iperf_test *test)
                                 iperf_printf(test, report_sender_not_available_summary_format, "SUM");
                         }
                         else {
-                          iperf_printf(test, report_sum_bw_retrans_format, mbuf, start_time, sender_time, ubuf, nbuf, total_retransmits, report_sender);
+                          iperf_printf(test, report_sum_bw_retrans_format, mbuf, start_time, sender_time, ubuf, nbuf, (int)total_retransmits, report_sender);
                         }
                 } else {
                     /* Summary sum, TCP without retransmits. */
```

neheb, 2024-06-08:
yeah, that worked.

closed 2024-06-09 by neheb