openwrt/packages #24119: znc: failure to build

egorenar, 2024-05-11:
Maintainer: @KanjiMonster 
Environment: (ARMv7 Processor rev 4 (v7l), OpenWrt SNAPSHOT r26260-b975221d3854)

znc 1.8.2 can no longer be built, please see the attached error log.
i was to build znc 1.9.0 but had to drop patches:
* 101-Reduce_rebuild_time.patch
* 120-openssl-deprecated.patch

znc 1.9.0 patch:
```
diff --git a/net/znc/Makefile b/net/znc/Makefile
index e2528909d3da..2203eeee09a9 100644
--- a/net/znc/Makefile
+++ b/net/znc/Makefile
@@ -8,13 +8,13 @@
 include $(TOPDIR)/rules.mk

 PKG_NAME:=znc
-PKG_VERSION:=1.8.2
-PKG_RELEASE:=6
+PKG_VERSION:=1.9.0
+PKG_RELEASE:=1

 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
 PKG_SOURCE_URL:=https://znc.in/releases \
                https://znc.in/releases/archive
-PKG_HASH:=ff238aae3f2ae0e44e683c4aee17dc8e4fdd261ca9379d83b48a7d422488de0d
+PKG_HASH:=8b99c9dbb21c1309705073460be9bfacb6f7b0e83a15fe5d4b7140201b39d2a1

 PKG_MAINTAINER:=Jonas Gorski <jonas.gorski@gmail.com>
 PKG_LICENSE:=Apache-2.0
```

Error log:

```
ninja: Entering directory `/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2'
[1/139] cd /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2 && /home/egorenar/Repositories/openwrt-rel/staging_dir/host/bin/cmake -D nightly= -D alpha_version= -D append_git_version=false -D gitcmd= -D srcdir=/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2 -D srcfile=/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/src/version.cpp.in -D destfile=/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/src/versionc.cpp -P /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/cmake/gen_version.cmake
[2/139] /home/egorenar/Repositories/openwrt-rel/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-13.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++  -I/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2=znc-1.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -std=c++11 -fPIC -include znc/zncconfig.h -MD -MT src/CMakeFiles/znclib.dir/FileUtils.cpp.o -MF src/CMakeFiles/znclib.dir/FileUtils.cpp.o.d -o src/CMakeFiles/znclib.dir/FileUtils.cpp.o -c /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/src/FileUtils.cpp
FAILED: src/CMakeFiles/znclib.dir/FileUtils.cpp.o
/home/egorenar/Repositories/openwrt-rel/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-13.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++  -I/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2=znc-1.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -std=c++11 -fPIC -include znc/zncconfig.h -MD -MT src/CMakeFiles/znclib.dir/FileUtils.cpp.o -MF src/CMakeFiles/znclib.dir/FileUtils.cpp.o.d -o src/CMakeFiles/znclib.dir/FileUtils.cpp.o -c /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/src/FileUtils.cpp
In file included from /home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/uenum.h:25,
                 from /home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/ucnv.h:52,
                 from /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/include/znc/Csocket.h:91,
                 from /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/include/znc/Socket.h:21,
                 from /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/include/znc/ExecSock.h:21,
                 from /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/src/FileUtils.cpp:18:
/home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/localpointer.h:561:26: error: parameter declared 'auto'
  561 | template <typename Type, auto closeFunction>
      |                          ^~~~
/home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/localpointer.h:573:76: error: template argument 2 is invalid
  573 |     explicit LocalOpenPointer(std::unique_ptr<Type, decltype(closeFunction)> &&p)
      |                                                                            ^
/home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/localpointer.h:583:78: error: template argument 2 is invalid
  583 |     LocalOpenPointer &operator=(std::unique_ptr<Type, decltype(closeFunction)> &&p) {
      |                                                                              ^
/home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/localpointer.h:599:59: error: template argument 2 is invalid
  599 |     operator std::unique_ptr<Type, decltype(closeFunction)> () && {
      |                                                           ^
/home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/uenum.h:69:1: note: invalid template non-type parameter
   69 | U_DEFINE_LOCAL_OPEN_POINTER(LocalUEnumerationPointer, UEnumeration, uenum_close);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/ucnv.h:597:1: note: invalid template non-type parameter
  597 | U_DEFINE_LOCAL_OPEN_POINTER(LocalUConverterPointer, UConverter, ucnv_close);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
[3/139] /home/egorenar/Repositories/openwrt-rel/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-13.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++  -I/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2=znc-1.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -std=c++11 -fPIC -include znc/zncconfig.h -MD -MT src/CMakeFiles/znclib.dir/Utils.cpp.o -MF src/CMakeFiles/znclib.dir/Utils.cpp.o.d -o src/CMakeFiles/znclib.dir/Utils.cpp.o -c /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/src/Utils.cpp
FAILED: src/CMakeFiles/znclib.dir/Utils.cpp.o
/home/egorenar/Repositories/openwrt-rel/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-13.2.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++  -I/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2=znc-1.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -std=c++11 -fPIC -include znc/zncconfig.h -MD -MT src/CMakeFiles/znclib.dir/Utils.cpp.o -MF src/CMakeFiles/znclib.dir/Utils.cpp.o.d -o src/CMakeFiles/znclib.dir/Utils.cpp.o -c /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/src/Utils.cpp
In file included from /home/egorenar/Repositories/openwrt-rel/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/unicode/uenum.h:25,
[... cut ...]
make[2]: *** [Makefile:377: /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/znc-1.8.2/.built] Error 1
make[2]: Leaving directory '/home/egorenar/Repositories/openwrt-rel/feeds/packages/net/znc'
time: package/feeds/packages/znc/compile#15.86#2.04#3.67
    ERROR: package/feeds/packages/znc failed to build.
make[1]: *** [package/Makefile:129: package/feeds/packages/znc/compile] Error 1
make[1]: Leaving directory '/home/egorenar/Repositories/openwrt-rel'
make: *** [/home/egorenar/Repositories/openwrt-rel/include/toplevel.mk:233: package/znc/compile] Error 2
```

egorenar, 2024-05-11:
This seem to be related to ICU update made recently: https://github.com/openwrt/packages/pull/24104

KanjiMonster, 2024-05-12:
I'll take a look.

esaaprillia, 2024-05-12:
same

```
[71/107] Generating src/harfbuzz-subset.cc with a custom command
[72/107] Compiling C++ object src/libharfbuzz-subset.so.0.60840.0.p/graph_gsubgpos-context.cc.o
[73/107] Compiling C++ object src/libharfbuzz-cairo.so.0.60840.0.p/hb-cairo-utils.cc.o
[74/107] Compiling C++ object src/libharfbuzz-cairo.so.0.60840.0.p/hb-cairo.cc.o
[75/107] Compiling C++ object src/libharfbuzz-icu.so.0.60840.0.p/hb-icu.cc.o
FAILED: src/libharfbuzz-icu.so.0.60840.0.p/hb-icu.cc.o 
aarch64-openwrt-linux-musl-g++ -Isrc/libharfbuzz-icu.so.0.60840.0.p -Isrc -I../src -I. -I.. -I/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include -I/home/runner/work/mosdns/mosdns/sdk/staging_dir/toolchain-aarch64_generic_gcc-13.2.0_musl/usr/include -I/home/runner/work/mosdns/mosdns/sdk/staging_dir/toolchain-aarch64_generic_gcc-13.2.0_musl/include -I/home/runner/work/mosdns/mosdns/sdk/staging_dir/toolchain-aarch64_generic_gcc-13.2.0_musl/include/fortify -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c++11 -fno-exceptions -fno-exceptions -fno-rtti -fno-threadsafe-statics -fvisibility-inlines-hidden -DHAVE_CONFIG_H -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/harfbuzz-8.4.0=harfbuzz-8.4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fPIC -Wno-non-virtual-dtor -MD -MQ src/libharfbuzz-icu.so.0.60840.0.p/hb-icu.cc.o -MF src/libharfbuzz-icu.so.0.60840.0.p/hb-icu.cc.o.d -o src/libharfbuzz-icu.so.0.60840.0.p/hb-icu.cc.o -c ../src/hb-icu.cc
In file included from /home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include/unicode/uset.h:36,
                 from /home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include/unicode/unorm2.h:35,
                 from ../src/hb-icu.cc:39:
/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include/unicode/localpointer.h:561:26: error: parameter declared 'auto'
  561 | template <typename Type, auto closeFunction>
      |                          ^~~~
/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include/unicode/localpointer.h:573:76: error: template argument 2 is invalid
  573 |     explicit LocalOpenPointer(std::unique_ptr<Type, decltype(closeFunction)> &&p)
      |                                                                            ^
/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include/unicode/localpointer.h:583:78: error: template argument 2 is invalid
  583 |     LocalOpenPointer &operator=(std::unique_ptr<Type, decltype(closeFunction)> &&p) {
      |                                                                              ^
/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include/unicode/localpointer.h:599:59: error: template argument 2 is invalid
  599 |     operator std::unique_ptr<Type, decltype(closeFunction)> () && {
      |                                                           ^
/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include/unicode/uset.h:358:1: note: invalid template non-type parameter
  358 | U_DEFINE_LOCAL_OPEN_POINTER(LocalUSetPointer, USet, uset_close);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/usr/include/unicode/unorm2.h:286:1: note: invalid template non-type parameter
  286 | U_DEFINE_LOCAL_OPEN_POINTER(LocalUNormalizer2Pointer, UNormalizer2, unorm2_close);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
[76/107] Compiling C++ object src/libharfbuzz-cairo.so.0.60840.0.p/hb-static.cc.o
[77/107] Compiling C++ object src/libharfbuzz-subset.so.0.60840.0.p/hb-subset-plan.cc.o
[78/107] Compiling C++ object src/libharfbuzz-subset.so.0.60840.0.p/hb-subset.cc.o
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:55: /home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/harfbuzz-8.4.0/.built] Error 1
```

KanjiMonster, 2024-05-12:
Should be fixed via https://github.com/openwrt/packages/commit/c5fcfc0c1e7c69db0929aadedc4906332782c87a.

closed 2024-05-12 by KanjiMonster