openwrt/packages #24148: [pbr] pbr has wan/wan6 interface names hard-coded, conflicts with wwan/wwan6 and travelmate

schuettecarsten, 2024-05-13:
pbr has wan/wan6 interface names hard-coded. On my setup, I have a wan interface and also a wwan/wwan6 interface to access internet via a wifi bridge. pbr does not detect this interfaces and waits for "wan" access forever. I suggest to add an option to configure the name of the wan interfaces.

stangri, 2024-05-13:
Upgrade to a current release from my repo and check README for option names.

stangri, 2024-05-13:
> That's great, maybe it's a good idea to make it possible to add more than one interface name - to allow cable WAN, wifi WAN (WWAN) and QMI on the same device, depending on what's attached to the USB port.

You're welcome to contribute the code.

> When do you plan to upgrade the packages in this repo?

It has been upgraded in this repo.

schuettecarsten, 2024-05-14:
> > When do you plan to upgrade the packages in this repo?
> It has been upgraded in this repo.

pbr is still 1.1.4...?

stangri, 2024-05-14:
> > > When do you plan to upgrade the packages in this repo?
> > > It has been upgraded in this repo.
> 
> pbr is still 1.1.4...?

That version already has the options for specifying wan/wan6 names, does it not?

schuettecarsten, 2024-05-14:
Yes, you're right.

closed 2024-05-14 by schuettecarsten