openwrt/packages #24166: prometheus-node-exporter-lua: nat_traffic.lua produce duplicate samples

redjumper, 2024-05-15:
After updating to Prometheus 2.52, Prometheus is reporting 'Duplicate sample for timestamp' errors.

> ts=2024-05-15T07:37:38.942Z caller=scrape.go:1777 level=debug component="scrape manager" scrape_pool=openwrt_node_exporter target=http://192.168.1.1:9100/metrics msg="Duplicate sample for timestamp" series="node_nat_traffic{dest=\"192.168.1.1\",src=\"192.168.2.2\"}"
> ts=2024-05-15T07:37:38.942Z caller=scrape.go:1777 level=debug component="scrape manager" scrape_pool=openwrt_node_exporter target=http://192.168.1.1:9100/metrics msg="Duplicate sample for timestamp" series="node_nat_traffic{dest=\"192.168.1.1\",src=\"192.168.2.2\"}

Maybe it is related to this change in [ Prometheus 2.52](https://github.com/prometheus/prometheus/releases/tag/v2.52.0)
> [ENHANCEMENT] Scrape: Increment prometheus_target_scrapes_sample_duplicate_timestamp_total metric on duplicated series during one scrape. https://github.com/prometheus/prometheus/pull/12933

acolombier, 2024-06-04:
It looks like there is a genuine issue with the exporter which is listing duplicate samples value, and Prometheus 2.52 has only started reporting it. 
Here is an example snippet for what I can `curl` on the metric endpoint
```
node_nat_traffic{dest="1.1.1.1",src="192.168.0.17"} x
node_nat_traffic{dest="1.1.1.1",src="192.168.0.17"} y
...
```
I guess the two options could be:
1. Sum up bytes into a single gauge cample
2. Include sport and dport to de-duplicate samples

Solution `1.` is probably the best as `2.` will increase significantly metric cardinality which is suboptimal for Prometheus performance and stability and would also break integration (AlertManager, Grafana, ...)

Let me know if you need some hep on fixing or testing @champtar ! :)

champtar, 2024-06-05:
@acolombier If you can prepare a PR with solution 1 that would be perfect, I'm slowly trying to cleanup the backlog and any help is appreciated

PR #24328 (2024-06-15): prometheus-node-exporter-lua: remove duplicated nat samples
Maintainer: @champtar 
Compile tested:  -
Run tested: `arm_cortex-a15_neon-vfpv4/ipq806x/generic/22.03.5 r20134-5f15225c1e`

Fixes duplicates samples for NAT gauge

Fixes #24166

_Disclaimer: this is my first time writing Lua._

I've patched my device manually and will monitor the behaviour during the next few hours.
commit 611b3724 (2024-06-18): prometheus-node-exporter-lua: remove duplicated nat samples

Merge duplicate src/dest samples by suming their value (bytes count)

Fixes #24166

[bump version number]

(cherry picked from commit cd8f67298c588a6e80c5991b6921222eea10a06f)
closed 2024-06-15 by champtar