openwrt/packages #24168: shadowsocks-libev: configure: error: MBEDTLS_CIPHER_MODE_CFB required

alex9434, 2024-05-15:
Maintainer: @yousong @ffontaine 
Environment: OpenWrt snapshot from today (fresh sync from git)

Description:
Error message when building shadowsocks-libev
```
checking for mbedtls_cipher_setup in -lmbedcrypto... yes
checking whether mbedtls supports Cipher Feedback mode or not... configure: error: MBEDTLS_CIPHER_MODE_CFB required
make[2]: *** [Makefile:130: /opt/rpi4/snapshot/openwrt/build_dir/target-aarch64_cortex-a72_musl/shadowsocks-libev-3.3.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
```
Similar issue has been raised with upstream: https://github.com/shadowsocks/shadowsocks-libev/issues/2868

suyoulin, 2024-05-24:
Hello. Did you have fix the fault? Thanks

M4TRIX04, 2024-05-30:
still unsolved

fuqiang03, 2024-05-30:
@zxlhhyccc 
[image]
完全官方op  main库

zxlhhyccc, 2024-05-31:
> 完全官方op main库

@fuqiang03 请添加我贴出来的补丁即可编译通过，补丁名字可以随便起，但能否可用不清楚，确定是可以运行。

Wang-P, 2024-06-14:
能编译，但不能正常使用，我是通过passwall2加载ss节点使用的

Falld0wn18, 2024-06-15:
> > 完全官方op main库
> 
> @fuqiang03 已修改补丁，我贴的是原来有问题的补丁，不好意思。但能否可用不清楚，绝对是可以编译通过和运行。

能编译通过且服务能启动，但测试是无法订阅和使用节点

Wang-P, 2024-06-26:
补充：今天重新编译了openwrt，使用了上面补丁，顺利编译成功，ss节点可以正常使用。mbedtls是3.6.0版，shadowsocks-libev为3.3.5版

yousong, 2024-06-26:
The upstream project has no update for about 2 years [Link](https://github.com/shadowsocks/shadowsocks-libev/commits/master/).   Now that it does not even compile with current openwrt environment, I'm going to remove this package and its accompanying luci app.

zxlhhyccc, 2024-06-29:
> 补充：今天重新编译了openwrt，使用了上面补丁，顺利编译成功，ss节点可以正常使用。mbedtls是3.6.0版，shadowsocks-libev为3.3.5版

你确定能正常使用吗？能截个图出来吗？如果能正常使用，我就提pr。。。

Wang-P, 2024-06-30:
> > 补充：今天重新编译了openwrt，使用了上面补丁，顺利编译成功，ss节点可以正常使用。mbedtls是3.6.0版，shadowsocks-libev为3.3.5版
> 
> 你确定能正常使用吗？能截个图出来吗？如果能正常使用，我就提pr。。。

当然确定。
以下是截图：
[image]
[image]

zxlhhyccc, 2024-06-30:
> 当然确定。

看你的截图，貌似是xray的ss功能：
[image]
你确定运行的是纯shadowsocks-libev而不是xray？

Wang-P, 2024-06-30:
晚上重新在vps上编译了shadowsocks-libev 3.3.5，libsodium为1.0.20版，mbedtls为3.6.0版，并使用了shadowsockts-windows科学上网，并关闭了路由翻墙，证实该补丁是有效的。
[image]

zxlhhyccc, 2024-06-30:
> 晚上重新在vps上编译了shadowsocks-libev 3.3.5，libsodium为1.0.20版，mbedtls为3.6.0版，并使用了shadowsockts-windows科学上网，并关闭了路由翻墙，证实该补丁是有效的。

我要确定在passwall或者passwall2或者ssrp使用我的补丁编译的shadowsocks-libev能翻墙么？

msdos03, 2024-07-03:
> > 晚上重新在vps上编译了shadowsocks-libev 3.3.5，libsodium为1.0.20版，mbedtls为3.6.0版，并使用了shadowsockts-windows科学上网，并关闭了路由翻墙，证实该补丁是有效的。
> 
> 我要确定在passwall或者passwall2或者ssrp使用我的补丁编译的shadowsocks-libev能翻墙么？ 还有个问题就是，客户端和服务端是不是都需要打补丁才能使用，还是服务端不需打补丁，只要客户端打了补丁也可以使用？

I have tested your patch in ssr-plus and it compiles and works fine. I think now you can create a pull request for it.

msdos03, 2024-07-03:
> > 晚上重新在vps上编译了shadowsocks-libev 3.3.5，libsodium为1.0.20版，mbedtls为3.6.0版，并使用了shadowsockts-windows科学上网，并关闭了路由翻墙，证实该补丁是有效的。
> 
> 我要确定在passwall或者passwall2或者ssrp使用我的补丁编译的shadowsocks-libev能翻墙么？ 还有个问题就是，客户端和服务端是不是都需要打补丁才能使用，还是服务端不需打补丁，只要客户端打了补丁也可以使用？

我使用你的补丁编译了shadowsocks-libev和ssr-plus，发现shadowsocks-libev无法正常工作。
为了测试，我使用`ss-local -c /tmp/tcp-only-ssr-retcp.json -b 127.0.0.2`启动了客户端，然后运行了
`curl --socks5 127.0.0.2:1234 https://google.com`结果得到了如下报错
`curl: (35) ssl_handshake returned: (-0x7280) SSL - The connection indicated an EOF`
再看ss-local的日志发现打印出` 2024-07-03 12:51:04 ERROR: invalid password or cipher`
我可以确认配置文件中密码是正确的

msdos03, 2024-07-06:
@zxlhhyccc 经过一段时间的研究和尝试，我修改了你的补丁使其现在至少在我这里可以正常工作了。我推测之前有人报告工作正常是因为没有使用aes-128-gcm算法，所以没有触发潜在问题。你觉得现在是时候进行pull request了吗？
```
--- a/m4/mbedtls.m4
+++ b/m4/mbedtls.m4
@@ -31,7 +31,7 @@ AC_DEFUN([ss_MBEDTLS],
   AC_COMPILE_IFELSE(
     [AC_LANG_PROGRAM(
       [[
-#include <mbedtls/config.h>
+#include <mbedtls/mbedtls_config.h>
       ]],
       [[
 #ifndef MBEDTLS_CIPHER_MODE_CFB
@@ -48,7 +48,7 @@ AC_DEFUN([ss_MBEDTLS],
   AC_COMPILE_IFELSE(
     [AC_LANG_PROGRAM(
       [[
-#include <mbedtls/config.h>
+#include <mbedtls/mbedtls_config.h>
       ]],
       [[
 #ifndef MBEDTLS_ARC4_C
@@ -64,7 +64,7 @@ AC_DEFUN([ss_MBEDTLS],
   AC_COMPILE_IFELSE(
     [AC_LANG_PROGRAM(
       [[
-#include <mbedtls/config.h>
+#include <mbedtls/mbedtls_config.h>
       ]],
       [[
 #ifndef MBEDTLS_BLOWFISH_C
@@ -80,7 +80,7 @@ AC_DEFUN([ss_MBEDTLS],
   AC_COMPILE_IFELSE(
     [AC_LANG_PROGRAM(
       [[
-#include <mbedtls/config.h>
+#include <mbedtls/mbedtls_config.h>
       ]],
       [[
 #ifndef MBEDTLS_CAMELLIA_C
--- a/src/crypto.c
+++ b/src/crypto.c
@@ -103,7 +103,7 @@ crypto_md5(const unsigned char *d, size_t n, unsigned char *md)
     if (md == NULL) {
         md = m;
     }
-#if MBEDTLS_VERSION_NUMBER >= 0x02070000
+#if MBEDTLS_VERSION_NUMBER < 0x03000000 && MBEDTLS_VERSION_NUMBER >= 0x02070000
     if (mbedtls_md5_ret(d, n, md) != 0)
         FATAL("Failed to calculate MD5");
 #else
--- a/src/aead.c
+++ b/src/aead.c
@@ -178,8 +178,8 @@ aead_cipher_encrypt(cipher_ctx_t *cipher_ctx,
     case AES192GCM:
     case AES128GCM:
 
-        err = mbedtls_cipher_auth_encrypt(cipher_ctx->evp, n, nlen, ad, adlen,
-                                          m, mlen, c, clen, c + mlen, tlen);
+        err = mbedtls_cipher_auth_encrypt_ext(cipher_ctx->evp, n, nlen, ad, adlen,
+                                          m, mlen, c, mlen + tlen, clen, tlen);
         *clen += tlen;
         break;
     case CHACHA20POLY1305IETF:
@@ -226,8 +226,8 @@ aead_cipher_decrypt(cipher_ctx_t *cipher_ctx,
     // Otherwise, just use the mbedTLS one with crappy AES-NI.
     case AES192GCM:
     case AES128GCM:
-        err = mbedtls_cipher_auth_decrypt(cipher_ctx->evp, n, nlen, ad, adlen,
-                                          m, mlen - tlen, p, plen, m + mlen - tlen, tlen);
+        err = mbedtls_cipher_auth_decrypt_ext(cipher_ctx->evp, n, nlen, ad, adlen,
+                                          m, mlen, p, mlen - tlen, plen, tlen);
         break;
     case CHACHA20POLY1305IETF:
         err = crypto_aead_chacha20poly1305_ietf_decrypt(p, &long_plen, NULL, m, mlen,
@@ -724,9 +724,9 @@ aead_key_init(int method, const char
     if (method >= CHACHA20POLY1305IETF) {
         cipher_kt_t *cipher_info = (cipher_kt_t *)ss_malloc(sizeof(cipher_kt_t));
         cipher->info             = cipher_info;
-        cipher->info->base       = NULL;
-        cipher->info->key_bitlen = supported_aead_ciphers_key_size[method] * 8;
-        cipher->info->iv_size    = supported_aead_ciphers_nonce_size[method];
+        cipher->info->private_base_idx       = 0;
+        cipher->info->private_key_bitlen = supported_aead_ciphers_key_size[method] * 8;
+        cipher->info->private_iv_size    = supported_aead_ciphers_nonce_size[method];
     } else {
         cipher->info = (cipher_kt_t *)aead_get_cipher_type(method);
     }
--- a/src/stream.c
+++ b/src/stream.c
@@ -174,7 +174,7 @@ cipher_nonce_size(const cipher_t *cipher)
     if (cipher == NULL) {
         return 0;
     }
-    return cipher->info->iv_size;
+    return cipher->info->private_iv_size;
 }
 
 int
@@ -192,7 +192,7 @@ cipher_key_size(const cipher_t *cipher)
         return 0;
     }
     /* From Version 1.2.7 released 2013-04-13 Default Blowfish keysize is now 128-bits */
-    return cipher->info->key_bitlen / 8;
+    return cipher->info->private_key_bitlen / 8;
 }
 
 const cipher_kt_t *
@@ -645,9 +645,9 @@ stream_key_init(int method, const char
     if (method == SALSA20 || method == CHACHA20 || method == CHACHA20IETF) {
         cipher_kt_t *cipher_info = (cipher_kt_t *)ss_malloc(sizeof(cipher_kt_t));
         cipher->info             = cipher_info;
-        cipher->info->base       = NULL;
-        cipher->info->key_bitlen = supported_stream_ciphers_key_size[method] * 8;
-        cipher->info->iv_size    = supported_stream_ciphers_nonce_size[method];
+        cipher->info->private_base_idx       = 0;
+        cipher->info->private_key_bitlen = supported_stream_ciphers_key_size[method] * 8;
+        cipher->info->private_iv_size    = supported_stream_ciphers_nonce_size[method];
     } else {
         cipher->info = (cipher_kt_t *)stream_get_cipher_type(method);
     }
```

zxlhhyccc, 2024-07-07:
@msdos03 问题确实解决了，我来提pr

yousong, 2024-07-10:
Sorry guys.  Let's move on.   #24540

closed 2024-07-10 by yousong